JMRI/JMRI #10558: DCS 210+ should show as a system connection

KenC57, 2022-01-09:
It has a USB connection like the DCS 240. But it isn't listed as a system connection. I presume using the PR4 or DCS240 system connection is equal to this. But we've added some of the others I think this got missed. When it first came out as the DCS210, that one didn't have the USB. But the DCS 210+ does include the USB.

danielb987, 2022-01-09:
@devel-bobm 
Can you take a look at this?

devel-bobm, 2022-01-09:
Yes, but I cannot get to it _any time soon_, because "life is getting in the way".  This task is one of several JMRI tasks which have been on my plate for a long time.

Also note that we do not have a DCS210+ "decoder definition" for configuring the command station via a JMRI Roster entry.  This is something else which is "needed".  This is also on my "to-do" list.

In current JMRI versions, I would select PR4 as the connection type and DCS210 as the command station type.  This ought to be as close as one can get at the current time.  Downsides?  The "configure command station opsws" tool does not have the correct OpSw settings.  Upside: " It will, so far as I know, handle _every other thing_ that can be done via the LocoNet menu.

(And I would also recommend _against_ using the integrated USB functionality on the DCS210+, because of the problems some users experience with the interface "locking-up" under unknown-to-me conditions.  This comment will likley appear in a help page for the DCS210+.)

londontk, 2022-01-28:
As of Friday 01/28/2022 This is a VERY BIG issue for me. Unable to proceed.
Maybe you could fix this soon.! I'm using an Intel NUC computer running Ubuntu 20.03.
Installed Java and JMRI. Ran DecoderPro which runs the DecoderPro Wizard. I set system manufacturer
to Digitrax and I'm unable to select a DCS210+ as my connection. PLEASE FIX!

BHoffman351, 2022-01-28:
@londontk Perhaps you should ask for a refund of the JMRI purchase price.

londontk, 2022-01-28:
Maybe I should begin looking seriously at an alternative for JMRI.!!!
Didn't get past the first screen. JMRI is not impressing me so far.

devel-bobm, 2022-01-28:
@londontk,

I do not know what has "set you off".  Your attitude is not impressing me.  But I will _attempt_ to provide a few  alternate solutions.  It is up to you to accept or decline the alternatives.

- Select the DCS240 in your JMRI Configuration Profile entry, even though you are using the DCS210+.  This _will_ pretty much "just plain work".  I do this occasionally.

- Select the "PR4" device as the "System Connection", and the select the "DCS210 command station" as the "Command Station Type".  That _also_ will work.  Again, this is something that I do occasionally.

Either of those options will work.

Note that  when you work with Digitrax hardware, JMRI itself _does not do much of anything_ with the "Command Station Type" information from the Configuration Profile.  That means that you will _most likely not_ see any difference in JMRI functionality when using a DCS210+ command station but configuring JMRI for a DCS240 command station or when JMRI is configured for interfacing via PR4 to a DCS210 command station. 

One place where you _cannot_ simply work with a different command station type is when configuring a command station via the JMRI "Symbolic Programmer" which is normally accessed via a JMRI "Roster" entry.  If you need to configure your DCS210+ command station options, do so using the methods described in the DCS210+ manual.

On the comment 
> Maybe you could fix this soon.!

I, like most JMRI developers, am a _volunteer_.  I work with JMRI, and help develop JMRI, as part of my hobby.  I am also the most-likely JMRI developer to "fix" this.  But my personal life includes commitments which take precedence over JMRI code development.  So if I need to code it, _it will have to wait until I have the hobby time available_.

Whining, by you or by any other JMRI user, _will not_ change that.

bobjacobsen, 2022-02-18:
See PR #10705

danielb987, 2022-02-19:
PR #10705 is now merged and will be included in the next JMRI test release 4.99.4.

closed 2022-02-19 by danielb987