JMRI/JMRI #10570: Python3 Issues

dsand47, 2022-01-11:
The AbstractAutomaton has a problem or I am doing it wrong.

```
import jmri as jmri
import java
from jmri.jmrit.automat import AbstractAutomaton

exec( open("jython/jmri_bindings.py3").read() )

class AutomatExample(AbstractAutomaton) :
    def init(self):
        self.count = 10
        return

    def handle(self):
        print('Count: {}'.format(self.count))
        self.count -= 1
        if self.count < 1: return False

        self.waitMsec(10000)
        return True

# end of class definition

a = AutomatExample()
a.setName("Automat test script")
a.start()
```

```
15:46:16,126 automat.AbstractAutomaton             WARN  - Unexpected Exception ends AbstractAutomaton thread [Automat test script]
AttributeError: 'PythonJavaExtenderClass' object has no attribute 'waitMsec'
	at <python> handle(<eval>:18:375-394)
	at org.graalvm.sdk/org.graalvm.polyglot.Value.invokeMember(Value.java:932)
	at com.oracle.truffle.host.adapters.AbstractAutomaton$$Adapter.handle(Unknown Source)
	at jmri.jmrit.automat.AbstractAutomaton.run(AbstractAutomaton.java:149)
	at com.oracle.truffle.host.adapters.AbstractAutomaton$$Adapter.run(Unknown Source)
	at java.base/java.lang.Thread.run(Thread.java:829)
```

Is the Jython option being used?

$ graalpython --jvm --experimental-options --python.EmulateJython

Syntax errors do not indicate the line number.  Finding them is difficult.  Basic errors, such as the print change, can be found using the command line, but that does not work after adding any JMRI references.

Should Python3 have be in its own directory.  Putting the bindings script in jython means that the jython directory will be around long after Jython is gone.

danielb987, 2022-01-11:
@bobjacobsen 
Can you look at this? Thanks!

bobjacobsen, 2022-01-11:
I think we're using the right options, but they might not be getting all the way down to the engine.  Will check.

This line:
```
at <python> handle(<eval>:18:375-394)
```
says the error was encountered on line 18.  I _think_ that 375-394 is the character position in the entire script (not the line, clearly) but I haven't seen documentation of that.  But Python 3 often tags syntax errors far away, when it finally gives up on parsing something.

dsand47, 2022-01-11:
The runtime waitMsec error is the line number.  Syntax errors have no clues that I can see.

Here is a typical syntax error:
```
     [java] 15:41:23,424 jsr223graalpython.GraalJSScriptEngine WARN  - Exception in eval [AWT-EventQueue-0]
     [java] 15:41:23,429 startup.ScriptButtonModel             ERROR - Unable to run script /Users/das/JMRI/_Profiles/Python3.jmri/AutomatonTest.py3. [AWT-EventQueue-0]
     [java] javax.script.ScriptException: org.graalvm.polyglot.PolyglotException: SyntaxError: invalid syntax
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.toScriptException(GraalJSScriptEngine.java:512)
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.eval(GraalJSScriptEngine.java:489)
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.eval(GraalJSScriptEngine.java:422)
     [java] 	at java.scripting/javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
     [java] 	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:356)
     [java] 	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:293)
     [java] 	at apps.startup.ScriptButtonModel$ScriptButtonAction.actionPerformed(ScriptButtonModel.java:69)
     [java] 	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
```

bobjacobsen, 2022-01-11:
Well, this is ugly.  Try replacing the wait line with
```
        self.__super__.waitMsec(10000)
```
(two underscores before and after).  I got this from the "Inheritance from Java" section here: https://www.graalvm.org/reference-manual/python/Jython/    

Still looking for a better solution...

dsand47, 2022-01-12:
> Well, this is ugly. Try replacing the wait line with
> 
> ```
>         self.__super__.waitMsec(10000)
> ```
> 
> (two underscores before and after). I got this from the "Inheritance from Java" section here: https://www.graalvm.org/reference-manual/python/Jython/
> 
> Still looking for a better solution...

That does work.

dsand47, 2022-01-12:
More issues.  Current script:
```
import jmri as jmri
import java
from jmri.jmrit.automat import AbstractAutomaton

exec( open("jython/jmri_bindings.py3").read() )

class AutomatExample(AbstractAutomaton) :
#     def __init__(self, secs):
    def __init__(self):
        self.secs = timer

    def init(self):
        self.count = 10
        self.sensor = sensors.getSensor('ISCLOCKRUNNING')
        return

#     def setup(self, secs):
#         print('do the setup: secs = {}'.format(secs))
#         self.secs = secs
#
    def handle(self):
        print('Count: {}'.format(self.count))
        self.count -= 1
        if self.count < 1: return False

        self.__super__.waitSensorActive([self.sensor])      # Sensor defaults to Active
        print('Running = {}'.format(self.sensor.getKnownState()))
        self.__super__.waitMsec(self.secs * 1000)
        return True

# end of class definition
timer = 5
# a = AutomatExample(5)
a = AutomatExample()
# a.setup(5)
a.setName("Automaton test script")
a.start()
```
----
I use a setup function to pass parameters to an AbstractAutomaton instance.  That fails with:

```
     [java] 18:36:16,626 startup.ScriptButtonModel             ERROR - Unable to run script /Users/das/JMRI/_Profiles/Python3.jmri/AutomatonTest.py3. [AWT-EventQueue-0]
     [java] javax.script.ScriptException: org.graalvm.polyglot.PolyglotException: AttributeError: foreign object has no attribute 'setup'
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.toScriptException(GraalJSScriptEngine.java:512)
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.eval(GraalJSScriptEngine.java:489)
```

It appears that only the parent class methods are recognized.

----
The alternate method is to use the standard Python `__init__(self, secs)` function.  That fails with:
```
     [java] 18:27:44,394 startup.ScriptButtonModel             ERROR - Unable to run script /Users/das/JMRI/_Profiles/Python3.jmri/AutomatonTest.py3. [AWT-EventQueue-0]
     [java] javax.script.ScriptException: org.graalvm.polyglot.PolyglotException: TypeError: __init__() missing 1 required positional argument: 'secs'
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.toScriptException(GraalJSScriptEngine.java:512)
     [java] 	at jmri.script.jsr223graalpython.GraalJSScriptEngine.eval(GraalJSScriptEngine.java:489)
```

----
Since neither method works, I have to resort to a global variable.

### JMRI Lockup
I have added a waitSensorActive which appears to work until a wait has to occur.  In this case the initial sensor state is Active.  When it is set Inactive in the sensor table, the wait occurs.  When this happens, windows can be selected but none of them respond to inputs such as mouse clicks.

I have attached a thread dump.

[threaddump.txt](https://github.com/JMRI/JMRI/files/7851850/threaddump.txt)

bobjacobsen, 2022-01-12:
I think we're seeing what "The created object won’t behave like a Python object but like a foreign Java object. " on that page means.  Which is a mess.  

The thread dump indicates a deadlock between the AWT queue thread and the "Automaton test script" thread by them both holding locks on 0x00000007db8e6278 (a ReentrantLock) _Why_ that's the case isn't yet clear.

We might have to rethink inheriting Python classes from Java classes.  Aside from AbstractAutomat, are there other places where people do that (inherit, not use) Java classes typically?

dsand47, 2022-01-12:
> I think we're seeing what "The created object won’t behave like a Python object but like a foreign Java object. " on that page means. Which is a mess.
>
> The thread dump indicates a deadlock between [...]

I am not clear about inherit vs. use.

Here is a list of Jython classes that refer to another class using `class abc(xyz):`
```
(AbstractAutomaton)
(ActionListener)
(AdFrame)
(AdScrollFrame)
(ChangeListener)
(CreateTransits)
(DefaultWeightedEdge)
(FileFilter)
(JmriJFrame)
(Jynstrument)
(Jynstrument, AddressListener, MouseListener)
(Jynstrument, ItemListener)
(Jynstrument, MouseListener)
(Jynstrument, PropertyChangeListener, AddressListener)
(Jynstrument, PropertyChangeListener, AddressListener, MouseListener)
(Jynstrument, PropertyChangeListener, AddressListener, WiiDeviceDiscoveryListener, WiiRemoteListener, Runnable)
(Jynstrument, PropertyChangeListener, AddressListener, jmri.ThrottleListener)
(MoveTrain)
(PropertyChangeListener)
(SerialPortDevice)
(Thread)
(WindowAdapter)
(XmlFile)
(java.awt.event.ActionListener)
(java.awt.event.ItemListener)
(java.awt.event.KeyAdapter)
(java.beans.PropertyChangeListener)
(javax.swing.JFrame)
(jmri.implementation.AbstractShutDownTask)
(jmri.implementation.AbstractTurnout)
(jmri.jmrit.automat.AbstractAutomaton)
(jmri.jmrit.automat.Siglet)
(jmri.jmrix.can.CanListener)
(jmri.jmrix.dccpp.DCCppListener)
(jmri.jmrix.grapevine.SerialListener)
(jmri.jmrix.lenz.XNetListener)
(jmri.jmrix.loconet.LocoNetListener)
(jmri.jmrix.loconet.SlotListener)
(jmri.jmrix.mqtt.MqttContentParser)
(jmri.jmrix.loconet.locostats.LocoNetInterfaceStatsListener)
(jmri.jmrix.mqtt.MqttEventListener)
(jmri.jmrix.pricom.pockettester.DataListener)
(jmri.jmrix.pricom.pockettester.DataSource)
(object)
(threading.Thread)
(xAPlib.xAPRxEventListener)
```

bobjacobsen, 2022-01-14:
The threading problem is a known issue/feature in the current GraalVM when we use it via a ScriptEngineFactory wrapper. (The approach I used).  It might be improved/fixed eventually, might not.  I think we should wait a bit and see.  But that means this is not 100% useful for JMRI scripting right now.

There's also an effort to create an installable Nashorn JavaScript interpreter that doesn't require GraalVM.  If that works, then we don't really need GraalVM on Java 15 or later to have JavaScript.  So that's another wait and see.

Finally, Native Image support for Windows is expected this year.  We'll see.  Right now, native image provides a faster/simpler startup on Linux, but only Linux (there's a partial solution on macOS, but it's only partial). 

Together, I think this means a lot of wait-and-see.  I'll add a note to that effect to the next release note and the PY3 page.

danielb987, 2022-02-28:
I think this Issue should stay open for now.

danielb987, 2022-06-05:
I assume that this Issue should stay open.

danielb987, 2022-09-04:
I suppose this issue should stay open for now.

bobjacobsen, 2022-10-19:
I'm going to close this for now.  We're a long way from moving to Java 17, when it will be needed.

closed 2022-10-19 by bobjacobsen