JMRI/JMRI #10589: CMRI Network Metrics window wrong size

KenC57, 2022-01-16:
It initially opens the Data Metrics and action buttons half not fitting in the window. Closing and reopening the window shows it isn't retaining the 'last window size' either.

jerryg2003, 2022-01-21:
See also Issue 6919 and 9542.

bobjacobsen, 2022-05-13:
With 4.99.7 on Mac, it opens fine for me. It also remembers it's position across open/close and quit/restart.

Could somebody please post a screenshot of how it looks when you:
 - Open the window
 - Set it to the size you want
 - Close the window
 - Reopen it?

Thanks

jerryg2003, 2022-06-24:
First picture: when CMRI Network Metrics window first opened.
Second: as resized by me.  Then shut JMRI (also did a "Save" under Preferences in case that made a difference - it didn't).
Third: JMRI reopened, Network Metrics window reopened as in first picture.

[image]

[image]

[image]

KenC57, 2022-06-24:
JerryG,  that seems to confirm the problem still exists in 5.0. I also tried and it seems to still fail to
save the size correctly or restore it correctly, at least in Windows. It fails if I close and reopen the tool
or if I exit and reboot.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

KenC57, 2022-06-24:
What I'm seeing now is while running JMRI, the user-interface.xml is updated for the CMRInetMetricsFrame details. If changed, it shows the new location and size. BUT when JMRI is restarted or the metric window is reopened, the user-interface.xml is written, and it reverts to the default from the code (width="845.0") which is not wide enough. I'm seeing height="410.0" and width="865.0" as a better set of sizes.

Seems two things. One the default is too small. Two is the launch of the window does not use the values from the user-interface.xml and the exiting value in it get over-written.

KenC57, 2022-08-23:
It still seems that this tool uses lots of fixed sizes and at least on Windows doesn't restore what the user-preferences might have saved. My fix just keeps with how the code was created otherwise and not look into why the last used size/location is ignored for Windows.

dsand47, 2022-08-23:
> It still seems that this tool uses lots of fixed sizes and at least on Windows doesn't restore what the user-preferences might have saved. My fix just keeps with how the code was created otherwise and not look into why the last used size/location is ignored for Windows.

Line 66 is overriding the JmriJFrame size.  Try commenting it out.

KenC57, 2022-08-25:
If I commented line 66, it would restore fine if there was valid data in the user-interface.xml but it gave a wrong layout for size if there was nothing in the user-interface.xml. If I could figure out where that initial value should be set, or if there was a restored value available, I could make it use that saved value instead.

dsand47, 2022-08-25:
I think the actual problem is how the window is built.

Here is a test that I put together.
```
diff --git a/java/src/jmri/jmrix/cmri/serial/cmrinetmanager/CMRInetMetricsFrame.java b/java/src/jmri/jmrix/cmri/serial/cmrinetmanager/CMRInetMetricsFrame.java
index 196b44d117..d2641f75cf 100755
--- a/java/src/jmri/jmrix/cmri/serial/cmrinetmanager/CMRInetMetricsFrame.java
+++ b/java/src/jmri/jmrix/cmri/serial/cmrinetmanager/CMRInetMetricsFrame.java
@@ -63,7 +63,6 @@ public class CMRInetMetricsFrame extends jmri.util.JmriJFrame {
         // For the class
         setTitle(Bundle.getMessage("MetricsWindowTitle") + Bundle.getMessage("WindowConnectionMemo") + _memo.getUserName()); // NOI18N
         setLayout(new FlowLayout(FlowLayout.LEFT));
-        setPreferredSize(new Dimension(845, 400)); // 415 375
         networkMetricsBorderTitled = BorderFactory.createTitledBorder(networkMetricsBorder, "Error Metrics", TitledBorder.LEFT, TitledBorder.ABOVE_TOP);
 
         // Set up the CMRInet ERROR metrics table
@@ -135,7 +134,6 @@ public class CMRInetMetricsFrame extends jmri.util.JmriJFrame {
         networkMetricsPanel.setPreferredSize(new Dimension(415, 300));  //425 300
         networkMetricsPanel.setVisible(true);
 
-        add(networkMetricsPanel);
 
         //--------------------------------------
         // Set up the CMRInet metrics DATA table
@@ -200,7 +198,6 @@ public class CMRInetMetricsFrame extends jmri.util.JmriJFrame {
         networkMetricsDataPanel.setPreferredSize(new Dimension(415, 300));  //425 300
         networkMetricsDataPanel.setVisible(true);
 
-        add(networkMetricsDataPanel);
 
         // Main button panel
         //------------------
@@ -239,7 +236,13 @@ public class CMRInetMetricsFrame extends jmri.util.JmriJFrame {
         });
         mainButtons.add(doneButton);
         mainButtons.setVisible(true);
-        add(mainButtons);
+
+        var wrapper = new JPanel();
+        wrapper.setLayout(new BorderLayout());
+        wrapper.add(networkMetricsPanel, BorderLayout.WEST);
+        wrapper.add(networkMetricsDataPanel, BorderLayout.EAST);
+        wrapper.add(mainButtons, BorderLayout.SOUTH);
+        add(wrapper);
 
         addHelpMenu("package.jmri.jmrix.cmri.serial.cmrinetmanager.CMRInetMetricsFrame", true);
```

[image]

closed 2022-08-25 by danielb987

KenC57, 2022-10-11:
Commenting it out does help IF the user-preference has reasonable sizes. But if
I delete the user-preferences, it comes up in a very odd size. Height is just a
bit short but the width is much wider as it puts the buttons to the right of the
other two panes.

Where should the size be set for that first time if there was nothing in the
user-preferences?