JMRI/JMRI #10590: Create Test Release 4.99.3

bobjacobsen, 2022-01-16:
This is the next release in the 5.0 cycle. It's intended to be created from the `HEAD` of the `j11master` branch.

bobjacobsen, 2022-02-11:
I'm going to start the release process tomorrow (Feb 11) around 2PM US Eastern time to let me get it done before too late that night.  I'll post another message when I start.  As a reminder, I'll be

 - dropping in a tag on master for last content from Java 8
 - merging most-recent master to j11 master
 - doing the release process to create 4.99.3 files for checking
 - doing the release process to create 4.27.1 files for checking
 
Then once those files are OK
 - merging j11master back to master
 - removing the j11master branch

it's important that no PRs get merged between the time this process starts and the merge is complete, so I'll be merging (even if early) all PRs with approval and CI checks, and marking all the others as WIP until the process is complete.

bobjacobsen, 2022-02-12:
Starting release builds.  Please don't merge PRs.

bobjacobsen, 2022-02-12:
Headlesstest:
[image]

bobjacobsen, 2022-02-12:
Headless test _not_ including run-separate tests:
[image]

dsand47, 2022-02-12:
There is a serious bug in the latest LogixNG change.

dsand47, 2022-02-12:
> There is a serious bug in the latest LogixNG change.

Issue #10689 created.

danielb987, 2022-02-12:
> Starting release builds. Please don't merge PRs.

I don't follow issues unless I subscribe to them and it might be true to other maintainers as well. So it might be good to post important messages like this on the JMRI developers list. Thanks.

bobjacobsen, 2022-02-13:
JMRI 4.99.3 has been released. Files are available in the GitHub release section.

closed 2022-02-13 by bobjacobsen