JMRI/JMRI #10617: Table window vertical resizing tables opened via StartUp action anomaly

jerryg2003, 2022-01-27:
Using 4.99.2 on Windows 10:  three different behaviors when resizing table windows that have been opened by a StartUp action (at least that's how many I found so far, only one of which is "correct" IMHO).  OTOH, all work "correctly" when started via the Tools->Table-> commands on the PanelPro main window!

1, When grabbing and lengthening the Sensor Table window, it will only display up to 17 sensors in the top frame.  Further lengthening only increases the size of the lower frame (where "Add" button is), which has no purpose.  Also true for Turnout Table window,

2, When grabbing and lengthening the Memory Table window, it displays an increasing number of lines in the top frame, but after displaying about 17 lines in the top frame, it starts lengthening the bottom frame as well (where the "Add" button is), which has no purpose.  Also true for Light Table and Signal Mast Table windows.  

3.  [Correct behavior:] When grabbing and lengthening the Signal Mast Table window, it displays an increasing number of lines in the top frame and does not change the size of the bottom frame.

Are these behavior variants somehow related to the fact that windows opened via StartUp actions do not have a left side frame showing the list of all tables?

jerryg2003, 2022-10-16:
I would like to reopen this to see if someone with the coding knowledge is as curious as I as to why a window opened via a StartUp action behaves differently (incorrectly) than one opened via the Tools/Tables menu. [I can sort of understand some Table windows behaving differently than other Table windows (see 1 and 2 in the original issue post) - different programmers - although that is not great either.]

Still there in JMRI 5.1.5 under Windows 10 with Java  11.0.16.1.

Thanks.

dsand47, 2022-10-16:
The difference is caused by the context.  The user-interface.xml file has an entry based on the Tools &rArr; Tables menu action and a different one when using a startup action to open the table.

```
        <window xmlns="" class="jmri.jmrit.beantable.ListedTableFrame:Sensors" height="158.0" locX="0.0" locY="1.0" width="437.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1" height="139.0" locX="928.0" locY="0.0" width="512.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.ListedTableFrame:Memory Variables" height="162.0" locX="974.0" locY="738.0" width="466.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1:1" height="145.0" locX="0.0" locY="755.0" width="341.0"/>
```
The second one is the sensor table startup action.

The 3rd and 4th items are memory variable windows.

The ListedTableFrame entries are the location and size of the entire window.  If each table is open using the full menu path, the table specific frame size is used.  If the table is selected from the left column list, it uses the current frame size.

The startup actions use an index number based on the startup actions row for location and size instead of the table name.

jerryg2003, 2022-10-16:
Thanks.  So if I understand this correctly, the StartUp action uses a different class for displaying the table window, and in the case of certain table types, those classes don't work as they should when a further manual resizing is done (that is, limiting the listed items to 17, and then only continuing to lengthen the lower frame).

dsand47, 2022-10-16:
On my computer the magic number is 14.

When the BeanTableFrame is built by AbstractTableAction (startup open table),  the height of the table data area is set to the height of the rows or 400, whichever is greater.

When the window area exceeds the data area by making the window taller, both the data area and the footer expand.

Since the data area height is set when the window is created, adding additional rows does not change the data area height.  Making the window taller to see the bottom rows will result in the expansion.

After doing a store and re-starting, the new data area height will include the new items.

Scrolling is not affected by this behavior.

KenC57, 2022-10-16:
If the start up action is treated special, how are you supposed to adjust it?

Is the startup action the same as if it was a button created by start up? This
latter is the one I use most often and frequently end up with some odd places or
sizes for the table and an unclear way to adjust that.

My questions related to the 'open tables' and not a specific single table.

dsand47, 2022-10-17:
Opening a table from a button uses the same process as using the start up action to open a table.

Once the table has been opened using either startup method, the location and size is recorded in user-interface.xml.  Any subsequent session will used the last known values.

The tricky part is that the **AbstractTableAction** values are recorded by index based on the location in the the startup actions list.  If the order is changed, the apparent location and size will change.  If you delete the last button and add a button for a different table, the new table will get the location and size for the previous table.

The other thing to remember is that startup actions and buttons are totally independent.  The **Tools &rArr; Tables** values are separate.

KenC57, 2022-10-17:
Will moving or resizing the table, after creation by a startup action, update
the startup action entry or only update the normal table entry? End effect, can
a user change where that table via the startup action appears? I've not found
normal ways to do that change (other than edit the preference file manually).

dsand47, 2022-10-17:
**Tools &rArr;  Tables** changes are saved in the ListedTableFrame entry.
```
        <window xmlns="" class="jmri.jmrit.beantable.ListedTableFrame:Sensors" height="424.0" locX="0.0" locY="1.0" width="692.0"/>
```

Startup action and button table changes are saved in the AbstractTableAction entry.
```
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1" height="709.0" locX="608.0" locY="0.0" width="512.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1:1" height="204.0" locX="202.0" locY="19.0" width="124.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1:2" height="302.0" locX="45.0" locY="242.0" width="454.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1:3" height="162.0" locX="0.0" locY="657.0" width="332.0"/>
        <window xmlns="" class="jmri.jmrit.beantable.AbstractTableAction$1:4" height="492.0" locX="0.0" locY="1.0" width="819.0"/>
```

The first 3 are startup actions.  Assuming no changes to the startup action list, they always end up in the same place.

However, the 4th and 5th are buttons.  The first button clicked will get the 1:3 entry and the second one will get the 1:4 entry.  This means the two tables can swap places unless you always click the buttons in the same sequence.  For fun, close the button based windows.  Now click left to right.  Close them again and click right to left.

jerryg2003, 2022-10-17:
I’m learning a lot about the inner workings of JMRI from this stream, but does any of it explain the inability to display more than 14 or 17 items in a table window initially opened via StartUp action? No matter what I try, only the lower frame keeps expanding after that (at least for Sensor and Turnout table windows)?

closed 2022-12-18 by stale[bot]