JMRI/JMRI #10618: LogixNG printout formatting using

jerryg2003, 2022-01-27:


danielb987, 2022-02-08:
@jerryg2003 
Thanks for point out this! I have thought about trying to fix it, but I have too much other things to do.
It would be very good if you have time to fix it.

jerryg2003, 2022-02-08:
…will give me an opportunity to learn about how to do that…

jerryg2003, 2022-02-18:
@danielb987 et al:  see PR #10724 .  Need people to try this out with large, complex panel files, especially complex LogixNG.  Copy panelfile-4-99-3.xsl to your computer.  Replace href target in line 2 of your panel files.  Let me know what you think of the output (remember that I mostly worked on Logix and LogixNG, but I willing to investigate and modify other areas.   Thanks.

danielb987, 2022-02-18:
@jerryg2003 
Thank you!

@dsand47 @bobjacobsen 
Do you have time to look at this? I have no experience with `xsl` files and almost never look at the xml files in a web browser.

bobjacobsen, 2022-02-19:
@jerryg2003  Could you try [the instructions for viewing files with the web server](https://www.jmri.org/help/en/html/doc/Technical/XmlView.shtml) to see if they're still correct?

jerryg2003, 2022-02-19:
@bobjacobsen I had read that so it is on my list to work on (going away for a week so it may be a while). Also, please see my question two comments above.  Thanks.

bobjacobsen, 2022-02-19:
Answer to question:  Some browsers don't find the .xsl file, some do.  The solution when they don't is to use the local web server.  That's what I use.

danielb987, 2022-02-19:
> @jerryg2003 Could you try [the instructions for viewing files with the web server](https://www.jmri.org/help/en/html/doc/Technical/XmlView.shtml) to see if they're still correct?

@bobjacobsen 
Not related to this PR. But the `sample panel XML file` in the second paragraph of the page is not found. It seems removed from the xml/samples folder of JMRI/JMRI. But `xml/samples/CrossoverExample.xml` works fine.

jerryg2003, 2022-02-19:
@danielb987 @bobjacobsen  I will:
1. Change the PR to put my new XSL code into panelfile-4-19-2.xsl rather than creating a new panelfile-4-99-3.xsl (please answer Q1 first so I can do all that is necessary at the same time). 
2. I will rewrite the instructions for viewing XML files and fix the broken link (in a week or so).

Q1: The files in the /xml/samples directory point to panelfile.xsl (no version number) so should I copy my code into that file too in my PR?  (When were panel files changed to start referencing an XSL panelfile-[version number].xsl? Are there others out there that should be changed?)
Q2: The index.shml file in the /xml/xsl directory is specific to decoders while that directory itself contains a variety of xsl files.  Is that what we want, or should that be changed too? 

Thanks.

danielb987, 2022-02-19:
> Q1: The files in the /xml/samples directory point to panelfile.xsl (no version number) so should I copy my code into that file too in my PR? (When were panel files changed to start referencing an XSL panelfile-[version number].xsl? Are there others out there that should be changed?)

I will let @bobjacobsen answer that since I don't know the answer myself.

bobjacobsen, 2022-02-19:
> Q1: The files in the /xml/samples directory point to panelfile.xsl (no version number) so should I copy my code into that file too in my PR? (When were panel files changed to start referencing an XSL panelfile-[version number].xsl? Are there others out there that should be changed?)

I think it make sense to have the panel files in /xml/samples reference the most recent .xsl and .xsd files.  Don’t change any of the older (lower version) .xsd or .xsl files.

See https://www.jmri.org/help/en/html/doc/Technical/XmlSchema.shtml for how we update the numbers.

> Q2: The index.shml file in the /xml/xsl directory is specific to decoders while that directory itself contains a variety of xsl files. Is that what we want, or should that be changed too?

That should not be changed.  Decoder files are common, and we run the XSLT transforms on them onto the web.  That index is for them.

See https://jmri.org/xml/XSLT

jerryg2003, 2022-02-20:
@bobjacobsen  

Hmmm... I tried changing one (CrossoverExample.xml) to panelfile-4-19-2.xsl and it doesn't work - different schema.  Here is what it has for a turnout:

` <turnout systemName="LT17" feedback="MONITORING" />
   `

Here is what a current panel of mine has for a turnout:

`<turnout feedback="ONESENSOR" sensor1="Test 01" inverted="false" automate="Off">
      <systemName>CT3001</systemName>
      <userName>Test TO</userName>
    </turnout>`

So it appears that these examples are from an earlier schema.  Even panelfile-2-9-6.xsl has the newer one.  So I guess I just leave these alone.

jerryg2003, 2022-02-20:
Also, found some other errors and omissions:  how do I find out what "type" 29 in a logix conditional schema is, and what the data values mean?  One of my panel files (which plays a sound) prints out "Unknown type. Report this as an error."

bobjacobsen, 2022-02-20:
> So it appears that these examples are from an earlier schema. Even panelfile-2-9-6.xsl has the newer one. So I guess I just leave these alone.

Leaving them alone is fine!  Otherwise, you could read them on e.g. 4.20 and rewrite them to get the content updated.

> Also, found some other errors and omissions: how do I find out what "type" 29 in a logix conditional schema is, and what the data values mean? One of my panel files (which plays a sound) prints out "Unknown type. Report this as an error.”

Not elegant, but I searched the logix code for 29 and found:

```
java/src/jmri/Conditional.java:    static final int TYPE_TRAIN_RUNNING = 29;
java/src/jmri/Conditional.java:    static final int ACTION_CONTROL_AUDIO = 29;
```

Which lead to the complete lists of Type (things you can check; line 685) and Action (things you can do; line 747) codes.

jerryg2003, 2022-02-25:
PR #10724 provides formatted LogixNG and many other improvements.  Further work will be done as new PRs.

closed 2022-02-25 by jerryg2003