JMRI/JMRI #10619: Single CV Programmer Long Address Limit incorrect for Loconet Attached Board Mode

n3ix, 2022-01-28:
The Single CV Programmer will not allow a Long Address greater than 10239 in Loconet Attached Board mode.  I believe that the maximum allowed address should be 16383 in this mode.

crosland, 2022-01-28:
If its a DCC address it cannot be greater than 10239 due to the way the address space is partitioned. It's not a true 14-bit space. This is the same for all systems.

If it's a Loconet address (i.e. not for a loco) then I don't know, but why would you be using the CV programmer in that case.

n3ix, 2022-01-28:
@crosland Hi.  As indicated it is a Loconet Attached Board and addresses go to 16383.  We routinely use high addresses for these boards so that CVs are not accidently changed by throttles.  In this case I needed to test the effect of Reading and Writing a CV that is not supported in the Decoder template.  The Single CV Programmer was the easiest solution, but the address range limits are incorrect in this case.

bobjacobsen, 2022-01-28:
I’ll fix this.

Does a “LocoNet Attached Board” have separate long and short addresses?  Or is it just one address space, i.e. like LNCV2 addressing?

n3ix, 2022-01-28:
@bobjacobsen Thanks Bob.  In my case we just use Long addresses.  I am not aware of anyone making a board that uses Short addresses.  I suppose that is possible since it is just an Ops Mode message.  The Single CV Programmer has another issue in that you can get in a mode where just the Short Address radio button is shown, and if you select it there is no way to deselect it.
[image]

n3ix, 2022-01-28:
I can create the radio button issue by selecting LNCV and then selecting another mode.  Long Address button is gone.

bobjacobsen, 2022-01-28:
See #10620. If you could test the bobjacoben/simpleProg branch, that would be great.

n3ix, 2022-01-29:
@bobjacobsen  Well it took all day to get this branch running on my jmri build machine (I had not built for a long time).  I have very low confidence I am running the right code.  It reports 4.25.10ish on start?  If that is actually the branch build then I cannot enter a long address above 10239 with LocoNet-attached Board selected.

bobjacobsen, 2022-01-29:
Sounds like you got a different branch.  Sorry for time taken.

Is this what you're looking for?

[image]

n3ix, 2022-01-29:
Yes, looks good.   

My Eclipse Ant build environment is probably messed up somehow.  Git says my fork of JMRI master is up to date and GitHub desktop says my local copy is up to date but when I build that I also get 4.25.10ish which makes no sense?  It's hard to try to do a build once every 6 months and have a clue what I'm doing :(

closed 2022-02-04 by n3ix