JMRI/JMRI #10621: Switchboard Panel and Webserver display of  the panel are not the same.

Phil-Klein, 2022-01-28:
[image]

I opened switch board editor, change from loconet to internal turnouts, reduced the number of buttons displayed to 2, set one turnout to open and the other to closed.  The screen shot shows the difference between what the actual panel displays and what is displayed via JMRI webserver.

mstevetodd, 2022-01-28:
There are actually quite a few differences between the web view and the JMRI panel view. Enough that I suspect some or all are design choices and intentional. @silverailscolo did this development. Hopefully, he will comment on this?

Phil-Klein, 2022-01-28:
[dark territory.txt](https://github.com/JMRI/JMRI/files/7962332/dark.territory.txt)

Phil-Klein, 2022-01-28:
Steve

Its uploaded.

Phil

silverailscolo, 2022-01-29:
Phil, the issue you show is caused by two separate choices I made (they could be adjusted).
The web view of a Switchboard fills the full area, resulting in the rectangular shapes. In JMRI the stay in the frame.
But I think what you mean to point out in this issue is the different coloring. My visual preference was to provide a boldly colored rectangle, like the web display. The button display in JMRI was created as the most basic graphic, a java JButton. However I never managed to set the fill color of these. IIRC all that I could set was either the border or the text color. Standing further away from the screen, I agree this is not distinct enough.
If someone can help us find the command for that, I would be happy to implement it, making both views more alike.
Egbert

bobjacobsen, 2022-01-29:
If you can point me to the right place in the code, I can take a look.

mstevetodd, 2022-01-29:
> I can take a look.

Thanks, @bobjacobsen but I know where to make the change, just wanted to be sure @silverailscolo was OK with changing it. I'm testing it now.

mstevetodd, 2022-01-29:
Example of changes: http://jmri.mstevetodd.com:12080/panel/Switchboard/Switchboard

@Phil-Klein This change will take a while to show up in production. If you'd like to try it now, copy the following files to the respective folders in your JMRI, stepping on the files by the same names:
http://jmri.mstevetodd.com:12080/dist/web/css/panel.css
http://jmri.mstevetodd.com:12080/dist/web/js/panel.js
Be sure to refresh your browser to get the new files.

Phil-Klein, 2022-01-30:
Steve

I renamed the original files, copied the new files.

I get the same result.  

Phil

mstevetodd, 2022-01-30:
> Steve
> 
> I renamed the original files, copied the new files.
> 
> I get the same result.  
> 
> Phil

Sounds like your cache is not cleared. Restart JMRI and forcibly clear your browser's cache. Use a regular browser, not an embedded browser in a phone app.

Phil-Klein, 2022-01-30:
Steve

Clearing the cache resolved the issue.

Thanks

Phil

closed 2022-01-30 by Phil-Klein