JMRI/JMRI #10622: Switchboard Editor feature request.

Phil-Klein, 2022-01-28:
When utilizing User names, I would like to be able to turn off displaying system names.  The buttons become too cluttered with both.  However, I can see that other people may want to display both.

silverailscolo, 2022-02-13:
@Phil-Klein could you check out the latest release and confirm the new Switchboard > Options menu > Labels choice works for you?

Phil-Klein, 2022-02-16:
Egbert.

 

It works great!  Thanks

 

Phil

 

<https://github.com/JMRI/JMRI/issues/10622#issuecomment-1038350572> , or
unsubscribe
7527ANCNFSM5NBUE57A> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AXQLSHVFQRF4ZAEVNZMAFNTU27527A5CNFS
M5NBUE57KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHXR7R3A

closed 2022-02-16 by silverailscolo