JMRI/JMRI #10623: Request: Export  updated CV values to text (or .CSV) file

McLae, 2022-01-28:
The use cases:
1. no hardware connection to use Decoder Pro,. User uses Simulator mode to make changes, then exports changed CV values to enter manually.
2. The user inspect the list of changes CVs to verify no unwanted changes were made.
3.  The user exports the list of changes and posts to a friend the list of changes.
4.  The user exports the list of changes and uses as reference when programming fleet of Locos.
5.  If export all CV values is an option, user exports list of CVs and posts list to help group to validate settings. 

Implementation:
Menu option to export CV values (may be two options, export all or export changed)
Change list would pull from CV pane all items with status of changed, not written.
All list would pull list of CVs from CV pane regardless of status. 
Might  have use for option to pull CVs by each of three status, not read (yellow), changed (orange), or read (White).

bobjacobsen, 2022-02-18:
You can do this already in DecoderPro.

Open an roster entry. On the "Roster Entry" pane, there's a File menu.  Select Export..., then the file type you want.

dheap, 2022-02-18:
@bobjacobsen What @McLae is asking for is an option to export only those CVs that have a  **Changed** status, not the whole list as a present.

ESU's LokProgrammer software has such a feature and @McLae would like the same feature in JMRI.

bobjacobsen, 2022-02-18:
Ah, thanks.  My mistake.

dheap, 2022-02-18:
Refer this  [jmriusers](https://groups.io/g/jmriusers/topic/88481388#199412) thread

bobjacobsen, 2022-02-20:
DecoderPro CV's have multiple states possible:
```
UNKNOWN   (red) - usually generated by a read or write error
EDITED    (orange) - changed through the GUI
READ      (default color) - read from decoder
STORED    (default color) - written to decoder
FROMFILE  (yellow) - read from roster entry
```

Which one(s) do you want to write out?  Just EDITED, or some others too?

bobjacobsen, 2022-02-20:
See PR #10728. That only writes the EDITED state, but it would be easy to modify. Also only writes CVS format for modified-only, not the PR1 or PR1WIN formats.

[image]

dheap, 2022-02-20:
My view is that EDITED is the main state for which an exported CV value is meaningful/useful.

McLae, 2022-02-20:
EDITED is the main use case. This will generate a list of CVs a NOOB may use to program using a throttle. Might convince these folks to start using JMRI to determine what CVs to change (nose of the camel).
Written might be a secondary use, to record what was changed in a specific session. (Future feature?)30 years of programming, planning before hand can make feature expansion easy or hard! :)
Thomas Stephens DeSoto, TX

bobjacobsen, 2022-02-20:
#10623 sets it up so that it's easy to add move menu items that write different (sets of) states.  It's probably more work to find good (short, clear) labels for them and keep the menu clear & usable than it is to technically add more items. So we can address additional use cases as needed....

closed 2022-02-21 by danielb987