JMRI/JMRI #10633: Zimo CVs Missing

PhillTrain, 2022-01-31:
Issue Type: __Feature Request__

Good morning,
I use mainly Zimo decoders and find your JMRI program very useful, thank you for making it available.
I only have one problem, it doesn't list all the Zimo CV's, I use serveral not listed and as they are not there I can't save them.
Don't know if other decoder makes also are missing CV's.
is there an update for this?
If not would it be possible to update to allow me to veiw all Zimo CV's.
keep up the good work.
best regards,
phill.

DecoderPro 4.26+R381c8dfc32
Java HotSpot(TM) Client VM Oracle Corporation 25.311-b11 mixed mode
Windows 10 10.0 x86

crosland, 2022-01-31:
It might help if you can narrow the problem down by giving specific 
examples of decoder model and CV(s). The decoder firmware version would 
also be useful if that is readable.

It may be that Zimo have updated their firmware since the decoder 
definitions were last updated.

PhillTrain, 2022-02-13:
> CV No.
> 109 to 111. Seems to be present.
> 170 to 177. 183 to 189. Related to Servo's 3 and 4. Presumably nobody got round to adding them when Zimo added them. Hopefully not too difficult to add, as may be duplicates of Servo 1 and Servo 2.
>
>   1.  Sound related, a new feature Zimo added and nobody spotted. I'm making some updates so that may be caught.
>   2.
>      *
>
> Present,  though in process of being moved to menus at top of pane (JMRI 4.99.3).   They don't really fit as CV's for decoderPro as they are not values to "record", but actions which put the decoder into a different state (pseudoprogramming or load calibration runs).
>
> 339 to 340. 342. 343. 379. Sound related, diesel/electric locos. A new feature Zimo added and nobody spotted.
> 593 to 598. Sounds for panto movement. New features Zimo added and nobody spotted
> 700 to 737. 740 to 743. Seem to be sound related, but not clear from a quick read of the German documentation where they'd go. Requires more investigation.
>
> Some of these might get added when a volunteer gets round to working out what they are, and where they go. I'm making some changes at present, but I'm not clear on what all of them do, or where they should appear.

Hello Nigel,
Many thanks for reply.

Best regards,
Phill.

nigelcliffe, 2022-02-13:
Further update.    

Summary -  some of this may be "large decoder" elements, which needs someone else to update it.  
Where I've added things to my files, its to the "small decoder" files, which typically cover decoders MX600-MX660, aiming to have these files in the next test release (4.99.4).

170 to 177. 183 to 189.   Appear to be for Servo's 3 and 4.  Which would be large-scale decoders only ?   Zimo manual for small decoders only have two Servo outputs, so not appropriate for those.    
CV270 is listed in German manual for small decoders (usually most up to date) as "not yet implemented".     

CV339/340 - added to my file.  
CV342/343 are present in decoder file for version 40 decoders, so is this a different decoder which support it, but not showing ? 
CV379, CV593-598  -  added to my file. 

CV700 to 737, and 740 to 743 - I've not tried to understand these yet.

closed 2022-04-16 by stale[bot]