JMRI/JMRI #10638: LogixNG - Jython Functions

dsand47, 2022-02-01:
The process for creating Jython functions to extend LogixNG functions has changed since the initial release.  This will require updating chapter 9 of the LogixNG reference.  The changes have been identified and appear to work.

There are exceptions that occur if the Formula Functions help is displayed when there is a Jython function.

The script has extra print statements to show the changes to the calculate function parameters.

[Archive.zip](https://github.com/JMRI/JMRI/files/7981163/Archive.zip)

dsand47, 2022-02-01:
The Jython class needs another function:
```
  def getConstantDescriptions(self):
    return 'None'
```

danielb987, 2022-02-03:
> There are exceptions that occur if the Formula Functions help is displayed when there is a Jython function.

Is this resolved by the updated documentation in  #10640?

dsand47, 2022-02-03:
> > There are exceptions that occur if the Formula Functions help is displayed when there is a Jython function.
> 
> Is this resolved by the updated documentation in #10640?

Yes.  A required function, `getConstantDescriptions()`, had been added which is what caused the exceptions.

closed 2022-02-03 by danielb987