JMRI/JMRI #10669: CBUS Node Manager GUI Enters Endless Loop

crosland, 2022-02-09:
I have discovered an issue in 4.26 :( Luckily I suspect no one has used this part of JMRI yet (node manager gui for servo module).

I need some help in suggesting how to handle the underlying cause.

We have a table data model for the CBUS Node variables with a basic table display. No problems there. I added a new pane that displays the data more graphically with sliders and spinners, etc. This implements tableChanged() to capture changes to the table data, e.g., in Servo8BaseEditNVPane,java

When a user interacts with this gui I want the changes to be applied to the table data model, ultimately through fireTableRowsUpdated().

When the table data changes for any other reason I want the changes to be reflected in the new gui, through the tableChanged() method.

Add to this the outgoing CBUS NVSet message which results in a property change being fired to update the table (which gets propagated to the guiand things start to get messy.

There's a particular problem with JSpinners. Changing the data programmatically in response to the table changing causes the spinner's change listener to fire, the same as if the user had interacted with the gui. The change listener causes a new update to the table, which causes the spinner to be updated, and so on, and the code gets stuck in an endless loop.

Add to this the outgoing CBUS NVSet message which results in a property change being fired to update the table (which gets propagated to the gui, which changes the Spinner, which fires a new table update, ...) and things start to get messy.

I also saw problems with JSliders but the issue does not appear in the current code.

Following a suggestion found on StackExchange I tried using JSpinner.isFocusOwned() to determine when the change was (likely) due to user interaction, but this never seems to return true.

I am guessing the underlying cause is at least partly due to having two data models, the overall one in the table and the individual ones in the Spinner components.

Another suggestion is to use flags to disable the change listeners but I worry things may get out of sync.

I will try adding code to only propagate changes from the table and in the Spinner change listener when the data really has changed, e.g., the spinner component causes a table data change and so does the CBUS message. The second of these will see the data has already changed.

I also thought about custom renderers for the table data but I think any use of JSpinner will result in similar behaviour.

Any other suggestions welcome.

@icklesteve can you think of anything?

danielb987, 2022-02-09:
@crosland 
The first thing I recommend you to do is to update the release notes for 4.26, explaining this issue. That way, the risk is lower that users will be trapped by it. And if they are, there is a possibility that they will read the release note and see the problem.

https://github.com/JMRI/website/blob/master/releasenotes/jmri4.26.shtml

Note that the release note is in the JMRI/website repository.

crosland, 2022-02-09:
Good point! Done! PR #458

danielb987, 2022-02-09:
> Good point! Done! PR #458

Thanks! PR is merged.

danielb987, 2022-02-09:
Is it possible to test this issue with a simulator? I don't have any CBUS hardware.

crosland, 2022-02-09:
Not easily, we would need to add code to simulate the module as well, or populate the Node Manager Table in some way.

bobjacobsen, 2022-02-09:
For the JSpinner, I'd create a new subclass that overrides `setValue` to suppress any action if the value is actually the same as the current one.

icklesteve, 2022-02-09:
@crosland I'm going to attempt to reproduce with a new simulated module.
Any idea how many Event Variables a CANMIO_SVO has?
Ta, Steve.

crosland, 2022-02-09:
Actually, the JSpinner seems to know that he value has not changed.

I am looking at what happens with the JSliders (I had a similar issue with these previously and can't recall what I did that fixed it), as these do not result in the endless loop. When a slider is changed and an NVSET sent to the layout, an internal CBUS reply is generated which causes all of the table to be updated, including the spinner values. This does not trigger the spinners to fire again, so I think they do look at the old and new values.

I need to trace what is different when the initial activity is a click on the spinner.

@icklesteve is there any reason why SINGLENVUPDATE property change has to trigger a complete table update?

This code from CbusNodeNVTableDataModel.java

```
    public void propertyChange(PropertyChangeEvent ev){
        if (ev.getPropertyName().equals("SINGLENVUPDATE")) {
            int newValue = (Integer) ev.getNewValue();
            resetNewNvs();
            fireTableRowsUpdated(newValue,newValue);
        }
        else if (ev.getPropertyName().equals("ALLNVUPDATE")) {
            resetNewNvs();
            fireTableDataChanged();
        }
    }
```
calls resetNewNVs() which iterates over the table firing a change for every row, even for a single update. This is n't the root of the problem but it would help to have less updates flying around.

crosland, 2022-02-09:
> Any idea how many Event Variables a CANMIO_SVO has?
If it's running the canservo8c firmware then it's 4 EVs per event
```
;this code assumes a one byte EV giving the servo numbers and a second EV for polarity. One bit per servo.
;EV1 = 	servo number.  Can have several servos per event.
;EV2 = 	polarity for that event
;EV3 = 	Used to teach the feedback responses. See below.
;EV4  	Available but mot used yet.
```

crosland, 2022-02-09:
What I learned today is that the spinner model is based on Number as it's value type. Even if it is constructed with Doubles you can assign an int and the value changes to Integer. This seems to plays havoc with the change events and I was seeing three calls to stateChanged() for a single click in the GUI.

The endless loop is fixed but I think there is still scope to tidy things up a bit, such as unnecessary whole table updates.

crosland, 2022-02-10:
OK, I've found some other similar problems, that don't lock up the code.

I am wondering if running Java 11 has changed the behaviour in some subtle way, or I'm just looking in more detail.

icklesteve, 2022-02-10:
I've been focused on other areas so not had a chance to add some other modules to the SPI.

Work in 1 hour, will revisit tomorrow.

Have found a Bundle tweak if you could include? see
https://github.com/icklesteve/JMRI/commit/5ebaf11cd8a7d2fb19c11711eb6d807af98a8a3c

crosland, 2022-02-14:
@icklesteve I have included the bundle change.

crosland, 2022-02-21:
PR #10703 fixes this

closed 2022-02-21 by crosland