JMRI/JMRI #10673: OperationsPro restore doesn't completely handle RFID tags

swalton00, 2022-02-09:
If you use OperationsPro tools->Restore to move the OperationsPro data to a different profile or a different machine, it successfully restores all the data with no error messages. However, the next time you open an OperationsPro panel, an Error level Exeception is thrown for every car that has an RFID tag. There is no apparent impact on any functionality, just a large number of Errors reported. An easy way to clean up the data is to do a Cars->Tools->Export of all the cars. Follow that with a Settings->Tools->Delete all Cars and end up with a Cars->Tools->Import of the Cars data just exported. 

The implication here is that Cars->Import does something that Settings->Tools->Restore should be doing.

danielb987, 2022-02-09:
@DanielBoudreau 
Is this something for you to look at?

DanielBoudreau, 2022-02-09:
I will take a look at this next week.

DanielBoudreau, 2022-02-09:
@swalton00 Can you send me your set of backup files? daboudreau@hotmail.com

Thanks,
Dan

swalton00, 2022-02-09:
I've looked at it some more and there is a mismatch between "OperationsPro" and "PanelPro" data. The missing tags don't show up in a PanelPro Table of tags.  The backup and restore operation considers only OperationsPro data, but the error is about missing PanelPro data. If you clear the OperationsPro data, it doesn't affect PanelPro data (and shouldn't in my view). Yet should OperationsPro be saving and restoring the directory of IDTags? I think some discussion is called for before I jump in and just register the tags. 

The answer is to call IdTagManager with a provideIdTag(newTag) for every car restored that has a tag. If there is already a tag for it, no problem. If there is no preexisting tag, that will create it. But when to call it. I'm reluctant to go through the data for each car after the restore but before the restart. The restore only moved in new XML into the OperationPro directory and there is no saying what state things are in until after the restart.

Yet the biggest question is where the boundary should be OperationsPro data and PanelPro data. Should RFID tags be entirely under control of OperationsPro? How does PanelPro use them? I see them appearing as Reporters, but what other functionality is there?

danielb987, 2022-02-09:
RFID tags might be used by scripts so they must not be removed from PanelPro.

@pabender 
Can you look at this issue?

swalton00, 2022-02-13:
After tracing through the code and debuging it, it seems like OperationsPro should just not use the PanelPro tags. Not including tests, there are only three places where IdManager is used to retrieve or save the tags. OperationsPro doesn't do anything with the User Name portion and the code where the RollingStock attempts to set the Location based on a Tag change will never fire becuase getTrackByReporter(Reporter r) will never fire because a TimeoutReporter will never be .equals() with an RfidReporter. Thus it will never set the Car location. 

I believe that removing the IdTagManager reference is appropriate. That will remove the error messages and totally seperate PanelPro tags from OperationsPro tags. Setting the Reporter to a specific track location may work for some, but I don't see how a point location would work for a Car. The RFID tag fires as the car passes over the reader, but that could be on the way in to the location, or on the way out from the location, or even just moving within the location. If the car is part of a train and is being moved to a location beyond the reader, using the reader to set the location could be either correct, or totally wrong (and thus really mess up what OperationsPro thinks about the location of the car).

The three references are 1) in the RollingStock class, where the setRfid() method sets the tag value of the car (this is the cause of the error messages, when the first list of cars is built). 2) ImportCars, where a new car is imported and there is an RFID tag present (this is why importing cars works, but restoring OperationsPro data does not). and 3) the RollingStockEditFrame, where the ComboBox is created (this is only time the User Name is referenced). This third reference is one where the IdTagManager reference needs to to remain, but it also needs to include IdTags that are on Cars, but not in the RFID Tag table. (Currently it is not possible to set a car to tag that is on a different car, but has not been read on this JMRI and thus is not in the table).

DanielBoudreau, 2022-02-13:
The car location in operations should not be set by detection.  The car’s location and track is where the database thinks the car should be. “Where Last Seen” is what gets updated by RFID detection.

swalton00, 2022-02-13:
I very strongly agree that the Location of a car should not be changed by detection. However, in the propertyChange event for "whereLastSeen", line 901 of RollingStock does setLocation(newTrack.getLocation(), newTrack) where newTrack is the track where the Reporter is located. This is not executed because getTrackByReporter does not correctly compare the Reporter with the RfidReporter which was assigned to a Location. Thus whereLastSeen never gets set to an OperationPro Location.

My whole point in this discussion is that OperationsPro tags should be totally seperate from PanelPro tags, other than choices for tags to be assigned to a Car.

pabender, 2022-02-14:
> I very strongly agree that the Location of a car should not be changed by detection. 

I have long disagreed with the assertion that the operations code shouldn't use the tag's location.  When the RFID tag's position is updated, you know (absolutely) where the car was at that point in time.  That may disagree with what operations knows about the car,  but it is highly valuable when trying to locate lost cars.   

The prototype uses RFID information (AEI tags) to determine where cars actually are, and that's what I want to model with respect to how the operations code behaves.  That isn't possible today because we aren't allowed to set the location without setting the track too.

> However, in the propertyChange event for "whereLastSeen", line 901 of RollingStock does setLocation(newTrack.getLocation(), newTrack) where newTrack is the track where the Reporter is located. This [...]

Could you elaborate please

https://github.com/JMRI/JMRI/blob/c9c2bff73f8fe2f5e70662d225721e664c64fb8c/java/src/jmri/jmrit/operations/locations/LocationManager.java#L128-L146

getTrackByReporter looks correct to me, with RfidReporter extending Reporter the comparison is correct.

pabender, 2022-02-14:
> After tracing through the code and debuging it, it seems like OperationsPro should just not use the PanelPro tags. Not including tests, there are only three places where IdManager is used to retrieve or save the tags. OperationsPro doesn't do anything with the User Name portion 

I don't believe this is actually correct.  The Operations code SHOULD (because this is the way everything SHOULD be working) be asking for the display name of the tag.  I haven't looked to see if it actually does or not.  

> and the code where the RollingStock attempts to set the Location based on a Tag change will never fire becuase getTrackByReporter(Reporter r) will never fire because a TimeoutReporter will never be .equals() with an RfidReporter. Thus it will never set the Car location.

Ahhh... so that is what you were refering to...

The managers creating the RfidReporter's are actually returning an RfidReporter decorated with a timeout reporter:

https://github.com/JMRI/JMRI/blob/c9c2bff73f8fe2f5e70662d225721e664c64fb8c/java/src/jmri/jmrix/rfid/generic/standalone/StandaloneReporterManager.java#L36-L48

So if the property change has reference to the raw reporter,  the equals on the TimeoutReporter needs to be updated so that it can compare the decorated reporter, not just the TimeoutReporter.  That certainly is a bug.

swalton00, 2022-02-14:
The propertyChange listener in RollinngStock.java, when a "whereLastSeen"
event occurs, does a locationManager.getTrackByReport( ) passing in the tag
that was detected, to determine if the Reporter has been set to a location.
The locationManager checks every known Track to see if the reporter matches
a Reporter on one of the known tracks. The problem is that the comparison
is done by (track.getReporter().equals(r) and the Reporters (and the code
hierarchy) is using the default .equals() method and, since they are of
slightly different classes, they are technically not equal. (The correct
approach would be to define a .equals(Object o) that, if both are an
instance of "Reporter", then compare the System Names and consider them
equal if those are equal. If the Reporter matches a Location, then it sets
the Location of the Car to the Location of the Reporter. Since they don't
match, the Location is not set and the Last Seen is not updated.

I have several problems with this: I don't think a Car should have its
Location changed because it passed over a reader. I'm fine with saving the
"whereLastSeen" field (I wish that were easier to observe), but a reader is
a point detector: regardless of where you put it, it only reflects that one
point. If you put the reader at the entrance to a siding, you don't know if
the Car is on the way in, or on the way out (or even just passing over
because the next car in needs to be pulled. An OperationsPro train will be
updating the Car's Location and I would be quite upset if the known
OperationsPro location were to be changed by the tag reader. Of course, I
can easily avoid that by not associating a reader with a Location (by the
way it only checks the Tracks and not the Locations). My readers are
located at the entrance to the Yard and not on a specific track.

Look at the change I just put in (it's in the just released code). That
opens a window for that specific reader and displays the tag, the road and
car number (if there is a car that has that tag). If the tag does not have
an associated Car, a button lets you choose from a list of all the cars (by
default, those without tags). If there is an associated Car, the Location
and Track as shown as a ComboBox -- if you change them, you can click a
button to set the location. Another button lets you bring up the Edit Car
window for that Car.

I plan to use the window for the Yardmaster to check the incoming/outgoing
trains for the correct cars. Between Ops sessions, I plan to pull each
track over the reader to verify that what OperationsPro thinks matches what
it really is. I expect others will find different ways to do this.

pabender, 2022-02-14:
> The propertyChange listener in RollinngStock.java, when a "whereLastSeen" event occurs, does a locationManager.getTrackByReport( ) passing in the tag that was detected, to determine if the Reporter [...]

Right, and this is where the bug is.

> I have several problems with this: I don't think a Car should have its Location changed because it passed over a reader. I'm fine with saving the "whereLastSeen" field (I wish that were easier to [...]

This is correct, it is a point detector.  BUT If you're moving the car locations along with a train, the last known point location is a better location for the car than the place where the train picked it up.

> If you put the reader at the entrance to a siding, you don't know if the Car is on the way in, or on the way out (or even just passing over because the next car in needs to be pulled. An [...]

And this is where Dan and I have disagreed on approach in the past.  The current code updates the location based only on tracks with reporters.  At one point in time, the code updated the location based on location and not just tracks.  When it did that, it would set the track to null (which is why Dan didn't like it).  But this behavior reflects the way that AEI readers (the prototype's RFID implementation)  are used on the prototype.  AEI readers tend to be placed at key points in the system (entrances and exits from yards and junctions in particular).   From that information gathered as a train passes from one location to another you can get a list of the cars in a train in standing order AND you can get a list of cars that were left somewhere between those two points.  

If you know the switch positions, it is possible to determine what cars were pushed onto a given track with only a single real RFID reader.  (see https://github.com/pabender/Sensys2014/blob/master/ReporterTrackContents.py ).

swalton00, 2022-02-14:
But, I went out with a switch list saying pick up car 123 at siding A  and
deliver it to siding C. Along the way, we passed over a reader (at location
B). I really don't think JMRI should set car 123 to be at siding B.
OperationsPro thinks it's part of a train, and the final location will be
C.

If I put a reader at the entrance to a siding, that still only tells me
that the car passed over the reader. One of my Locations has 3 "Tracks" --
it's actually 3 different spots on the same physical track (because I want
to route cars to those three industries differently. The outer track
actually has a delay built in. That means, the engineer,  in working the
siding needs to pull the cars from the outer industry to get to the inner
cars (which are the ones he really wants). That means those outer cars pass
over the reader several times (but don't really change location). As I see
it, there are two ways to work this. I want to work this as an engineer and
a yardmaster who are given a switch list telling them which cars they
should pick up and which ones to drop. I will use the RFID reader to assist
the yardmaster in verifying incoming and outgoing cars. That means the JMRI
has the final word. Of course crews can make mistakes so I want to pull a
string of cars and run them over a reader to verify their location for the
next operating session.

I can see if you want to let engineers pick up whichever cars they want,
(or you are using some other system for setting up your operating
sessions), then if you want JMRI to help you track the locations of cars,
then forcing the location based on what the reader says is the proper
answer. I can see it working either way - if you want to use the second
way, then set the location of your readers. If you don't want the reader to
interfere with the location of the car, then don't tell JMRI a Track
location for the reader.

I can also see why others would have a problem with a car having a Location
without a Track. The way the software is organized there really isn't a
hierarchy of Location then Track within Location - there are
Locations/Tracks  They are sort of the same entity. I would not have
written the code that way, but I did not write the code. And criticizing
other peoples code for things like that is a good way to start a religious
war -- people have good reasons for doing it a certain way.

My bigger issue is about the split between PanelPro tags and OperationsPro
tags. Should there be a separation between the two and is the current
crossover correct. I don't ever use User Names on tags (and from the
OperationsPro standpoint, they aren't visible). But PanelPro allows you to
have them and the tags are sorted by User Name and if that is not present,
by System Name. Right now there is what I believe is a bug -- if you
restore the OperationsPro data to a different system, you will get an Error
message for every tag that has not been read on the JMRI profile. The issue
comes up because OperationsPro only copies Operations files when it does a
restore, and PanelPro stores tags in a different location.

pabender, 2022-02-14:
> But, I went out with a switch list saying pick up car 123 at siding A and deliver it to siding C. Along the way, we passed over a reader (at location B). I really don't think JMRI should set car 123 to be at siding B. OperationsPro thinks it's part of a train, and the final location will be C. 

Oh, I understand that.  Cars also get moved when you move the location of the train, but that movement is just an assumption that the move actually happened.

RFID allows us to get a better picture that the move actually happened as expected, and this is the use case I'm going for.   The idea is to consider what happens when someone forgets to spot a car or picks up a car they were not supposed to.  In either case, the car ends up at the train's destination (physically) but the operations code says the car is somewhere else. 

Without updating the location of the car based on where the reader is (even without a track) The operations software has no clue where the car actually is, so it can't take corrective action to add the car to a different train to get spotted in the right location.

> My bigger issue is about the split between PanelPro tags and OperationsPro tags. Should there be a separation between the two and is the current crossover correct. I don't ever use User Names on [...]

Ok, so IDTags are named beans associated with a layout. All Named Beans in JMRI have both a UserName and a SystemName.

When you select an RfidTag (or any other IdTag) to associate with a car, it uses the NamedBeanComboBox to pull the IDTag information.  When the IDTag has a user name, that UserName is displayed in the drop down box.  If it doesn't have a user name, then the system name is used.  (I just checked with current head of master, and this works as expected, at least when adding cars)

Operations uses the same tags that the rest of JMRI uses.  The only thing it stores within its own code is a pointer to the IDTags.  This is currently storing the system name of the tag, though that is wrong.  Why is that wrong?  Well, It's possible for either the the Tag ID associated with a given piece of rolling stock to change OR for the tag ID to be associated with a different piece of rolling stock ( in the first case, what happens if the tag breaks or is lost (falls off the car).  The second case may be more rare, but in my case it could happen if i accidentally swapped floors on two Micro Trains boxcars, for example).  

Being able to swap the physical thing on the layout with the logical thing in software is one of the reasons we have UserNames in the first place.  

In the big picture, IDTags are used on the whole layout, and may be used for more than one thing at a time, but every use of a tag is referencing the same physical object on the layout (which is the tag, not the piece of rolling stock to which it is attached).  (IDTags can also be used for block tracking, for example).

So you might ask yourself why aren't IDTags stored in the panel file for the layout?  Well, the reason for that is that we want to save the state of the IDTags even if we don't save the panel file which contains the layout description.  You probably aren't experiencing the same issues with the Reporters you have associated at a location because you are also loading the panel file containing the reporter definitions.  The Reporters aren't stored with the Operations data either (and they shouldn't be).

DanielBoudreau, 2022-02-14:
The other issue with using the same “location” field for operations and rfid is that you can run operations in real time.  A car’s location is updated by operations as the user moves the train in real time.  The reader would update car to where is actually is, which might be part of the train’s route or not.  So the real time conductor could get confused to where and when a car should be removed or added to a train.  Today’s operations code didn’t consider the case where an outside agent could update the car’s location and how to properly handle the various scenarios.

Updating a car’s location when it hasn’t been assigned to a train shouldn’t be an issue for operations.  Updating a car’s location once it has been scheduled for a pick up or when it’s part of train could break things. The code could be enhanced, but it would also need a lot of testing to cover all of the different cases.  IMO, better to keep the two locations separate, expected and actual.

Dan

Scott Walton

pabender, 2022-02-14:
> Updating a car’s location when it hasn’t been assigned to a train shouldn’t be an issue for operations. Updating a car’s location once it has been scheduled for a pick up or when it’s part of train [...]

My primary use case for RFID in operations, which I mentioned above, is actually always about updating the car location after it was supposedly spotted where it was destined.... i.e. I want to know when either a car wasn't really spotted where expected OR when the car was picked up when it wasn't supposed to have been.  In both of those cases, the cars are no longer in a train (if you move the train in real time) but aren't where they are supposed to be.

I only have a handful of readers, and they are placed at the entrance to yards and at interchange points.  I don't consider it cost effective to place tag readers on every track, though there are some who want to do that.

KenC57, 2022-02-14:
Listening to this, I think it becomes important that a reader has options for where it is and what it should
do. If you use readers to check the cars going out in a train, you'd want it to update the train somehow. But
that would require the readers include the ability to recognize the train. At the end of the train, if a car
wasn't seen in the train, it needs to be delete. If a car is in the train, but not in the manifest, wherever
that car was before gets changed to be in this train now and heading to a yard as lost.

Now if a reader was at a location, and only serving one industry, it would know it made it to that track. But
as pointed out, if that track had multiple spots, it still doesn't know which. If it was placed where it saw
cars arriving in a town, it could know any cars not on the outbound train (presuming it had a reader at both
ends) must be somewhere in the town. Ok, no exit reader, then if it didn't make it to the next town, it must
have stayed in that first town.

But all that is quite complex with seeing or not seeing, knowing it is part of some train, etc... Here I
suggest we get back to the 'use case' as a great way to identify where a reader is useful for operations.
Think of where a reader might be placed, what it would see, and what should it do about it. From all of this,
it is sounding more useful to just have that 'where last seen' and higher level or human interactive gets to
decide if car locations deserve to be updated.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

pabender, 2022-02-14:
> From all of this, it is sounding more useful to just have that 'where last seen' and higher level or human interactive gets to decide if car locations deserve to be updated.

Which is what the current behavior is.

DanielBoudreau, 2022-02-15:
I think it would be neat when the two disagree to color the row red or whatever.  I don’t use RFID so not a good person to implement.

Dan

KenC57, 2022-02-15:
Paul's use case, at yards and interchanges, would be best I think at reordering the cars in the manifest to
show the 'as seen' order of the cars in the train. That's about as far as I'd see a reader being good for
under those setups. That would make switching easier since you know how blocked (or not) the cars really are
in the train. That I could see as something automation could do with enough safety. The train is its current
location.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

pabender, 2022-02-15:
> Paul's use case, at yards and interchanges, would be best I think at reordering the cars in the manifest to show the 'as seen' order of the cars in the train. That's about as far as I'd see a reader [...]

I can actually get the car ordering information in other ways.

I actually do that using the collection on the internal reporters.  The script I linked to above is the key to that.

swalton00, 2022-02-15:
Actually the current behavior is to do nothing. Because of the bug (lack of
an equals method), there is no reporter found at any location. Thus
whereLastSeen is never set. The current code, if a Location were found, is
to set whereLastSeen and also to set the Location for the Car.

Tomorrow, I will add an equals() method (probably to Reporter). I believe I
should also add check for whether the Car is in a Train and, if it is, skip
the setting of the Location. Also, I will change the setRfid() method to
use provideIdTag() instead of getIdTag(). That will create the tag in the
current system, and thus avoid the Error message we are currently seeing.

pabender, 2022-02-15:
> Actually the current behavior is to do nothing. Because of the bug (lack of an equals method), there is no reporter found at any location. 

You are making an assumption that you shouldn’t make.  Not all reporters use the TimeoutReporter decorator, and the bug only impacts Reporters decorated that way.

For other reporters ( such as Transponding Reporters or RailCom reporters ) the comparison does the right thing and will set the whereLastSeen. ( there is actually an equals defined in AbstractNamedBean and the reporters ( except the TimeoutReporterDecorator) are all derived from that.  ( decorators specifically are not named beans,  they decorate some other thing, in this case a reporter, which is a named bean ).

The issue with TimeoutReporter was introduced about 2 years ago.

pabender, 2022-02-15:
> Tomorrow, I will add an equals() method (probably to Reporter). 

See PR #10706.  This makes sure the decorator does a check of the underlying object in its equals method.