JMRI/JMRI #10689: LogixNG bug in PR #10679

dsand47, 2022-02-12:
Index out of range exception:

```
     [java] 10:36:31,674 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/Users/das/JMRI/_Configs/H&D/MILW%20H&D%20Phase%201.xml) in adaptor of type jmri.jmrit.logixng.implementation.configurexml.DefaultDigitalActionManagerXml Exception: java.lang.StringIndexOutOfBoundsException: String index out of range: -1
     [java] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
     [java] java.lang.StringIndexOutOfBoundsException: String index out of range: -1
     [java] 	at java.base/java.lang.String.substring(String.java:1841)
     [java] 	at jmri.jmrit.logixng.actions.configurexml.ActionMemoryXml.load(ActionMemoryXml.java:175)
     [java] 	at jmri.jmrit.logixng.implementation.configurexml.DefaultDigitalActionManagerXml.loadActions(DefaultDigitalActionManagerXml.java:139)
     [java] 	at jmri.jmrit.logixng.implementation.configurexml.DefaultDigitalActionManagerXml.load(DefaultDigitalActionManagerXml.java:94)
```

Caused by empty `otherTableCell` element:

```
    <ActionMemory class="jmri.jmrit.logixng.actions.configurexml.ActionMemoryXml">
      <systemName>IQDA:AUTO:0131</systemName>
      <memory>Mpls Status</memory>
      <addressing>Direct</addressing>
      <reference />
      <localVariable />
      <formula />
      <memoryOperation>SetToString</memoryOperation>
      <otherConstant>OK</otherConstant>
      <otherTableCell />
      <otherVariable />
      <otherFormula />
      <MaleSocket>
        <AbstractDebuggerMaleSocket class="jmri.jmrit.logixng.tools.debugger.configurexml.DebuggerMaleDigitalActionSocketXml" />
        <AbstractMaleSocket enabled="yes" locked="no" system="no" catchAbortExecution="no" class="jmri.jmrit.logixng.implementation.configurexml.DefaultMaleDigitalActionSocketXml">
          <errorHandling>Default</errorHandling>
        </AbstractMaleSocket>
      </MaleSocket>
    </ActionMemory>
```

The following diff fixes the problem.  There may be others.

```
diff --git a/java/src/jmri/jmrit/logixng/actions/configurexml/ActionMemoryXml.java b/java/src/jmri/jmrit/logixng/actions/configurexml/ActionMemoryXml.java
index fdca727fbb..239a6bcd02 100644
--- a/java/src/jmri/jmrit/logixng/actions/configurexml/ActionMemoryXml.java
+++ b/java/src/jmri/jmrit/logixng/actions/configurexml/ActionMemoryXml.java
@@ -172,30 +172,32 @@ public class ActionMemoryXml extends jmri.managers.configurexml.AbstractNamedBea
                 if (elem != null) {
                     boolean result = false;
                     String ref = elem.getTextTrim();
-                    String[] refParts = ref.substring(1).split("[\\[\\]]");  // Remove first { and then split on [ and ]
-//                    System.out.format("refParts.length: %d, '%s', '%s'%n", refParts.length, refParts[0], refParts[1]);
-                    if (refParts.length == 3) {
-                        String table = refParts[0];
-                        String[] rowColumnParts = refParts[1].split(",");
-                        if (rowColumnParts.length == 2) {
-                            String row = rowColumnParts[0];
-                            String column = rowColumnParts[1];
-//                            System.out.format("Table: '%s', row: '%s', column: '%s'%n", table, row, column);
-
-                            h.setTableNameAddressing(NamedBeanAddressing.Direct);
-                            if (table != null) {
-                                NamedTable t = InstanceManager.getDefault(NamedTableManager.class).getNamedTable(table);
-                                if (t != null) h.setTable(t);
-                                else h.removeTable();
+                    if (!ref.isEmpty()) {
+                        String[] refParts = ref.substring(1).split("[\\[\\]]");  // Remove first { and then split on [ and ]
+    //                    System.out.format("refParts.length: %d, '%s', '%s'%n", refParts.length, refParts[0], refParts[1]);
+                        if (refParts.length == 3) {
+                            String table = refParts[0];
+                            String[] rowColumnParts = refParts[1].split(",");
+                            if (rowColumnParts.length == 2) {
+                                String row = rowColumnParts[0];
+                                String column = rowColumnParts[1];
+    //                            System.out.format("Table: '%s', row: '%s', column: '%s'%n", table, row, column);
+
+                                h.setTableNameAddressing(NamedBeanAddressing.Direct);
+                                if (table != null) {
+                                    NamedTable t = InstanceManager.getDefault(NamedTableManager.class).getNamedTable(table);
+                                    if (t != null) h.setTable(t);
+                                    else h.removeTable();
+                                }
+                                h.setTableRowAddressing(NamedBeanAddressing.Direct);
+                                h.setTableRowName(row);
+                                h.setTableColumnAddressing(NamedBeanAddressing.Direct);
+                                h.setTableColumnName(column);
+                                result = true;
                             }
-                            h.setTableRowAddressing(NamedBeanAddressing.Direct);
-                            h.setTableRowName(row);
-                            h.setTableColumnAddressing(NamedBeanAddressing.Direct);
-                            h.setTableColumnName(column);
-                            result = true;
                         }
+                        if (!result) throw new JmriConfigureXmlException("otherTableCell has invalid value: "+ref);
                     }
-                    if (!result) throw new JmriConfigureXmlException("otherTableCell has invalid value: "+ref);
                 }
 
                 elem = shared.getChild("otherVariable");
```

danielb987, 2022-02-12:
Thanks. I apologize for this. I will fix it ASAP.

The problem exists in ActionMemoryXml and ActionLocalVariableXml.

dsand47, 2022-02-12:
> Thanks. I apologize for this. I will fix it ASAP.
> 
> The problem exists in ActionMemoryXml and ActionLocalVariableXml.

There is another problem but I have not figured out where it is yet.

danielb987, 2022-02-12:
> Caused by empty `otherTableCell` element:

This is fixed by PR #10690.

danielb987, 2022-02-12:
> There is another problem but I have not figured out where it is yet.

Can you explain the problem?

dsand47, 2022-02-12:
> > There is another problem but I have not figured out where it is yet.
> 
> Can you explain the problem?

I am creating a module.

[image]

The disabled line is actually a local variable.  I think it is just a label problem.  However, it did get a NPE.
```
     [java] 11:44:06,739 implementation.DefaultConditionalNG   WARN  - ConditionalNG IQC:AUTO:0052 was aborted during execute: java.lang.NullPointerException [JMRI LogixNGThread]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrit.logixng.actions.ActionLocalVariable.getTableRow(ActionLocalVariable.java:628)
     [java] 	at jmri.jmrit.logixng.actions.ActionLocalVariable.lambda$execute$0(ActionLocalVariable.java:709)
     [java] 	at jmri.util.ThreadingUtil.lambda$runOnGUIWithJmriException$0(ThreadingUtil.java:184)
     [java] 	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:303)
```

I also tried a memory variable.
```
     [java] 11:59:21,344 implementation.DefaultConditionalNG   WARN  - ConditionalNG IQC:AUTO:0052 was aborted during execute: java.lang.NullPointerException [JMRI LogixNGThread]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrit.logixng.actions.ActionMemory.getTableRow(ActionMemory.java:610)
     [java] 	at jmri.jmrit.logixng.actions.ActionMemory.lambda$execute$0(ActionMemory.java:716)
     [java] 	at jmri.util.ThreadingUtil.lambda$runOnGUIWithJmriException$0(ThreadingUtil.java:184)
     [java] 	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:303)
```

danielb987, 2022-02-12:
> The disabled line is actually a local variable. I think it is just a label problem.

Yes, copy/paste error. Fixed in #10690.

danielb987, 2022-02-12:
> I am creating a module.

Problem found. Will fix.

closed 2022-02-13 by dsand47