JMRI/JMRI #10711: Default values for speed table

marko-pi, 2022-02-15:
I propose that default values for speed table could be specified within XML of a decoder.  There is a rather ancient sample for this 

```
<variable CV="67" item="Speed Table">
<speedTableVal>
<speedTableEntry step="1" value="0"/>
<speedTableEntry step="2" value="0"/>
...
<speedTableEntry step="27" value="0"/>
<speedTableEntry step="28" value="0"/>
</speedTableVal>
<label>Speed Table</label>
</variable>

```
but it is not working.

Best regards

PUlvestad, 2022-02-15:
Related users group thread
https://groups.io/g/jmriusers/message/200895

dheap, 2022-02-15:
> Related users group thread
> https://groups.io/g/jmriusers/message/200895

```
This proposal looks easier to implement:
<variable CV="67" item="Speed Table" default="0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0">
<speedTableVal/>
<label>Speed Table</label>
</variable>
```

I'd further suggest that:
```
<variable CV="67" item="Speed Table" default="0">
<speedTableVal/>
<label>Speed Table</label>
</variable>
```
would be treated as an equivalent form, as we do with multiple masks.

marko-pi, 2022-02-16:
I also think that

<variable CV="67" item="Speed Table" default="0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0">

is nicer.

By the way, it is highly unlikely that all default values will be zero.  I just put zeroes as an example.  No need for an equivalent form, IMHO.

dheap, 2022-02-16:
With SplitVal We have to do something if too few masks are supplied. So we [use last mask for all following CVs if fewer masks than the number of CVs listed were provided](https://github.com/JMRI/JMRI/blob/295d9280cd6b02192406d7e485ac5a63310ef117/java/src/jmri/jmrit/symbolicprog/SplitVariableValue.java#L113).

Likewise we need to do something meaningful with too few defaults. I'd suggest we do the same.

marko-pi, 2022-02-16:
IMHO, this is fine as a way to avoid error.  After all, it is on the author of the decoder to specify the correct number of masks.

bobjacobsen, 2022-02-20:
This is a workaround:
```
<variable item="Speed Table" CV="67">
    <speedTableVal/>
</variable>
<variable item="default CV 71" CV="71" default="14"><decVal/></variable>
<variable item="default CV 72" CV="72" default="21"><decVal/></variable>
<variable item="default CV 73" CV="73" default="28"><decVal/></variable>
```

marko-pi, 2022-02-20:
> This is a workaround:
>
> ```
> <variable item="Speed Table" CV="67">
>     <speedTableVal/>
> </variable>
> <variable item="default CV 71" CV="71" default="14"><decVal/></variable>
> <variable item="default CV [...]

I am sorry, this did not work for me .

dheap, 2022-02-20:
> I am sorry, this did not work for me .

Maybe because you didn't set the default values for CVs 67,68,69,70 and this resulted in a non-monotonic curve. See:
https://github.com/JMRI/JMRI/blob/ae29730d4176e40af7b2810cdec9afa3e3a83cfe/java/src/jmri/jmrit/symbolicprog/SpeedTableVarValue.java#L63

marko-pi, 2022-02-20:
```
			<variable CV="67" item="Speed Table">
				<speedTableVal />
				<label>Speed Table</label>
			</variable>
			<variable CV="67" item="CV 67 default" default="0"><decVal/></variable>
			<variable CV="68" item="CV 68 default" default="1"><decVal/></variable>
			<variable CV="69" item="CV 69 default" default="2"><decVal/></variable>
			<variable CV="70" item="CV 70 default" default="3"><decVal/></variable>
			<variable CV="71" item="CV 71 default" default="4"><decVal/></variable>
			<variable CV="72" item="CV 72 default" default="5"><decVal/></variable>
			<variable CV="73" item="CV 73 default" default="6"><decVal/></variable>
			<variable CV="74" item="CV 74 default" default="7"><decVal/></variable>
			<variable CV="75" item="CV 75 default" default="8"><decVal/></variable>
			<variable CV="76" item="CV 76 default" default="9"><decVal/></variable>
			<variable CV="77" item="CV 77 default" default="10"><decVal/></variable>
			<variable CV="78" item="CV 78 default" default="11"><decVal/></variable>
			<variable CV="79" item="CV 79 default" default="12"><decVal/></variable>
			<variable CV="80" item="CV 80 default" default="13"><decVal/></variable>
			<variable CV="81" item="CV 81 default" default="14"><decVal/></variable>
			<variable CV="82" item="CV 82 default" default="15"><decVal/></variable>
			<variable CV="83" item="CV 83 default" default="16"><decVal/></variable>
			<variable CV="84" item="CV 84 default" default="17"><decVal/></variable>
			<variable CV="85" item="CV 85 default" default="18"><decVal/></variable>
			<variable CV="86" item="CV 86 default" default="19"><decVal/></variable>
			<variable CV="87" item="CV 87 default" default="20"><decVal/></variable>
			<variable CV="88" item="CV 88 default" default="21"><decVal/></variable>
			<variable CV="89" item="CV 89 default" default="22"><decVal/></variable>
			<variable CV="90" item="CV 90 default" default="23"><decVal/></variable>
			<variable CV="91" item="CV 91 default" default="24"><decVal/></variable>
			<variable CV="92" item="CV 92 default" default="25"><decVal/></variable>
			<variable CV="93" item="CV 93 default" default="26"><decVal/></variable>
			<variable CV="94" item="CV 94 default" default="27"><decVal/></variable>

```

It is not working.  Cross my heart.

bobjacobsen, 2022-02-20:
It works for me.

 - Edit that into xml/decoders/nmra/cv67speedTableBasic.xml
 - Open DecoderPro
 - Click "New"
 - Select NMRA -> NMRA standard CV definitions
 - In the window that opens, open the Speed Table tab
 - And there it is.

What _exactly_ are you doing, and what does "not working" mean?

marko-pi, 2022-02-20:
I don't have "New" in my DecoderPro version 4.26+R381c8dfc32 on Raspberry Pi.

If I use "New Loco", there is no "NMRA"

bobjacobsen, 2022-02-20:
"new Loco" should lead to a screen like this:
[image]

If you don't have that, how are you making a new roster entry to test your revised decoder file?

marko-pi, 2022-02-20:
I am sorry, I didn't see NMRA because it is not in alphabetic order.

Yes, it works if I edit in xml/decoders/nmra/cv67speedTableBasic.xml

However, if I make my own decoder file, then it does not work.  Default values are not changed.

bobjacobsen, 2022-02-20:
- Does your version validate?  (DP File menu -> Validate XML file)
 - If so, please attach your file here (will need to add a .txt extension) and I'll take a look at it.

marko-pi, 2022-02-20:
Yes, I validated.  I am attaching the file.

Thanks.

[Public_Domain_MPED_mul.txt](https://github.com/JMRI/JMRI/files/8104119/Public_Domain_MPED_mul.txt)

bobjacobsen, 2022-02-20:
- I copied that file (with a .xml extension) to xml/decoders
 - recreated decoder index
 - New Loco, selected Public Domain -> MPED DIY -> MPED Basic LocoMotive
 - Looked at speed table and see:
[image]

marko-pi, 2022-02-20:
That's strange.  Using the same procedure I get this:

[image]

bobjacobsen, 2022-02-20:
I suspect you're not changing the file you think you're changing. The decoder definition files can be the xml/decoders directory in two places: in the program directory, or in the preferences directory.  The preferences directory is used first.  So check that you have your file in only one of those places.

marko-pi, 2022-02-20:
OK, I put my XML in _programming_ **xml/decoders** directory.  I also edited **decoderIndex.xml** in _programming_ **xml** directory by adding 

```
      <family name="MPED DIY" mfg="Public-domain and DIY" file="Public_Domain_MPED_mul.xml">
        <model model="MPED Basic Locomotive" />
      </family>

```
I also recreated decoder index just in case.  **I did not make any change to preferences directory**.

It actually works fine except for that speed table problem.

Is it possible that speed table behaviour depends on whether my decoder definition is either in preferences or programming directory?

bobjacobsen, 2022-02-20:
> Is it possible that speed table behaviour depends on whether my decoder definition is either in preferences or programming directory?

Extremely unlikely.  The file is opened, and then everything else is just reading and interpreting the contents.

Try changing something else, e.g. a label, and make sure that change shows up when you use your decoder definition.

bobjacobsen, 2022-02-20:
> I also recreated decoder index just in case. I did not make any change to preferences directory.

Well, actually you did.  That recreated decoder index should have been put in your preferences directory.  In the preferences directory you should now see a `decoderIndex.xml` file. (if not, then you might be looking in the wrong place). Is there a decoders/ directory there?  Does it have anything in it?

marko-pi, 2022-02-20:
I have started from scratch, reinstalling the program.  Now it is working.  I guess there must have been some mix up with my directories.

I also found out that this

```
                        <variable item="Speed Table" CV="67">
				<speedTableVal />
				<label>Speed Table</label>
			</variable>
			<variable CV="67" item="CV 67 default" default="0"><decVal/></variable>
			<variable CV="68" item="CV 68 default" default="0"><decVal/></variable>
                        ...
			<variable CV="93" item="CV 93 default" default="0"><decVal/></variable>
			<variable CV="94" item="CV 94 default" default="0"><decVal/></variable>
		</variables>

```

still does not work.  **So in order for this workaround to work, _all_ default CVs must be specified and each following must be larger than the previous one.**

So is the idea of

`<variable CV="67" item="Speed Table" default="1,2,3,...,28">
`
still actual, or should this issue be closed?

bobjacobsen, 2022-02-20:
> So in order for this workaround to work, all default CVs must be specified and each following must be larger than the previous one.

No, see my original example.

JMRI does insist that each CV be greater than or equal to the prior one (alternately, each one be less than or equal to the next one).  But you don’t have to specify them all, and they can be equal from one to the next.

bobjacobsen, 2022-02-20:
There does seem to be a problem with setting them all to zero.  All 1 works, but all zero causes some internal computation to fail at the end of the initialization.  Not sure where, exactly, maybe the interpolation setup.  Why would you want a speed table that's all zero?

marko-pi, 2022-02-20:
Specifying just _two_ zeroes also does not work.

Putting all zeroes is not important, but I used zeroes for testing (not being aware as this is exactly the problem).

bobjacobsen, 2022-02-20:
If anybody wants to look at adding true defaults, here's info in hopes that it's useful:
 - the `speedTableEntry` approach is in the schema, but not otherwise implemented
 - the `SpeedTableVarValue` object is created in `VariableTableModel`. Right after ctor would be the obvious place to load defaults
 - But there's a race condition because the `SpeedTableVarValue` ctor sets the values to a default ramp, then attaches a listener, then sets the state. (That order is probably necessary to get the color right on the display) Those events are still being processed from the event queue after `VariableTableModel` has moved on to the next variable.
 - So it would be better to pass into the ctor a list or array of default values that the `SpeedTableVarValue` could set instead of the linear ramp ones. Then there'd be no race condition with setting them afterwards.

closed 2022-04-22 by stale[bot]