JMRI/JMRI #10735: JMRI drops TCP connection; doesn't reconnect.

TheKenWiley, 2022-02-21:
I have installed an RFID reader system which uses TCP via CBUS in JMRI.  I can start JMRI and read tags without problems.  If left alone for any period of time, JMRI no longer accepts incoming data from the reader.  It does not report any problems.  It shows the connection still active in the main screen, but the only way to re-establish the connection is to restart JMRI.  I will include an email segment from a friend having the same problem:

I have looked at the sequence of events between JMRI and the hub.  JMRI initiates the TCP connection when it starts.  The hub sends regular "keep alive" messages to make sure the JMRI is still connected, which JMRI acknowledges.  However, JMRI does not send out "keep alive" messages, so if the hub goes away, JMRI is not aware of this. When the hub comes back, therefore, JMRI is not aware that the TCP connection has been lost.
What should happen is that once JMRI has established a connection with the hub (and remember, in this situation JMRI is the "master" and the hub is the "slave"), JMRI should send out regular "keep alive" messages, to check that the hub is still there.  If the hub goes away, JMRI would therefore be aware of this, and after a short time would end the TCP connection within JMRI, and start a sequence of trying a new TCP connection on a regular basis.  When the hub comes back, the new TCP connection would be established, and data can then start to flow between hub and JMRI.  At present, a new TCP connection is only ever attempted when JMRI starts - so if the TCP connection is lost, you have to shut JMRI down and start it up again.

bobjacobsen, 2022-02-22:
When the connection stops working, is there anything in the JMRI system log?  I.e. a message like "Exception in receive loop" or "Exit from receive loop"?

If not, the TCP connection hasn't been broken. Something else is going wrong.

TheKenWiley, 2022-02-22:
I am down loading Java 11, JMRI 4.99.3 as I write this.  I will let you know the first time it happens.  I was using the last production release of JMRI, 4.24 I believe.

danielb987, 2022-02-22:
@bobjacobsen 
> When the connection stops working, is there anything in the JMRI system log? I.e. a message like "Exception in receive loop" or "Exit from receive loop"?
> 
> If not, the TCP connection hasn't been broken. Something else is going wrong.

Are you sure about this? Usually when a device closes the TCP connection, it sends a TCP packet that notifies the other part that the connection is closed. But if the device drops dead without sending any packet, the other part will not get any notification. And in this case, the only way the other part can know that the connection is broken is when it tries to send anything without getting any acknowledge.

I don't know anything about CBUS so I might be wrong in this case, but the first message from @TheKenWiley in this thread makes perfectly sense to me if JMRI doesn't send any "keep alive" and if the other part suddenly drops dead without alerting JMRI. If that device has an internal "keep alive" feature that reboots it when it has crashed, it might reboot itself to start working again. But JMRI isn't aware of it and therefore doesn't try to reconnect.

danielb987, 2022-02-22:
A practical demonstration of the above:

I have a Raspberry Pi connected with WiFi. I connect to it with SSH from my Windows 8.1 computer. While connected, I remove the power to the Raspberry Pi, causing it to drop dead without being able to notify my Windows computer. I reconnect the power to the Raspberry Pi and it starts. But my SSH client on the Windows computer hasn't still realised that the connection is dead. It's not until I write something on the keyboard, which causes the SSH client to send some data, that the SSH client understands that the connection is gone.

bobjacobsen, 2022-02-22:
TCP connections don't need keep-alive messages. There's a [well-defined protocol ](https://www.geeksforgeeks.org/tcp-connection-termination/) that determines whether the connection has closed, regardless of whether characters are crossing it or not.

It's certainly possible that one end or the other has lost the ability to send, or perhaps to receive, those characters.  But that's different from actually having the connection close.

bobjacobsen, 2022-02-22:
> If that device has an internal "keep alive" feature that reboots it when it has crashed, it might reboot itself to start working again. 

Do we have evidence whether that's happening or not?  If so, that seems to be the thing to fix.

danielb987, 2022-02-22:
> TCP connections don't need keep-alive messages. There's a [well-defined protocol ](https://www.geeksforgeeks.org/tcp-connection-termination/) that determines whether the connection has closed, regardless of whether characters are crossing it or not.

Yes, if and when JMRI sends something. But if the RFIDs are the only things connected to CBUS and JMRI doesn't read them unless the hub sends a message telling JMRI that the RFID has been updated, JMRI will not send any message if the hub goes away. My point is that if JMRI is only listening on the connection, it will not be notified if the hub drops dead.

bobjacobsen, 2022-02-22:
I understand those examples. I don't think JMRI can drop the connection without being aware of it. (See the title of the issue) Nor do I think that the connection itself can be lost (e.g. losing connectivity) at the JMRI end without that being discovered. 

If the problem is that the other-end socket is crashing without resetting the connection, then perhaps JMRI could time out, close and reopen the connection as a retry, and have mechanisms to retry that retry when it occasionally fails. Reopening the connection after a crash will almost certainly still result in missed reads of the RFID tags.  It seems better to figure out why the other end is crashing and fix that rather than trying to cope with it from outside.

TheKenWiley, 2022-02-23:
I haven't had the chance to check any of this since my first posting.  I'm running window10, jmri4.24, java8.  The creator of the RFID reader is testing and believes he has found the problem, not in JMRI, that makes it appear as though the connection has dropped.  I apologize for the original posting.

icklesteve, 2022-02-23:
Hi @TheKenWiley 

It may be a quick fix to kick-start the JMRI reconnect by starting the CBUS Fast Clock on Startup.
As this sends out regular messages, if one is unable to be sent, the exception is caught and the reconnect attempts start.

> The hub sends regular "keep alive" messages to make sure the JMRI is still connected, which JMRI acknowledges.

Interesting, there's nothing in the CBUS protocol for transport connection keep-alive,
if I could ask what form are these messages are in?

Ta,
Steve.

MartinNotts, 2022-02-24:
Danielb987 is correct.  If the hub or Pi drops dead for any reason, such as the power being removed (maybe inadvertently), then when the hub or Pi powers up again, there is no TCP connection alive.  However, JMRI still thinks that the connection is alive.  Analysing the problem using Wireshark, once JMRI, which is the master has established a TCP connection (SYN, SYN/ACK, ACK) then JMRI just sits there waiting for messages.  Because JMRI does not send any "keep alive" messages of any sort, JMRI has no idea if the hub or Pi is still there.  Now, after about 8000 seconds (2.25 hours), JMRI realises it has not received any messages, and issues a RESET packet.  Immediately after that, JMRI starts a new TCP connection.  If the connection has not failed during the 2.25 hours with no data, then the RESET packet tells the hub to terminate the TCP connection, and then JMRI and the hub start a new TCP connection. If the connection has failed, then a new connection is established.  I can see this quite clearly on the CBUS console window in JMRI - every 2.25 hours without any activity, the old connection is dropped, and a new connection is made.
However, there is no indication at all on JMRI that the connection has dropped, even when no messages are being received. If JMRI were to send "keep alive" messages regularly, then it would [1] realise the connection had dropped, and could initiate a reset and a new TCP session and [2] inform the user that the connection had dropped.
To icklesteve, there is a method within TCP to send a "keep alive" connection, it is just a packet with no information in it, and all it is looking for is an ACK packet from the other end.  The hub sends these packets out every 20 seconds or so, and if JMRI does not respond for several "keep alive" packets, then the hub assumes JMRI has gone away without issuing a proper FIN or RESET packet, and drops the TCP connection.
If JMRI issued "keep alive" packets regularly, then it would know if the hub had gone away, could inform the user, and could try and re-initiate the TCP connection.
In this example, I am only referring to JMRI on the MERG CBUS console.

MartinNotts, 2022-02-24:
Right, I have done some more tests with Wireshark
Using icklesteve's suggestion, I have set up a fast clock, and I can see that JMRI is sending out a packet once a minute.  This packet is being acknowledged by the hub, and I am able to send RFID data into JMRI, so all is fine.
The following is a timeline, in whole seconds, of what happens when the power on the hub is disconnected, and then reconnected.
67 - JMRI sends fast clock, hub acknowledges
80 - hub is powered down
90 - hub is powered up, but obviously the TCP connection is now no longer valid into the hub.
127 - JMRI sends fast clock, receives no ACK
127 - two TCP re-transmissions, no ACKs
128 - TCP re-transmission, no ACK
130,133,141,155,182,239,352,475,598.720,843,966 - TCP retransmissions, no ACK
However, during this time, the CBUS console window is showing fast clock transmissions, with no indication that they are not being received.
1090 - JMRI send SYN to restart connection on different output port number
1090 - Hub sends SYN, ACK
1090- JMRI sends ACK
1090 - Hub sends CBUS General ACK
1090 - JMRI sends ACK
1146 - JMRI sends fast clock, hub sends ACK
By 1091 connection is re-established.  So, it has taken (1090-127) 963 seconds for JMRI to re-establish connection, and during this time, the user has no idea that the connection has failed!  This is a lot better than the 8000 seconds recorded earlier for JMRI to issue a RESET with no bus activity, but it is still 16 minutes, and JMRI looks as if it is working.  
During the time between 90 and 1090, the Raspberry Pi know that the hub is connected, as ARP packets are being exchanged.

closed 2022-04-26 by stale[bot]