JMRI/JMRI #10747: French signals for aspect signalling

John-Christopher95, 2022-02-24:
Hi, I have created xml-files and icons for French signals.
   https://fr.wikipedia.org/wiki/Signalisation_ferroviaire_en_France
almost all aspects are covered.
French signals are complex, so logix will have to be used.
the behavior of the signal is not exactly the same as reality for the flashing, it is voluntary.
the flashing signal is either on or off, for convenience the signal always remains on and a flash indicates the flashing.

I use my own decoders in Arduino under C/MRI, the signal is used with Output Matrix Mast and each aspect corresponds to an output word..
there are not many compatible decoders for french signals, so arduino is simple and cheap.
 (I use combination of output, like 0001, 0010, 0011... each combination of output (1001,1002,1003,1004) is an aspect for the decoder).
the files are not perfect, I am not sure of the logic of the signals between them then complex case, but it is to be tested for each one.
the signals are chosen and are not all present, the present signals can be used for signals with fewer aspects (with fewer lamps),  C use D, E use F, G use H, 2F can drive horizontal or vertical signal.
You can try it and if its 'a correct put them in new releases of JMRI

files for xml/signals:
[SNCF-2015.zip](https://github.com/JMRI/JMRI/files/8134830/SNCF-2015.zip)

files for preferences/resources/icons:
[SNCF.zip](https://github.com/JMRI/JMRI/files/8134835/SNCF.zip)

bobjacobsen, 2022-02-24:
Working on this now.  Some minor modifications are needed to get the files to validate, no problem there.  could you give me your full name, please, for the copyright notice?  (You can email it to rgj1927@gmail.com if you prefer)

bobjacobsen, 2022-02-24:
Signal XML files have to pass a validation step.  I fixed a couple small issues where I could figure out the proper content, but am stumped on two of them.

In xml/signals/SNCF-2015/appearance-BLJ_GUIDON.xml, there's this section:
```
  <specificappearances>
    <danger>
      <aspect>GuidonLF</aspect>
      <aspect>GuidonMF</aspect>
    </danger>
    <permissive>
      <aspect>GuidonLO</aspect>
      <aspect>GuidonMO</aspect>
    </permissive>
    <held>
      <aspect>GuidonLO</aspect>
      <aspect>GuidonMO</aspect>
    </held>
    <dark>
      <aspect>GuidonLO</aspect>
      <aspect>GuidonMO</aspect>
    </dark>
  </specificappearances>
```
with two `<aspect>` elements under each appearance.  Only one is allowed.  Could you tell me which to keep?

Also, in xml/signals/SNCF-2015/aspects.xml there are some incomplete elements that are missing `<speed>` elements. Could you tell me what to put on these?
```
 <aspect>
      <name>IDDark3</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction éteint</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>25</dccAspect>
 </aspect>
 <aspect>
      <name>ID1_3</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>26</dccAspect>
 </aspect>
 <aspect>
      <name>ID2_3</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>27</dccAspect>
 </aspect>
 <aspect>
      <name>ID3_3</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>28</dccAspect>
 </aspect>
 <aspect>
      <name>IDDark_4</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction éteint</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>29</dccAspect>
 </aspect>
 <aspect>
      <name>ID1_4</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>30</dccAspect>
 </aspect>
 <aspect>
      <name>ID2_4</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>31</dccAspect>
 </aspect>
 <aspect>
      <name>ID3_4</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>32</dccAspect>
 </aspect>
 <aspect>
      <name>ID4_4</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>33</dccAspect>
 </aspect>
 <aspect>
      <name>IDDark_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction éteint</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>34</dccAspect>
 </aspect>
 <aspect>
      <name>ID1_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>35</dccAspect>
 </aspect>
 <aspect>
      <name>ID2_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>36</dccAspect>
 </aspect>
 <aspect>
      <name>ID3_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>37</dccAspect>
 </aspect>
 <aspect>
      <name>ID4_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>38</dccAspect>
 </aspect>
 <aspect>
      <name>ID5_5</name>
      <rule>Article ?</rule>
      <indication>Indicateur de Direction</indication>
      <description>tableau fixe lumineux.</description>
      <comment>Indique la direction géographique.</comment>
      <route>Either</route>
      <dccAspect>39</dccAspect>
 </aspect>
```

Thanks again for the signal definition.

Bob

John-Christopher95, 2022-02-24:
hello Bob, 
For xml/signals/SNCF-2015/appearance-BLJ_GUIDON.xml, 
I understand, Maybe i have to realise two xml file.
-GuidonLF & GuidonMF are the same = Stop
LF is a light bar and MF a manual signal for small station (protected by disque signal).
-GuidonLO & GuidonMO = Clear (and start signal after LF or MF) 

I think that two files is the solution, one for manual_GUIDON signal and one for light_GUIDON  signal.
and change links under aspects.xml too.
BLJ is a light bar on protection signal for indicate a short stop lenght at station.

see (in french)
https://lesiteferroviaire.pagesperso-orange.fr/signaux%20arret%20assimile.htm#Ancre6

For xml/signals/SNCF-2015/aspects.xml
the IDxxx is just additional light on signal for indicate direction ( logix with turnout)
 the number of light indicate direction count left to right 
it's not a signal with stop or clear indication, no affect on speed.
whe have to indicate something for speed in this case ?

see (in french) 
https://lesiteferroviaire.pagesperso-orange.fr/Les%20indications%20de%20direction.htm

thanks for your job
Jc

John-Christopher95, 2022-02-24:
Hello Bob,
the two xml files for replace xml\signals\SNCF-2015\appearance-BLJ_GUIDON.xml (file to erase)
[appearance-Light-GUIDON.zip](https://github.com/JMRI/JMRI/files/8136699/appearance-Light-GUIDON.zip)
[appearance-Manual_GUIDON.zip](https://github.com/JMRI/JMRI/files/8136700/appearance-Manual_GUIDON.zip)

the aspects.xml new file 
[aspects.zip](https://github.com/JMRI/JMRI/files/8136722/aspects.zip)

for index.shtml, line to replace 
    <li><a href="appearance-Manual-GUIDON.xml">guidon manuel</a>
    <li><a href="appearance-Light-GUIDON.xml">BLJ et guidon lumineux</a>

thanks
Jc

John-Christopher95, 2022-02-24:
error in file name, sorry
[appearance-Manual-GUIDON.zip](https://github.com/JMRI/JMRI/files/8136735/appearance-Manual-GUIDON.zip)
JC

John-Christopher95, 2022-02-24:
Bob, 
for speed, TIVR indicate speed to applie at turnout entry (no permanent, applie with turnout position, explain the dark signal).
TIVA is for indicate to decrease speed for obtain speed indicated at TIVR, only one speed by signal for a turnout.
I have nothing, because it's not clear how to indicate this (create a special speed ?) or the speed is managed by turnout ?
JC

closed 2022-02-26 by danielb987

bobjacobsen, 2022-02-26:
This is now available in the latest development build which you can download here:  https://builds.jmri.org/jenkins/job/development/job/packages/910/