JMRI/JMRI #10748: LogixNG Module NPE when using delayed execution

dsand47, 2022-02-24:
If a module is called by different CondtionalNGs while a timer in the module is running, an NPE can occur.
```
     [java] 12:55:14,045 implementation.DefaultConditionalNG   WARN  - ConditionalNG IQC:AUTO:0001 was aborted during execute: java.lang.NullPointerException [JMRI LogixNGThread]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrit.logixng.implementation.DefaultMaleDigitalActionSocket.execute(DefaultMaleDigitalActionSocket.java:44)
     [java] 	at jmri.jmrit.logixng.tools.debugger.DebuggerMaleDigitalActionSocket.execute(DebuggerMaleDigitalActionSocket.java:35)
     [java] 	at jmri.jmrit.logixng.implementation.DefaultFemaleDigitalActionSocket.execute(DefaultFemaleDigitalActionSocket.java:37)
     [java] 	at jmri.jmrit.logixng.actions.ExecuteDelayed$InternalFemaleSocket.execute(ExecuteDelayed.java:461)
     [java] 	at jmri.jmrit.logixng.implementation.DefaultConditionalNG.internalExecute(DefaultConditionalNG.java:151)
     [java] 	at jmri.jmrit.logixng.implementation.DefaultConditionalNG.lambda$execute$0(DefaultConditionalNG.java:141)
     [java] 	at jmri.jmrit.logixng.util.LogixNG_Thread.lambda$new$0(LogixNG_Thread.java:143)
     [java] 	at java.base/java.lang.Thread.run(Thread.java:829)
 ```

The NPE occurs for ignore on repeat and individual timers.  It does not occur when restart timer is used.  However, the first action is lost.   

The attached xml file has two sensors, a LogixNG with two conditionals and a module.  The module timer runs for 5 seconds.

[module-test.txt](https://github.com/JMRI/JMRI/files/8135154/module-test.txt)

danielb987, 2022-02-25:
@dsand47 
Confirmed. I understand why it happens but not yet on how to solve it.

danielb987, 2022-02-25:
**Explanation of the problem**

The symbol table with the current local variables is stored in the ConditionalNG. Each action and expression can get the conditionalNG by calling `super.getConditionalNG()`.

However, there are two cases when this doesn't work that needs special handling:

1. When calling a Module. The module can be called from several different ConditionalNGs, so it needs to store which ConditionalNG called it last. When the action CallModule calls a Module, the Module stores the current ConditionalNG in a map there the current thread is the key. When someone later asks for the ConditionalNG, the Module uses the current thread to look up the ConditionalNG.

2. When something is scheduled later, for example by ExecuteDelayed. In this case, ExecuteDelayed stores the ConditionalNG and uses it later when executing its child action.

The problem is that (1) and (2) collides when ExecuteDelayed is used in a Module. So I need to rework this to get it to work.

danielb987, 2022-04-11:
Work in progress. This needs to be fixed.

closed 2022-10-01 by danielb987