JMRI/JMRI #10758: [Regression] ESU function map pane not populated on current master

newHeiko, 2022-02-27:
Steps to reproduce:

. Import the attached roster entry into your roster
. Select it
. Click on "Program" (no matter the settings, Edit Only works)
. Switch to "Function Map" tab

Expected results, as on [v4.26](https://github.com/JMRI/JMRI/releases/tag/v4.26):
[image]

Results on master, as of yesterday:
[image]

So by "not populated" I mean all the graphical elements are created, but not filled with data from the roster file. Instead they behave as if all the associated CVs were 0.

Saving the roster entry also sets all the associated CVs to 0.

Changing the first line and saving the roster entry correctly saves, but on re-loading the saved change is not re-loaded.

I ran a "git bisect" and it points to commit 7ecfc2b6e4 introducing this. Since this is the first time I'm doing that, please take this with a grain of salt.

Thanks,
Heiko
[6641.xml.zip](https://github.com/JMRI/JMRI/files/8148771/6641.xml.zip)

bobjacobsen, 2022-02-27:
Thanks for noticing it and tracking it down!  Will fix.

bobjacobsen, 2022-02-27:
Thanks again.  Could you check #10760?  It looks right to me, but I'm not an ESU user.

dheap, 2022-02-27:
Just waking up and starting up computers. I'll check this against some of my ESU definitions.

dheap, 2022-02-27:
Confirmed the problem with existing V4 and 5 decoders.

danielb987, 2022-02-28:
Is this issue resolved by PR #10760? If so, this issue can be closed.

bobjacobsen, 2022-02-28:
I believe it's resolved.

closed 2022-02-28 by bobjacobsen

newHeiko, 2022-02-28:
I can confirm that #10760 resolves this issue.

Heiko