JMRI/JMRI #10795: Speedometer usability issues

bobjacobsen, 2022-03-08:
Michael Homscheidt has been testing the new KPF-Zeller speedometer support.  (This has a new serial interface, but uses the rest of the common speedometer code).  He has some observations:

> (1) The text area at the bottom right reads "Bachrus MS-DCC“ but shoudl be something like "KPF-Zeller Speed-Cat“
> (2) There are two radio buttons Forward and backward. My expectation is that only one is active. Means, if Forward is active and I press Backward then Forward becomes inactive. But, at least in my Germay localisation, both radio buttons are active. I then have to manually deactivate the other one.
> (3) When exporting the speed profile no file suffix is added. I would expect to have .csv automatically added
> (4) The exported speed profile has two top lines. The first with the name of the device (here Bachrus, see (1)) and a second one "Step,Speed(fwd KPH)“ because I selected KPH in the speedometer window. But from the values these are MPH and not KPH
> 
> Within the nex days I will try to figure out how to automatically speed match locomotives. My current issue seems to be that DecoderPro seems not to try to set CVs with Programming on The Main but seems to expect the loco on the programming track. I try to figure this out wit my Digitrax DCS52 and will try to write a guide which could be published on the Speedometer site.

I'll take item (1), (2) and the top-line part of (4).  Somebody who knows the speedometer code (better than me) will have to fix the KPH/MPH issue in (4).  I'm not sure whether (3) is a problem or not; that could be argued either way, I think.

danielb987, 2022-03-08:
> I'm not sure whether (3) is a problem or not; that could be argued either way, I think.

For me, it's very useful to have an extension on all files. For example, If I know that there is a csv file somewhere in the `jmri/jmrit/logixng` folder, but I have forgot where, I can do a `find jmri/jmrit/logixng -name "*.csv"` to find it.

Extensions also helps me to know what type a file is, if it's a Java file, a text file or an image.

Is it common on Mac or Linux to have files without extensions?

steamfan, 2022-03-09:
On Mac OS it is common to have file suffixes.

bobjacobsen, 2022-03-09:
The question is not whether to have suffixes.  The question is whether JMRI should automatically apply them here.

When JMRI proposes a full file name, for example when it has a default name for a file, that includes a suffix.  But that's not what's happening here.  Here, a file chooser with an _empty_ filename is provided.  If the user types a name it's used; if it has a suffix that's used, and if not, then not.

Should we add a suffix to a filename that was typed without one?  That's not obvious to me at all.

Alternatively, we could provide a default filename, including a suffix.  But I'm not sure what that filename should be, given that people might want to keep multiple profiles.

bobjacobsen, 2022-03-09:
PR #10800 has improvements for (1), (2) and the naming part of (4)

jerryg2003, 2022-03-10:
I can only relate my experience as a Windows user: if an application says it is exporting a text file, I expect that application to append ".txt" to whatever I type as the file name -- except if I have already typed ".txt" in which case it should not duplicate that.  Similarly, if it says it is exporting a csv format file, I expect it to append ".csv" and if it is exporting an xml file, I expect it to append ".xml" to whatever I type.  

The only program I can recall that leaves the suffix entirely up to the user is Notepad++ when you create a totally new file.  Notepad, otoh, always appends .txt, even if you have typed a different suffix.

crosland, 2022-03-10:
It's a long time since I worked on this code, but the selection of both 
forward and reverse radio buttons is deliberate, and correct, to run the 
profiling in each direction, plotting them in different colours. If the 
user only want to profile one direction then they can deselect the 
other.

See SpeedoConsoleFrame.profileTimeout():
                 if ((profileDir == ProfileDirection.FORWARD)
                         && dirRevButton.isSelected()) {
                     // Start reverse profile
                     profileDir = ProfileDirection.REVERSE;

I haven't had time to look at the other issues yet.

Andrew

bobjacobsen, 2022-03-10:
Thanks!

In that case, I’ll change then to regular buttons to avoid confusion.

Bob

> On Mar 10, 2022, at 5:57 AM, Andrew Crosland ***@***.***> wrote:
> 
> It's a long time since I worked on this code, but the selection of both 
> forward and reverse radio buttons is deliberate, and correct, to run the 
> profiling in each direction, plotting them in different colours. If the 
> user only want to profile one direction then they can deselect the 
> other.
> 
> See SpeedoConsoleFrame.profileTimeout():
> if ((profileDir == ProfileDirection.FORWARD)
> && dirRevButton.isSelected()) {
> // Start reverse profile
> profileDir = ProfileDirection.REVERSE;
> 
> I haven't had time to look at the other issues yet.
> 

—
Bob Jacobsen
***@***.***

bobjacobsen, 2022-04-20:
Addressed, at least in large part, by #10800 and #10854

closed 2022-04-20 by bobjacobsen