JMRI/JMRI #10797: Problem allocating enough memory to Java heap

Betcsdad, 2022-03-09:
I can’t get my 64 bit Windows 7 computer with 8gb of main memory to allocate more than 270 meg of Java heap space.

This machine is running JMRI version 4.26 with the 64 bit version of Java 11.

PS my other Windows 7 laptop is a 32 bit machine and it allocates over 900 meg  to the Java heap space.  It is running JMRI 4.26 and the 32 bit version of Java 11

PUlvestad, 2022-03-09:
Users group thread for reference
https://groups.io/g/jmriusers/message/202280

klk32003, 2022-03-09:
On my 64 bit Windows 7 computer with 4 GB of memory I get 1988 megabytes heap size (see below).

There is nothing special on Windows system start in my case. Open a CMD window to run ```msconfig``` and to see what Windows is loading at system start in your case.
Please compare your both computers. This could be a clue.

```
2022-03-09 11:06:15,554 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2022-03-09 11:06:15,572 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Klaus\JMRI\log\messages.log [main]
2022-03-09 11:06:15,573 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Klaus\JMRI\log\session.log [main]
2022-03-09 11:06:15,583 apps.Apps                             INFO  - PanelPro version 4.26+R381c8dfc32 starts under Java 11.0.13 on Windows 7 amd64 v6.1 at Wed Mar 09 11:06:15 CET 2022 [main]
2022-03-09 11:06:17,645 apps.Apps                             INFO  - Starting with profile Meine_JMRI_Bahngesellschaft.3f4bdbef [main]
2022-03-09 11:06:17,784 node.NodeIdentity                     INFO  - Using d1eef2e2-ce66-4d2c-ba02-adb52e5d7842 as the JMRI storage identity for profile id 3f4bdbef [AWT-EventQueue-0]
2022-03-09 11:06:18,148 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2022-03-09 11:06:18,150 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2022-03-09 11:06:18,478 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2022-03-09 11:06:18,478 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Klaus\JMRI\Meine_JMRI_Bahngesellschaft.jmri\ [main]
2022-03-09 11:06:18,478 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Klaus\JMRI\Meine_JMRI_Bahngesellschaft.jmri\ [main]
2022-03-09 11:06:18,479 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Klaus\JMRI\ [main]
2022-03-09 11:06:18,479 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Klaus\ [main]
2022-03-09 11:06:18,479 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2022-03-09 11:06:19,122 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2022-03-09 11:06:21,341 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 11.0.13 [AWT-EventQueue-0]
2022-03-09 11:06:21,344 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.2 [AWT-EventQueue-0]
2022-03-09 11:06:46,473 LoggingTest.py                        INFO  - Allocated maximum heap size 1988 megabytes [AWT-EventQueue-0]
2022-03-09 11:08:22,337 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
```

devel-bobm, 2022-03-09:
> with the 64 bit version of Java 11.

This is _insufficient_ information to define a particular JAVA vendor's JAVA Runtime Environment  release.  Java implementations are provided by a number of vendors, and the _release number_ can be a critical piece in debugging JAVA issues.

Betcsdad, 2022-03-09:
Thank you for the response.  I’m going to be out of town for a couple days and won’t be able to access the computer again until Friday.

Tom Pearson

Betcsdad, 2022-03-09:
I’m sorry I forgot to include that it was Java version 11.0.14 from AZUL .  I noticed it now says 11.0.14.1+1 

Tom Pearson

bobjacobsen, 2022-03-09:
If you run the LaunchJMRI.exe file (the usual launcher) with the /debug switch it will provide additional information about what it's doing.  Not a windows expert, not sure how to do that.

Betcsdad, 2022-03-22:
It appears some of my responses to the emails haven't made it here.  I have removed programs from the startup list, uninstalled the Azul version of Java and installed the Oracle version, etc. but the script never shows more than 260 megs of heap space.

 Here's the info and activating the debug switch

CommandLine: "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /debug apps.gui3.dp3.DecoderPro3
Default options: 
AppName: DecoderPro3
Class: apps.gui3.dp3.DecoderPro3
Parameters: 
Noisy: 6
Force32bit: 0
Profile: 
Alternate launcher: 1
Testing for x64...
Result: 1
Setting x64 registry view...
Checking 'JRE'...
Checking 'Java Runtime Environment'...
JavaPath: C:\Program Files\Zulu\zulu-11-jre\\bin\javaw.exe
Delete file: C:\Users\MikeA\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Remove folder: C:\Users\MikeA\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\MikeA\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files\Zulu\zulu-11-jre\\bin\javaw.exe C:\Users\MikeA\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Result: 0
JExePath: C:\Users\MikeA\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
PhysicalMemory: 7989m
Check if more than 4GB memory
More than 4GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 1997m
Checking maximum heap size...
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\MikeA\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\MikeA
to: C:\Users\MikeA
user.home OK
Options:   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djogamp.gluegen.UseTempJarCache=false -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x64;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1997m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\MikeA\JMRI\systemfiles" -Djmri.log.path="C:\Users\MikeA\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\apiguardian-api-1.1.0.jar;lib\batik-anim-1.14.jar;lib\batik-awt-util-1.14.jar;lib\batik-bridge-1.14.jar;lib\batik-constants-1.14.jar;lib\batik-css-1.14.jar;lib\batik-dom-1.14.jar;lib\batik-ext-1.14.jar;lib\batik-gvt-1.14.jar;lib\batik-i18n-1.14.jar;lib\batik-js-1.8.jar;lib\batik-parser-1.14.jar;lib\batik-script-1.14.jar;lib\batik-shared-resources-1.14.jar;lib\batik-svg-dom-1.14.jar;lib\batik-svggen-1.14.jar;lib\batik-transcoder-1.14.jar;lib\batik-util-1.14.jar;lib\batik-xml-1.14.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-compress-1.18.jar;lib\commons-csv-1.9.0.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\commons-logging-1.2.jar;lib\commons-net-3.6.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.10.0.jar;lib\jackson-core-2.10.0.jar;lib\jackson-databind-2.10.0.jar;lib\JavaVersionCheckWindow.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom2-2.0.6.jar;lib\jetty-http-9.4.28.v20200408.jar;lib\jetty-io-9.4.28.v20200408.jar;lib\jetty-security-9.4.28.v20200408.jar;lib\jetty-server-9.4.28.v20200408.jar;lib\jetty-servlet-9.4.28.v20200408.jar;lib\jetty-util-9.4.28.v20200408.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jinputvalidator-0.8.0.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.5.jar;lib\jmri.script.jsr223graalpython.jar;lib\jna-5.9.0.jar;lib\jna-platform-5.9.0.jar;lib\joal.jar;lib\json-schema-validator-1.0.28.jar;lib\jsoup-1.13.1-javadoc.jar;lib\jsoup-1.13.1.jar;lib\jsplitbutton-1.3.1.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.2.jar;lib\libusb4java-1.3.0-darwin-aarch64.jar;lib\libusb4java-1.3.0-darwin-x86-64.jar;lib\libusb4java-1.3.0-linux-aarch64.jar;lib\libusb4java-1.3.0-linux-arm.jar;lib\libusb4java-1.3.0-linux-x86-64.jar;lib\libusb4java-1.3.0-linux-x86.jar;lib\libusb4java-1.3.0-win32-x86-64.jar;lib\libusb4java-1.3.0-win32-x86.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\org-openide-util-lookup-RELEASE126.jar;lib\org-openide-util-RELEASE126.jar;lib\org.eclipse.paho.client.mqttv3-1.2.5.jar;lib\pi4j-core-1.2.jar;lib\purejavacomm-1.0.5.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\thumbnailator-0.4.8.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.3.0.jar;lib\usb4java-javax-1.3.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.28.v20200408.jar;lib\websocket-client-9.4.28.v20200408.jar;lib\websocket-common-9.4.28.v20200408.jar;lib\websocket-server-9.4.28.v20200408.jar;lib\websocket-servlet-9.4.28.v20200408.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.1.jar;lib\xercesImpl-2.12.0.jar;lib\xml-apis-ext-1.3.04.jar;lib\xmlgraphics-commons-2.6.jar
Check for any prepended/appended classpath entries
Final ClassPath: .;classes;jmri.jar;lib\apiguardian-api-1.1.0.jar;lib\batik-anim-1.14.jar;lib\batik-awt-util-1.14.jar;lib\batik-bridge-1.14.jar;lib\batik-constants-1.14.jar;lib\batik-css-1.14.jar;lib\batik-dom-1.14.jar;lib\batik-ext-1.14.jar;lib\batik-gvt-1.14.jar;lib\batik-i18n-1.14.jar;lib\batik-js-1.8.jar;lib\batik-parser-1.14.jar;lib\batik-script-1.14.jar;lib\batik-shared-resources-1.14.jar;lib\batik-svg-dom-1.14.jar;lib\batik-svggen-1.14.jar;lib\batik-transcoder-1.14.jar;lib\batik-util-1.14.jar;lib\batik-xml-1.14.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-compress-1.18.jar;lib\commons-csv-1.9.0.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\comm [... cut ...] jar;lib\websocket-api-9.4.28.v20200408.jar;lib\websocket-client-9.4.28.v20200408.jar;lib\websocket-common-9.4.28.v20200408.jar;lib\websocket-server-9.4.28.v20200408.jar;lib\websocket-servlet-9.4.28.v20200408.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.1.jar;lib\xercesImpl-2.12.0.jar;lib\xml-apis-ext-1.3.04.jar;lib\xmlgraphics-commons-2.6.jar" apps.gui3.dp3.DecoderPro3 
Return code from process: 0
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

bobjacobsen, 2022-03-22:
That’s really odd.

The "-Xms96m -Xmx1997m “ options say that JMRI is telling Java to provide a minimum of 96MB memory, and allow the memory usage to go up to 1997MB.  That’s all that the JMRI startup can say.  There must be something else, other than JMRI’s startup, that’s limiting how much Java can allocate.

Do you have a global JAVA_OPTS or _JAVA_OPTS set?

This stack overflow:
https://stackoverflow.com/questions/4401396/could-not-reserve-enough-space-for-object-heap

Says it might be a 32-bit java issue. Are you running java as 32 bit or 64 bit? Apparently 64bit works better with large physical memory

Bob

> On Mar 22, 2022, at 2:39 PM, Betcsdad ***@***.***> wrote:
> 
> It appears some of my responses to the emails haven't made it here. I have removed programs from the startup list, uninstalled the Azul version of Java and installed the Oracle version, etc. but the script never shows more than 260 megs of heap space.
> 
> 

—
Bob Jacobsen
***@***.***

Betcsdad, 2022-03-22:
Thank you.  I went looking in “environment variables” and found a max of 256 meg had been set.  I deleted the line, restarted and now have 1998 meg allocated to. A max heap size.

Not sure how or who set that up, but it’s gone now

Thanks again,

Tom Pearson

dheap, 2022-03-22:
@Betcsdad Good to hear you've found the cause and solved it. There had to be something specific to your installation.

Dave

Betcsdad, 2022-03-23:
I did the initial Java and JMRI installation on this laptop years ago.  I have no idea how a fixed maximum heap size would have been set, but I’m glad it’s working now

closed 2022-03-23 by danielb987

Betcsdad, 2022-10-11:
I am currently running the 64 bit version of Java from Azul.  I’m not sure how to set a global Java anything 😉

Tom