JMRI/JMRI #10834: Issues on CreateDecoder help page

AFromAu, 2022-03-26:
Page: https://www.jmri.org/help/en/html/apps/DecoderPro/CreateDecoder.shtml

> This new file should go in the `decoders` subdirectory in the `JMRI` preferences directory so that the program can find it. The `DecoderProConfig2.xml` file lives in the preferences directory, so you can search for that file to locate it. (See the [configuration files page](https://www.jmri.org/help/en/html/apps/DecoderPro/Files.shtml#location) for further details about how to find that directory and its contents)

Issues:
* `DecoderProConfig2.xml` does not exist on my Windows PC with JMRI 4.26. `DecoderProConfig3.properties` exists in Home/JMRI.
* It's not clear whether I should create the `decoders` subdirectory should be in the `JMRI` directory or in `JMRI/preferences` (which contains only `preferences.properties`).  There is also `JMRI/My_JMRI_Railroad.jmri`, but I assume that is for railroad-specific information and not for new decoders.

dheap, 2022-03-26:
@AFromAu 

There are definitely some issues with some of this page.

> This new file should go in the decoders subdirectory in the JMRI preferences directory so that the program can find it. The DecoderProConfig2.xml file lives in the preferences directory, so you can [...]

The link points to an old page and instead should point to:
<https://www.jmri.org/help/en/html/doc/Technical/ProfileFileStructure.shtml>

The `decoders` subdirectory must be created in the User Files Location, as specified in the updated link.

The existing paragraph mentions the "preferences directory,". This is confusing because it is referring  to the portable name (="preference:",used in XML files) of the User Files Location, not a directory called `preferences`.

Dave in Australia

bobjacobsen, 2022-05-15:
Closed by #10986

closed 2022-05-15 by bobjacobsen