JMRI/JMRI #10836: Translate decoder definitions in JMRI

RodrigoFXE, 2022-03-27:
Hello everyone, I hope with this I do not break any rules of the JMRI developer community
I have only been using JMRI for a few years but I found it excellent, but since I live in Mexico I speak Spanish and I would like to contribute to JMRI by translating into Spanish the definitions of the decoders that we use in DecoderPro and if possible the texts in the JMRI interface
I must clarify two things about myself, the first is that I have no idea about how to write code and the second is that I am a blind person, I don't know if that is too much of an impediment to being able to do what I want to contribute.
I have entered the JMRI folders and where the definitions are saved and I have opened them to see how to try to translate them but I did not understand what I have to change to do it, if you can tell me where and how I have to do it to be able to translate the definitions I will I will appreciate
I leave my email for those who want to contact me to explain how to make the contribution I want to make to JMRI
Greetings to all
rodrigo92fxe@gmail.com

bobjacobsen, 2022-03-27:
(Also replied via email)

Thank you for wanting to help with this!

Most of the translation looks like this:

```
  <variable CV="4" item="Decel">
    <decVal max="31"/>
    <label>Deceleration Rate</label>
    <label xml:lang="da">Decelerations Forsinkelse (0-31)</label>
    <label xml:lang="it">Decellerazione (0-31)</label>
  </variable>
```

Where there’s a “variable” element for the English default followed by multiple elements for different languages.

I found this example:

 ```
 <enumChoice choice="Off" value="0">
    <choice>Off</choice>
    <choice xml:lang="de">Aus</choice>
    <choice xml:lang="es">De</choice>
  </enumChoice>
```

It shows that the language code for Spanish is “es”.  It’s also an example of another form, where a “choice” element is followed by multiple choice elements for the different languages.

Are the railroad terms in Spain and Mexico the same?  If not, there’s also a way to do “es” for default Spanish, and then add special “es-mx” terms for the Mexico variant.

Are there decoders you’d like to start with?  I can help you find the files for those.

Many decoder definitions use files from the

xml/decoders/nmra

and

xml/decoders/parts

directories. If you start there you can make a lot of progress quickly. 

Let me know how I can help please.

RodrigoFXE, 2022-03-27:
Hello Bob
I replied to your email, thank you very much for your help!