JMRI/JMRI #10846: Digitrax interface modification?

mfgranville, 2022-03-30:
There is a discussion in the Digitrax-Users group about using DCC addresses above 9983 and extended addresses in the (0)-127 range. It has been confirmed that a JMRI throttle will allow accessing addresses up to 10239 and Engine Driver and WiThrottle will allow accessing addresses up to 9999 when using a Digitrax command station to run trains.

No one has yet confirmed that extended addresses (0)-127 can or cannot be accessed because the JMRI Digitrax interface does not give a short/long choice for that address range. However, one user reported:
“Observations:
The DCS51 can generate DCC long address packets in the range 0-127.
The DCS51 can program DCC long address packets in the range 0-127.
How ?
Using the JMRI - Tool - Send DCC packet. The IMM packet is generated and DCC long address can be observed on the DCC track. I may assume that DCS50, DB150, DCS100 etc. can use the same mechanism - the IMM packet to carry "raw DCC packet".
The challenge is that the IMM packet with the DCC long address is one time event only. It looks like it is not in the refresh queue, similarly like e.g. accessory packets.”

So, the question now is – if JMRI developers can confirm this observation, are they willing to alter the Digitrax interface so that short/long becomes a choice when using addresses 1-127 with JMRI throttle, Engine Driver, and WiThrottle to run a train with a Digitrax command station? Or, is this just not possible? Perhaps the distinction between short and long addresses is made by the command station with no info other than address value coming from the throttle?

My personal interest is the following – I use NCE PowerPro at home and Digitrax DCS200 at the club. One of my favorite locos is a PRR GE 44-tonner #9999 which I have never brought to the club because I assumed I could only access up to 9983. I really didn’t want to get into the address programming syndrome just to go between home & club. Since I use Engine Driver at the club, I now know I can run this loco at the club without changing the address. Could it become possible to also transport locos with long address 1(or 0)-127?

Mark Granville

bobjacobsen, 2022-03-30:
The JMRI code change is very easy if the hardware supports it.  Whether it will make trouble for Digitrax users is a more subtle question.

Are there Digitrax throttles that can operate long address 1-127? I.e. physical devices, either made by Digitrax or somebody else?

bobjacobsen, 2022-03-30:
Looking through LocoNet PE hardware spec, I don't think the hardware can handle 0-127 as long addresses.  Addresses are handled by "slots" (Digitrax's name for the refresh facility) as an integer, without any long/short information.

mfgranville, 2022-03-30:
No Digitrax throttles that can send long addresses 1-127. I suspect it may
not be a possible modification because it may be that the Digitrax command
station makes the long/short decision based on the address value. I don't
think the Digitrax throttles send long/short info. But all that is above my
pay grade, so I've asked here.

 

My club's DCS200 command station will not accept addresses above 9983 from a
DT402 or UT4 throttle, but will accept addresses up to 10,239 from a JMRI
throttle. This led to the question of long address below 128.

 

<https://github.com/JMRI/JMRI/issues/10846#issuecomment-1083675429> , or
unsubscribe
TFXBANCNFSM5SDJHTNA> .
<https://github.com/notifications/beacon/AYPRLHM4EAFSB5VPLZDT7NLVCTFXBA5CNFS
M5SDJHTNKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOICLZGJI

mfgranville, 2022-03-30:
I was afraid of that. The upper limit for long addresses seems to be imposed by the throttle but the short. Long seems to be imposed by the command station.

newHeiko, 2022-04-01:
> Looking through LocoNet PE hardware spec, I don't think the hardware can handle 0-127 as long addresses. Addresses are handled by "slots" (Digitrax's name for the refresh facility) as an integer, without any long/short information.

FWIW, I agree (and already stated so on the digitrax-users list on 3/23 UTC). Unless Digitrax' extensions to LocoNet beyond the PE add some sort of band-aid, there is no way to get a long address 0-127 into a LocoNet command station.

(Discounting the DCC IMM packet, which is just passing through the command station, not getting into the refresh logic etc.)

I suggest this issue be closed WONTFIX, unless anyone wants to implement a full-blown DCC refresh stack in JMRI and use only DCC IMM packets on LocoNet - which would drop all interoperability with LocoNet hardware throttles.

Heiko

closed 2022-04-01 by bobjacobsen

mfgranville, 2022-04-01:
I suggest this issue be closed WONTFIX

 

Excuse my ignorance. Is this something I need to do as author of the issue, or will one of you developers do this?

 

Mark Granville