JMRI/JMRI #10849: Error 306

RobertRydin, 2022-03-31:
Issue Type: __Bug__

I am using a Digitrax DCS 52 and anMacBook Air running Montery (12.1)

I have a Serial to USB converter cord - attached to a hub which attaches to the USB C port on my mac.

In trying to identify a locomotive on my programming track, I consistently get the message "Timeout talking to commnd station (306)e

Locos I have purchased with DCC installed run correctly.  Locos whare i have installed decoders run forward and backward - but have no light functionality.

I am unable to identify the intalled decoder as I cannot identify it on DecoderPro.

I've been told by a credible source this is due to change APPLE has made - there is no workaround.  Is this true?

Please help?

Thanks,

Robert Rydin
robert@robertrydin.com
r

DecoderPro 4.26+R381c8dfc32
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.321-b07 mixed mode
Mac OS X 12.2.1 x86_64

dsand47, 2022-03-31:
> Issue Type: **Bug**
> 
> I am using a Digitrax DCS 52 and anMacBook Air running Montery (12.1)
> 
> I have a Serial to USB converter cord - attached to a hub which attaches to the USB C port on my mac.
> 
> In trying to identify a locomotive on my programming track, I consistently get the message "Timeout talking to commnd station (306)e
> 
> Locos I have purchased with DCC installed run correctly. Locos whare i have installed decoders run forward and backward - but have no light functionality.
> 
> I am unable to identify the intalled decoder as I cannot identify it on DecoderPro.
> 
> I've been told by a credible source this is due to change APPLE has made - there is no workaround. Is this true?
> 
> Please help?
> 
> Thanks,
> 
> Robert Rydin [robert@robertrydin.com](mailto:robert@robertrydin.com) r
> 
> DecoderPro 4.26+R381c8dfc32 Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.321-b07 mixed mode Mac OS X 12.2.1 x86_64

Robert,

After getting the 306 error, go to Help => System Console.  Click on "Copy to clipboard" and paste into a reply.

RobertRydin, 2022-04-07:
Somewhere in the JRMI messages, I saw a reference to Apple MacBook Air withe the M1 chip.  I have the M1 Chip.

Can someone point me to that message or related help?  Still getting the "306" error- my Mac is not seeing the DCS 52

bobjacobsen, 2022-05-13:
If this is still a problem for you, please contact the people on JMRIusers group at https://groups.io/g/jmriusers  There are people there who can walk you through debugging this.  It could be that you need to install drivers, or it could be a permissions issue, or something else.

closed 2022-07-12 by stale[bot]