JMRI/JMRI #10887: Slot Max: Loconet Slot not being released by JMRI Throttle.

Pugwash1, 2022-04-17:
When a throttle is released the slot is not released as there are still active listeners which is used to check if the throttle is being used by someone/something else. It may effect other shared throttle use cases.
I think it was caused by the removal of this code.

`--- a/java/src/jmri/jmrit/throttle/ThrottlesTableModel.java
+++ b/java/src/jmri/jmrit/throttle/ThrottlesTableModel.java
@@ -56,11 +56,7 @@ public class ThrottlesTableModel extends AbstractTableModel implements AddressLi
 
     @Override
     public void notifyAddressReleased(LocoAddress addr) {
-        if (addr instanceof DccLocoAddress ) {
-            DccLocoAddress la = (DccLocoAddress) addr;
-            fireTableDataChanged();
-            jmri.InstanceManager.throttleManagerInstance().removeListener(la, this);
-         }
+        fireTableDataChanged();
     }
 
`

bobjacobsen, 2022-04-17:
Do you know which commit that was?  Seeing it in context might make clearer why it was done.

devel-bobm, 2022-04-17:
It is important to note that code in java/src/jmrit affects _all_ connection types, not just LocoNet-type connections.  Perhaps the coder did not realize the problems it could cause, such as you apparantly see in LocoNet functionality.

This emphasizes the need for better tests of throttles, including LocoNet throttles, so that such jmrit changes would not cause LocoNet-specific failures.

Pugwash1, 2022-04-17:
#10853 was the pull.
It was removed from the table model because the table model didn't add it. But it wasn't moved to where it was created. I think there's a reason it's removed there as it's the only place which knows when it can be removed...

danielb987, 2022-04-17:
My apologize.

I think the correct fix is to not have `ThrottlesTableModel` as an `AddressListener`, but instead let the classes that registers it as an AddressListener to have its own AddressListener that calls `ThrottlesTableModel.fireTableDataChanged()`. That will remove the confusion.

The reason for my change was that ThrottlesTableModel is unable to remove itself from the ThrottleManager since it doesn't know which ThrottleManager the loco address belongs to. With #10853, you can have one throttle with loco 123 for the LocoNet connection and another throttle with loco 123 for the XpressNet connection.

danielb987, 2022-04-18:
I think #10888 will fix this.

danielb987, 2022-04-20:
Closing this Issue since it's fixed by #10888.

closed 2022-04-20 by danielb987