JMRI/JMRI #10896: Rotated signal masts disappear when displaying certain aspects.

dsand47, 2022-04-21:
This is an issue identified in the user group:  https://groups.io/g/jmriusers/message/203723

Icons with the GIF NETSCAPE Application Extension disappear if they have been rotated.

The first part of the GIF file shows the extension:
```
das@dmba SL-2-high % hexdump -C rule-239.gif
00000000  47 49 46 38 39 61 19 00  2f 00 70 00 00 21 ff 0b  |GIF89a../.p..!?.|
00000010  4e 45 54 53 43 41 50 45  32 2e 30 03 01 01 00 00  |NETSCAPE2.0.....|
```

Loading the file using **GIMP** and then doing an `export` removes the extension and the file works fine when rotated. 

There are 361 aspect icons with the extension.

dsand47, 2022-04-22:
I have identified two solutions to this issue.

1. Modify the rotated animated icon change to include all icons:
```
diff --git a/java/src/jmri/jmrit/catalog/NamedIcon.java b/java/src/jmri/jmrit/catalog/NamedIcon.java
index de7060a37c..91bc90ec66 100644
--- a/java/src/jmri/jmrit/catalog/NamedIcon.java
+++ b/java/src/jmri/jmrit/catalog/NamedIcon.java
@@ -109,9 +109,9 @@ public class NamedIcon extends ImageIcon {
 
                 int numFrames = gifReader.getNumImages(true);
 
-                // No need to keep the GIF info if it's not animated, the old code works
-                // in that case.
-                if (numFrames > 1) {
+                // Treat all icons as animated since some might be tagged as animated even
+                // though they only have 1 frame.
+                if (numFrames > 0) {
                     gifState.mStreamMd = gifReader.getStreamMetadata();
                     gifState.mFrames = new IIOImage[numFrames];
                     gifState.mWidth = 0;
```

2.  Use a tool such as GIMP to review the 666 icons in resources and remove the NETSCAPE flag if there is only one frame.

Option 2 does not include icons in user file locations.

bobjacobsen, 2022-04-22:
Solution (1) has the great advantage that it works for any new image files that come along in the future, user-provided images, etc.  There's a (small; probably very small) performance disadvantage for image files with the problem, but it's restricted to just those. So I'd vote in favor of (1)

closed 2022-04-23 by danielb987