JMRI/JMRI #10921: DCCppThrottleTest Warnings / Throttle Function Array Size

icklesteve, 2022-04-28:
In DCCppThrottleTest testFunctionFormats() 
https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/dccpp/DCCppThrottleTest.java#L151

The tests ensure Function 61 can be sent.

The normal AbstractThrottle constructor supports Functions 0-28.
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/AbstractThrottle.java#L72

AbstractThrottle warns when a value higher than the array size is encountered, hence the 3 Warnings in a standard test run :
```
WARN  - Unhandled update function number: 61 jmri.jmrix.dccpp.DCCppThrottle [main] jmri.jmrix.AbstractThrottle.updateFunction()
WARN  - Unhandled get function: 61 [main] jmri.jmrix.AbstractThrottle.getFunction()
WARN  - Unhandled update function number: 61 jmri.jmrix.dccpp.DCCppThrottle [main] jmri.jmrix.AbstractThrottle.updateFunction()
```

3 ways to fix this.

A - Set a maximum number of functions in DCCppThrottle constructor ( 99 ? )
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/AbstractThrottle.java#L86

B - Assert the warnings in the test class.
As warning comes from different thread, use something like
```
JUnitUtil.waitFor(()-> ( JUnitAppender.checkForMessageStartingWith(
            "Unhandled update function number: 61") !=null ), "no unhandled function 61 on set true");
```

C - Update the AbstractThrottle function array size to expand whenever a higher function number is encountered.
Would this cause issues with other jmrix Throttles? Would require further investigation.

Imho the best option depends on dccpp protocol / dccpp hardware support.

Any thoughts @mstevetodd  ?

Ta, Steve.

mstevetodd, 2022-04-28:
The DCC++EX code supports up to 32767 functions. (I asked the author)

Not sure what C would entail, so A looks like a good path for me. Is there another adapter that already handles this properly I could copy?

icklesteve, 2022-04-28:
The only deviation from 0-28 functions that I can find is EcosDccThrottle, where in the constructor it calls 
`super(memo,32);`
This supplies array size for functions 0-31

So first step is adding the number of functions to the super call in DCCppThrottle at lines 42 and 56.

After that, DCCppThrottleTest will need tweaking
See the over rides at https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/ecos/EcosDccThrottleTest.java#L443

32767 is a decent array size, especially when most users won't be using the full range.

Note that 2 arrays of the specified size are created when in the AbstractThrottle constructor,
the normal function array, also the momentary Function array, so to use the full spec.

Is it really worth holding the extra 65400 flags per Throttle when most users won't be using the full range?

With 10 open Throttles, that's 654000 flags.

Certainly worth having a play with this before deciding how to proceed.

Steve.

bobjacobsen, 2022-04-28:
In it's infinite wisdom, the NMRA defined multiple _overlapping_ function spaces:  The usual 0-31 one and a second one called "Binary States" that are accessed using "Binary State Control Instructions".   See page 7 of 

https://www.nmra.org/sites/default/files/s-9.2.1_2012_07.pdf

Note that this extra address space _overlaps_ with the function space:  They each provide a 1, a 2, a 3 etc that are disjoint.  

It's not clear how the existing JMRI throttles should support the Binary States. They could be defined as "functions" 32-32767 but that doesn't give access to Binary States 1-31.  Or throttles could be extended to implement them as something separate.

mstevetodd, 2022-04-29:
Updated response from UKChris regarding DCC++EX:
_The 0-28 range is what we keep internally and what gets broadcast back to all throttles. The extended range is there for completeness because dcc has that, and someone wanted it,  so we support it._ 
So, I'm just going to change the 61 to 16 and move along. That'll prevent the warning.

closed 2022-05-03 by danielb987