JMRI/JMRI #10930: Block/Portal/Path Warnings

PurkissA, 2022-05-01:
Issue Type: __Bug__

Not sure if this is a bug or design feature but the Block/Portal/Path Warnings does not display, if you use tabbed oblock editing option, as it does if the preference tabbed oblock editing option is not chosen.

PanelPro 4.99.6+R7578dc0c9b
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.15+10-LTS mixed mode
Windows 10 10.0 amd64

danielb987, 2022-05-01:
@petecressman 
Is this something you want to look at?

PurkissA, 2022-05-01:
Well yes. I keep having to switch between tabbed and non tabbed versions. My thought is if it shows in one way (non tabbed entry as separate window) automatically, it should show in the other (tabbed entry), possibly as another tab). Not essential but being consistent across display versions.

petecressman, 2022-05-02:
I am unsure what the issue is. It may be philosophical. I don't know the intention of the developer of the tabbed version. Perhaps his idea is to only report errors if they occur and not try to anticipate possible errors. Premature warnings can be an irritant to an accomplished user.

Rather than having to switch preferences and restart, Block/Portal/Path Warnings can be accessed from CircuitBuilder. menus. Please look at that way of defining OBlocks, Portals and Paths. And then decide whether the inconsistencies among the three are desirable or faults.

PurkissA, 2022-05-02:
I have pointed out inconsistencies which arise from changing one option. I will leave it at that. It is up to the developers whether that is a design feature or error.

closed 2022-07-02 by stale[bot]