JMRI/JMRI #10957: MERG CBUS Event Table display

Ancientdays, 2022-05-08:
with the GUI display set to Nimbus the text on the MERG CBUS Node manager appears black and is not easy to read 

See screen image here 
[image]

Regards

Allan Gardner

icklesteve, 2022-05-08:
Hi @Ancientdays , welcome to the JMRI project.

I've uploaded a tweak to the selected row text colour which improves readability, #10960  .

Thanks for reporting,
Steve.

Ancientdays, 2022-05-08:
Hi Steve many thanks for the update. How do I access the latest version?

icklesteve, 2022-05-09:
Hi Allan,

The update should appear in the main code base once the change has been approved and merged,
with at least 24 hours before the change has been submitted.
This helps stop unwanted changes appearing in the code and gives people a chance to comment on the update.

The unofficial developement installers can be found at https://builds.jmri.org/jenkins/job/development/job/packages/

Steve.

closed 2022-05-09 by danielb987

danielb987, 2022-05-09:
@Ancientdays 
The PR is now merged and the change will be included in the next JMRI test release.

Ancientdays, 2022-05-09:
Many thanks, folks

Allan

PR #10960 (2022-05-09): CbusCommonSwing - set selected text colour
fixes #10957