JMRI/JMRI #10985: JMRI stopped working

Vitov1, 2022-05-13:
Greetings to all, I need your help, I have a problem with the JMRI installation.
1.- I had JMRI installed and it worked very well with version 4.99.8. But it stopped working and only this info comes out:

[image]

2.- I ignore the msg and Panel Pro opens, but when trying preferences, this information appears:

[image]
Finally, I have uninstalled JMTI version 4.99.8 and I have tried with version 4.26 and when opening it I get this information:

[image]
Please can you help me with this, because I have tried in many ways without success in the installation of JMRI. Thank you, big hug.

danielb987, 2022-05-13:
@Vitov1 
Welcome to the JMRI project!

> Finally, I have uninstalled JMTI version 4.99.8 and I have tried with version 4.26 and when opening it I get this information:

When you uninstall JMRI, it seems that JMRI is still running on the computer. Reboot the computer and check that it's not installed. Then try to install again.

Vitov1, 2022-05-13:
Hello Daniel, you know that I have uninstalled everything including Java, I have made sure that there is no JMRI files, but there is no way to reinstall JMRI, this problem is really strange, believe me I have done everything. The only thing left for me is to reinstall WIN 10. But it doesn't make sense, but if necessary, I will do it. Thanks for your help Daniel.

danielb987, 2022-05-13:
Hello Vitov,

Could you ask this question on the JMRI users group?
Join here:  https://groups.io/g/jmriusers/join
Browse here:  https://groups.io/g/jmriusers/topics

There are more people there to help you out on this. I'm not sure how to solve this, but I think others have ideas.

Once you have joined the users group and you send a message to the group, it might take a couple of days until a moderator approves the message. This is for spam protection and applies to all new users.

Vitov1, 2022-05-13:
Thank you Daniel, I really appreciate your information.

danielb987, 2022-05-21:
@Vitov1 
Is this issue resolved? If so, it would be good to close the issue.

Vitov1, 2022-05-21:
Hi Daniel, fortunately this issue was solved successfully. Tnks for all coop. a hug from Quito- Ecuador

danielb987, 2022-05-21:
You are welcome!

closed 2022-05-21 by danielb987