JMRI/JMRI #11042: Create Test Release 4.99.10

bobjacobsen, 2022-05-28:
This is the next release in the 5.0 cycle. It is intended to only contain specific changes and bug fixes.

danielb987, 2022-05-30:
@bobjacobsen @DanielBoudreau 
PR #11044 is approved, CI is OK and it has milestone 4.99.10. Should it be merged now?

DanielBoudreau, 2022-05-31:
There are only test improvements in PR #11044.  No functional changes. I'd like to see if the latest changes fix the Jenkins test failures.

https://builds.jmri.org/jenkins/job/development/job/alltest/lastCompletedBuild/testReport/

bobjacobsen, 2022-05-31:
There's one change in a java/src file, so I think this is @DanielBoudreau 's call.

danielb987, 2022-05-31:
> There are only test improvements in PR #11044. No functional changes. I'd like to see if the latest changes fix the Jenkins test failures.

I'm fine with that. But @bobjacobsen wrote in [Test release 4.99.9 on Friday?](https://jmri-developers.groups.io/g/jmri/message/7355) that 4.99.10 will not be built from Head of Master, which means that if the 4.99.10 branch is already created, this PR needs to be merged in that branch.

danielb987, 2022-06-01:
@bobjacobsen 
Can we merge PRs for 5.1.1 now or should we wait until 5.0 is released?

bobjacobsen, 2022-06-12:
The release-4.99.10 branch has been created.

Jenkins will be creating files shortly at the [CI server](https://builds.jmri.org/jenkins/job/testreleases/job/4.99.10/)

bobjacobsen, 2022-06-12:
JMRI 4.99.10 has been released. Files are available in the GitHub release section.

closed 2022-07-10 by bobjacobsen