JMRI/JMRI #11043: Error configuring TurnoutOperation

dsand47, 2022-05-29:
While testing the DirtyManager changes I found an existing problem.

To re-create the problem follow these steps:

1. Start PanelPro with no files.
2. Go to Tools => Tables => Turnouts
3. Select Automation => Edit from the menu, click OK.
4. Add a turnout to the turnout table.
5. Click on Edit.
6. A Java exception occurs.

If the menu edit was not done, the edit of the turnout row works fine.  It appears that the menu based edit left an incomplete state or did not finish completely..

[session.log.zip](https://github.com/JMRI/JMRI/files/8792691/session.log.zip)

icklesteve, 2022-05-29:
I've had a play around with this, no quick fixes identified.
Would be minded to convert the TurnoutOperators to an SPI Interface.
This would replace the Reflection in TurnoutOperationConfig

Busy at work for a few days, will have another look on Wednesday if no one else gets to it before.

icklesteve, 2022-06-01:
With a fresh set of eyes over this, I spotted an issue with TurnoutOperations being added to the TurnoutOperationManager before being fully constructed, see #11052

icklesteve, 2022-06-05:
Hi @dsand47 , whenever convenient could you check that this has been resolved?

Thanks, Steve.

closed 2022-08-05 by stale[bot]