JMRI/JMRI #11063: new decoder file for Massoth "Pulsed Smoker for round boilers"

brucel1050, 2022-06-05:
hello,

hereby I attach a new decoder file for a rather new Massoth "Pulsed Smoker for round boilers" (product nr 8415001).

It would be lovely, if it could be added to the baseline package.

kind regards, Bruno

Bruno De Graeve

https://www.linkedin.com/in/brunodg/
[Massoth_pulsed_smoke_8415001.zip](https://github.com/JMRI/JMRI/files/8839941/Massoth_pulsed_smoke_8415001.zip)

`<?xml version="1.0" encoding="utf-8"?>
<?xml-stylesheet type="text/xsl" href="../XSLT/decoder.xsl"?>

<decoder-config xmlns:xi="http://www.w3.org/2001/XInclude" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/decoder.xsd">
  <version author="Bruno De Graeve brucel1050@gmail.com" version="1.0" lastUpdated="20220604"/>
  
  <decoder>
    <family name="Massoth eMOTION" mfg="Massoth Elektronik, GmbH">
      <model model="Pulsed Smoker for round boilers" maxInputVolts="24 V" maxMotorCurrent="N/A" maxTotalCurrent="1000 mA" numOuts="0" numFns="3" formFactor="G" connector="other" productID="8415001">
		<size length="54.0" width="426" height="30" units="mm"/>
      </model>
    </family>
    <programming direct="yes" paged="no" register="yes" ops="yes"/>
    <variables>
      <xi:include href="http://jmri.org/xml/decoders/nmra/shortAndLongAddress.xml"/>
      <variable CV="7" item="Decoder Version" tooltip="The decoder version number (read only)">
        <decVal/>
        <label>Software Version Number</label>
         <label xml:lang="it">Versione Decoder: </label>
        <label xml:lang="fr">Version décodeur: </label>
        <label xml:lang="de">Decoder Version: </label>
      </variable>
      <variable CV="8" readOnly="yes" default="123" item="Manufacturer" tooltip="The manufacturer's ID number (read only)">
        <decVal/>
        <label>Manufacturer ID</label>
        <label xml:lang="it">ID Costruttore: </label>
        <label xml:lang="fr">ID constructeur: </label>
        <label xml:lang="de">Hersteller ID: </label>
      </variable>
      <variable item="Programming lock - unlock" CV="15" default="162" tooltip="unlock programming lock (Key). Only if CV15 and CV16 are equal, programming of the decoder is possible.">
        <decVal/>
        <label>Programming lock - unlock</label>
      </variable>
      <variable item="Programming lock - lock" CV="16" default="162" tooltip="Preset fixed standard value for smoke generators (see manual). It's not recommended to change the value of CV16.">
        <decVal/>
        <label>Programming lock - lock</label>
      </variable>
	  <variable item="Load-dependent Operation" CV="49" mask="XXXXXXVX" default="1" tooltip="When load-dependency is activated (factory default) the vapor emission will depend on the acceleration (more emission) and deceleration (less emission)">
        <enumVal>
          <enumChoice choice="Off"/>
          <enumChoice choice="On"/>
        </enumVal>
        <label>Load-dependent Operation</label>
		</variable>
      <variable item="Clock Output" CV="49" mask="XXVXXXXX" default="" tooltip="Clock Output (see manual)">
        <enumVal>
          <enumChoice choice="Off"/>
          <enumChoice choice="On"/>
        </enumVal>
        <label>Clock Output</label>
		</variable>
      <variable item="Clock Input" CV="49" mask="XVXXXXXX" default="" tooltip="Clock Input (see manual)">
        <enumVal>
          <enumChoice choice="Off"/>
          <enumChoice choice="On"/>
        </enumVal>
        <label>Clock Input</label>
[... 264 lines cut ...]
	  <label><text>The Massoth Pulsed Smoke Generator for Round Boilers connects directly to track power and operates with DCC speed steps,</text></label>
	  <label><text>hall-effect sensor, or clock signal from a decoder.</text></label>
	  <label><text>With a newly designed fan and optimized airflow this smoke unit makes more smoke than ever.</text></label>
	  <label><text>Overheating and dry-run protection are integrated.</text></label>
	  <label><text>The pulsed smoke generator is equipped with a fully integrated decoder that features a standard loco address and a separate programming address.</text></label>
	  <label><text>POM programming of the smoke generator is thus possible.</text></label>
	  <label><text>the smoke generator's heating element and fan may be independently activated and controlled for optimum smoke effects</text></label>
	  <label><text>several CVs are available for setting temperature of the heating element,</text></label>
	  <label><text>preheating of the element, and various fan speed.</text></label>
	  <label><text>Load-dependent smoke generation is possible when the generator is connected to the SUSI interface terminal of a Massoth eMOTION decoder.</text></label>
	  <label><text></text></label>
	</column>
	  <column>
	  <label><text>Technical specifications:</text></label>
	  <label><text></text></label>
	  <label><text>Operating voltage : 18 -24V DCC</text></label>
	  <label><text>Operating current: 1000mA</text></label>
	  <label><text>Ambient operating temperature: -4°F to 122°F / -20°C to +50°C</text></label>
	  <label><text>Capacity: 5cc (optimum); 10cc (Maximum untill spill over)</text></label>
	  <label><text>Dimensions: 54x42x30mm</text></label>
	  <label><text></text></label>
    </column>
  </pane>
</decoder-config>
`

bobjacobsen, 2022-06-05:
Thanks!

See PR #11066

brucel1050, 2022-06-05:
After all the years I could use JMRI for "free", it was time to give something back ;-)

danielb987, 2022-06-06:
@brucel1050 
Thanks! Your files will be included in the next releases.

closed 2022-06-06 by danielb987