JMRI/JMRI #11091: CbusCreateBeanPaneTest.testCreateBean:76 "Reminder Dialog closed" did not occur in time

danielb987, 2022-06-12:
Several Windows CI tests has failed lately due to this error:
`CbusCreateBeanPaneTest.testCreateBean:76 "Reminder Dialog closed" did not occur in time`

```
2022-06-12T01:24:19.6429495Z java.lang.AssertionError: "Reminder Dialog closed" did not occur in time
2022-06-12T01:24:19.6429978Z 	at org.junit.Assert.fail(Assert.java:89)
2022-06-12T01:24:19.6430425Z 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:465)
2022-06-12T01:24:19.6431043Z 	at jmri.jmrix.can.cbus.swing.CbusCreateBeanPaneTest.testCreateBean(CbusCreateBeanPaneTest.java:76)
2022-06-12T01:24:19.6432524Z 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
```

See:
https://github.com/JMRI/JMRI/runs/6846730142?check_suite_focus=true , run `#2`
https://github.com/JMRI/JMRI/runs/6846727965?check_suite_focus=true , run `#2`
https://github.com/JMRI/JMRI/runs/6846751264?check_suite_focus=true , run `#1`

dsand47, 2022-06-12:
This is most likely caused by PR #11085 along with PR #11089.

PR #11089 sets enableStoreCheck true by default.  PR #11085 bypasses reminder dialogs if enableStoreCheck is true.

danielb987, 2022-06-12:
Thanks.

Could you try to call `setEnableStoreCheck(false)` in the `setUp()` and `resetProfileManager(Profile)` methods in the class JUnitUtil? The reason to why I think both places are needed is that resetProfileManager(Profile) overrides the setting from setUp() if it's called. And setUp() doesn't call resetProfileManager(Profile) so not all tests reset the profile.

dsand47, 2022-06-12:
PR #11092 has been submitted.

I also had to add the change to resetInstanceManager().

The CBUS test has an extra step which cleared the setup step.
```
        JUnitUtil.setUp();
        JUnitUtil.resetInstanceManager();
```

danielb987, 2022-06-12:
Closing this since it's solved by PR https://github.com/JMRI/JMRI/pull/11092.

closed 2022-06-12 by danielb987