JMRI/JMRI #11094: Decoder CSV Export file has Incorrect header.

dheap, 2022-06-13:
A created Decoder CSV Export file has an incorrect  header line of `"CV, value"`. It should be `CV,value`.

This causes a fatal error when you attempt to Import CSV File.

dheap, 2022-06-13:
I'm not in a position to track this one at present.

bobjacobsen, 2022-06-13:
Thanks. Could you please remind me how to export decoders in CSV?  I don't see it, but maybe I'm looking in the wrong places.

dheap, 2022-06-13:
With a Roster Entry, the File menu has Export… and Import… items.

Recently someone request a changed-only option, which you provided. Something may have broken then.

dheap, 2022-06-13:
#10623 was the PR I suspect.

bobjacobsen, 2022-06-13:
Thanks! Sorry for the trouble.  Fix is in #11096

danielb987, 2022-06-13:
I noted a couple of other things too with export/import CVs:

* If an error occurs, it's not handled. An exception is shown in the log but no error message is given to the user.
* The export exports to my Documents folder but the import exports from my profile folder.
* The export gives no extenstion to the file name. But the import expects a csv extension.

danielb987, 2022-06-15:
Closing this since it's solved by #11096.

closed 2022-06-15 by danielb987