JMRI/JMRI #11107: Multiple MQTT Lights, controlled by Fast Clock Follower, are not reliable.

rdschwemmer, 2022-06-19:
Experiment:
     Starting with no .jmri folder.
     Setup MQTT connection
     Add 2 MQTT lights 
           ML20, Controller: Fast Clock Follower, On 06:00 – Off 20:00 
           ML30, Controller: Fast Clock Follower, On 20:05 – Off 05:55 
     Setup the Fast Clock for 60:1
     File | Store all table content & panels...
     Add the .xml file to Start Up (preferences)
     File | Store all table content & panels…

At this point I monitor the MQTT broker with the command:
	mosquitto_sub -v -t '/#' | xargs -d$'\n' -L1 sh -c 'date "+%d.%m.%Y %T $0"' | ccze -m ansi
… in a shell window.  At this point the result is always as expected.  (Both Topics change ON & OFF every 12 minutes.)

However, if I then close Panel Pro and restart it. There is a > 50% chance that the MQTT broker will see one of the topics ‘yammering’ ON & OFF, about 17 times per second.  EG:
	18.06.2022 18:51:05 /trains/track/light/30 OFF 
	18.06.2022 18:51:05 /trains/track/light/30 ON 
	18.06.2022 18:51:05 /trains/track/light/30 OFF 
	18.06.2022 18:51:05 /trains/track/light/30 ON 
	18.06.2022 18:51:05 /trains/track/light/30 OFF 
	18.06.2022 18:51:05 /trains/track/light/30 ON 
And if I repeat closing Panel Pro and restarting it. There continues to be a > 50% chance that the MQTT broker will see one of the topics ‘yammering’.   The probability of success with only One Light, controlled by the fast clock, in the table is much higher.

This experiment has produced the same results with versions 4.25.6 & 4.99.10 running on Linux Mint 20.2 and with 4.99.10 running on Windows 10.  The broker is always Mosquitto 1.6.9-1 running on Linux Mint 20.2.

bobjacobsen, 2022-06-19:
Try setting "Light receive topic" in the preferences to be different than the light send topic.

[image]

rdschwemmer, 2022-06-19:
> Bob:
>
> Yes, based on opening & closing panel Pro 5 times, changing the "Light
> receive topic" did stop the 'yammering'!  I'm thinking that there is
> more to the story?
>
> Thank you very much for the quick response.

Now I see the "reply to" 
***@***.***

-------- Forwarded Message --------
Subject: 	Re: [JMRI/JMRI] Multiple MQTT Lights, controlled by Fast Clock 
Follower, are not reliable. (Issue #11107)
Date: 	Sun, 19 Jun 2022 09:19:31 -0500

bobjacobsen, 2022-06-19:
My hypothesis is that your broker is sending messages from JMRI back to JMRI.  The received message, when it matches the topic, sets the light's status.  What could be causing the yammering is e.g.
1) JMRI sends "off"
2) Time changes, JMRI decides to send "on" too
3) Reply to off comes back, JMRI changes to off
4) Reply to on comes back, JMRI changes to on
and so on.  But that's just a hypothesis, I haven't tried to recreate this.

rdschwemmer, 2022-06-19:
I need to reiterate that the 'yammering' does not start until Decoder Pro
is restarted, after adding 2 or more (mqtt) lights controlled by the fast
clock.  (the messagers are correct until the restart.)  And also, I'm
pretty sure that it does not happen if there is  only one mqtt light is
controlled by the fast clock.  I've  been running one light for months, and
have not noticed the problem.

Should I  run the 'experiment' with the broker on a different device?  I
think I have it on a raspberry pi.

closed 2022-08-31 by stale[bot]