JMRI/JMRI #11112: LogixNG feature request: display “listen” parameter value

jerryg2003, 2022-06-20:
Is it possible to display the “listen” parameter where appropriate on LogixNG conditional display and browse windows?  Otherwise you have to go into “edit” for each part of a conditional to see whether “listen” is checked.

For any expression or action where “listen” is available, e.g.:
`    ! A
        Publish by memory "MemoryData" (Listen) to topic by memory "MemoryTopic" (NoListen)`

Thanks.

danielb987, 2022-06-21:
I will have to think about that before implementing it. Some users might want it, some users might not.

danielb987, 2022-09-07:
I will reopen this PR since it would be good to fix this issue. But I don't know when or if I will be able to do it.

danielb987, 2022-09-12:
This will be solved by PR #11415.

closed 2022-09-13 by bobjacobsen