JMRI/JMRI #11146: Changing Address using Comprehensive Programmer

Alan-Lomax, 2022-07-02:
Issue Type: __Bug__

New install of JMRI V5. (with JRE 11)
New out of the box loco with new Loksound V5 decoder. Programming on programming track (via DCC++). First thing to do is change address away from 3. Using decoder pro in comprehensive mode the address change would not 'take'. It reported it as written and even reading it back (whole sheet) said it had changed but calling up a throttle said otherwise - it would only reposnd to address 3. Writing to CV1 using the CV programmer worked first time.

PanelPro 5.0+Rc441642522
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.15+10-LTS mixed mode
Windows 10 10.0 amd64

[profile.zip](https://github.com/JMRI/JMRI/files/9032222/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/9032223/session.log)

danielb987, 2022-07-02:
@Alan-Lomax 
Welcome to the JMRI project!

Which address are you trying to give the loco? I don't have much experience in programming locos, but locos can have a short or long address. 3 is a short address. If the new address is long, I think you need to tell it to use a long address instead of a short address.

closed 2022-09-03 by stale[bot]

Alan-Lomax, 2022-10-11:
Hi Daniel 

Thanks for getting back. I am quite conversant in changing addresses. In this case it was from 3 to 5 (both short addresses). My layout is small enough to not need longs.

The dcc++ has been in use for over a year. I used the same setup 2 weeks ago to change address on another loco. (That was Jmri  4.26 Jre 8) which happened without issue.

Al

"Numquam Urinare Contra Ventum”