JMRI/JMRI #11154: Cmri problem

steve2611967, 2022-07-04:
Can't do batches of cmri lights, have to do them one by one, but you can under cmri turnouts. Using version 5.0

KenC57, 2022-07-04:
What do you mean by 'batches' of lights?

danielb987, 2022-07-05:
@steve2611967 
Welcome to the JMRI project!

Do you mean that you can add CMRI turnouts in a sequential range, but that you cannot add CMRI lights in a sequential range?

steve2611967, 2022-07-09:
Yes

danielb987, 2022-07-09:
@steve2611967 
Did you intentionally closed this Issue? If there is an error, it would be good to fix it.

steve2611967, 2022-10-11:
No I closed it by accident, press wrong button.

steve2611967, 2022-10-11:
If I want to make say 8 lights for cmri, I have to do that one by one. You
can't put in a start number, say 2001 and set a sequential range.
Which you can in turnouts.

icklesteve, 2022-10-11:
Hi @steve2611967 I was going to have a look at this before it was closed, I presumed resolved!

I can't see an issue with adding sequential ranges for CMRI Lights, probably easier than CMRI Turnouts as there is no need to consider the number of Control Bits, see #11513 

Steve.

steve2611967, 2022-10-11:
You still can't because add sequentially box is ghosted out, it didn't use to be.

icklesteve, 2022-10-11:
Hi @steve2611967 , which version of JMRI was adding ranges of CMRI Lights via BeanTable possible?

I can't see any recent changes to
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/cmri/serial/SerialLightManager.java
regarding this, which is the where the code needs to updated .

I saw no problem with adding the feature of add ranges of CMRI Lights, hence created Pull Requested #11513 which will enable the sequential checkbox in future JMRI versions.

Steve.

PR #11513 (2022-10-13): Update CMRI Lights - MultipleAdditions
enable allowMultipleAdditions in create Light from BeanTable
fixes #11154
closed 2022-10-13 by danielb987