JMRI/JMRI #11156: Right click Layout Editor doesn't work

DanielBoudreau, 2022-07-05:
The "Marker" and operators loco icons no longer show the pop-up menu when the layout editor isn't in edit mode.

danielb987, 2022-07-05:
Is this with current master?

If so, could you create a simple panel with these things and upload here? You can attach the file to this Issue if you rename the file to .txt.

DanielBoudreau, 2022-07-05:
5.1.1 I’ll try current master and get back to you.

Dan

DanielBoudreau, 2022-07-05:
Broken with latest from master.  No right click menu.  You can easily test using the marker add loco feature.  Just create a new layout editor panel, disable edit mode, add loco icon.

Dan

DanielBoudreau, 2022-07-05:
Your fix was good for the panel editor.  Layout editor right click doesn’t produce the popup menu.  In edit mode, the popup menus do appear.

Dan

bobjacobsen, 2022-07-06:
I seem to recall that’s intentional.  The things in the pop-up menu are editor related.

Bob

> On Jul 5, 2022, at 7:33 PM, Daniel Boudreau ***@***.***> wrote:
> 
> Layout editor right click doesn’t produce the popup menu. In edit mode, the popup menus do appear.

—
Bob Jacobsen
***@***.***

DanielBoudreau, 2022-07-06:
Both the operations train icon and marker icons have popup menus that aren’t editor related.  This image is from 5.0 where it worked correctly

***@***.***

DanielBoudreau, 2022-07-06:
The problem was identified by one of our JMRI users:

https://groups.io/g/jmriusers/message/205898

dsand47, 2022-07-06:
Layout Editor markers work fine on macOS.  Add, move, popup and remove all work.

dsand47, 2022-07-06:
Here is a log of dragging a marker and two popups on macOS.  I don't have access to a Windows machine.

```
INFO  - ---- marker mouse pressed with meta, begin drag

INFO  - ---- marker mouse pressed delayed popup trigger
INFO  - ---- marker mouse released, popup trigger/delayed trigger, not dragging

INFO  - ---- marker mouse pressed delayed popup trigger
INFO  - ---- marker mouse released, popup trigger/delayed trigger, not dragging
```

The code changes.

```
diff --git a/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java b/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
index c702344592..a568d35978 100644
--- a/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
+++ b/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
@@ -3095,6 +3095,7 @@ final public class LayoutEditor extends PanelEditor implements MouseWheelListene
             // not in edit mode - check if moving a marker if there are any
             selectedObject = checkMarkerPopUps(dLoc);
             if (selectedObject != null) {
+                log.info("---- marker mouse pressed with meta, begin drag");
                 selectedHitPointType = HitPointType.MARKER;
                 startDelta = MathUtil.subtract(((LocoIcon) selectedObject).getLocation(), dLoc);
             }
@@ -3102,6 +3103,7 @@ final public class LayoutEditor extends PanelEditor implements MouseWheelListene
             // not in edit mode - check if a marker popup menu is being requested
             LocoIcon lo = checkMarkerPopUps(dLoc);
             if (lo != null) {
+                log.info("---- marker mouse pressed delayed popup trigger");
                 delayedPopupTrigger = true;
             }
         }
@@ -3626,6 +3628,7 @@ final public class LayoutEditor extends PanelEditor implements MouseWheelListene
             // requesting marker popup out of edit mode
             LocoIcon lo = checkMarkerPopUps(dLoc);
             if (lo != null) {
+                log.info("---- marker mouse released, popup trigger/delayed trigger, not dragging");
                 showPopUp(lo, event);
             } else {
                 if (findLayoutTracksHitPoint(dLoc)) {
@@ -3916,6 +3919,7 @@ final public class LayoutEditor extends PanelEditor implements MouseWheelListene
 
                 if (lo != null) {
                     showPopUp(lo, event);
+                    log.info("---- marker mouse clocked, popup trigger");
                 }
             }
         }
```

Hopefully, this will help determine the Windows issue.

It appears that Linux uses a right click and hold to drag the marker and control-right click for the popup.

danielb987, 2022-07-06:
> The "Marker" and operators loco icons no longer show the pop-up menu when the layout editor isn't in edit mode. I confirmed that it worked okay in 5.0. Bug was reported and found in 5.1.1. So this is a recent bug.

I have confirmed it and I'm working on it.

danielb987, 2022-07-06:
This is fixed by PR #11161.

danielb987, 2022-07-06:
@bobjacobsen 
I think I have got it right this time. But I'm getting more and more confused about the inconsistency in Java about this. See the method `isPopupTrigger()` in PR #11161. I need to check the OS, I need to check the event ID, and I need to check both META and BUTTON3 to get it to work.

Now I'm worried about the method `isMetaDown()`. Is it broken too? I don't want to fix it, unless I know it's broken. And I don't know if it is.

Do you know if other projects have similar problems like us, or is this specific to the JMRI project?

Or to put the question in another way: Are we handling mouse events the wrong way and get to pay the price for it? If so, what should we do instead, if it's possible to fix?

danielb987, 2022-07-06:
@bobjacobsen 
After a second thought, I realize that our problem is that we are using right mouse button to move items, instead of left mouse button. And therefore we cannot have popup on mouse button down.

bobjacobsen, 2022-07-07:
> After a second thought, I realize that our problem is that we are using right mouse button to move items, instead of left mouse button. And therefore we cannot have popup on mouse button down.

Right.  We have interactions that aren’t really mainstream Human Interaction cases, so we find odd things in the implementation.

Moving back to the basic interactions (click, drag, double-click, control/right click) would allow the code to be simpler and would help people who use multiple editor types.

But I’m not sure of that solution, though.  A lot of users have habits that would be hard to change.

danielb987, 2022-07-08:
> The "Marker" and operators loco icons no longer show the pop-up menu when the layout editor isn't in edit mode. I confirmed that it worked okay in 5.0. Bug was reported and found in 5.1.1. So this is a recent bug.

PR #11161 is merged. Can you check that it solves the problem for you?

DanielBoudreau, 2022-10-11:
Dragging objects using right click down is very unnatural for me.  Forever a Windows users.

Dan

icklesteve, 2022-10-14:
Control Panel Editor uses Left-click ( Win7 ) to drag objects around, feels much more natural.
Leave the right-click to open menus and switch to left-click for drag operations in all Editors imho.

geowar1, 2022-10-14:
I really (REALLY) wanted to fix this way back in 2016 when I started working on JMRI… but for two reasons I didn’t: 1) all the documentation (manuals, web pages & tutorials) to date would have to be fixed or be wrong and 2) the LayoutEditor mouse event dispatch code is a huge tangle of if/else if/else if…else… logic with overlapping cases… a total nightmare of spaghetti code to clean up… and I could never develop the will power to tackle it.