JMRI/JMRI #11157: Panel Editor can't move marker or train icon with right click

DanielBoudreau, 2022-07-05:
Test version 5.1.1 no longer allows the marker or operations train icon to drag to a new location using the right click on a mouse.

DanielBoudreau, 2022-07-05:
With a little more testing, it seems that all items can't be dragged even if the option to allow repositioning is enabled.

DanielBoudreau, 2022-07-05:
@danielb987   Is the the bug being addressed by your PR 11155?

danielb987, 2022-07-05:
Yes, I think so. Could you test #11155 and verify?

DanielBoudreau, 2022-07-05:
I don’t know how to pull a PR into my IDE.  However, I’ll test after your PR gets merged into master.

danielb987, 2022-07-05:
> I don’t know how to pull a PR into my IDE. However, I’ll test after your PR gets merged into master.

Someone needs to approve the PR. If you do that, the PR can be merged within an hour, either by you or me, and you can then test it.

DanielBoudreau, 2022-07-05:
PR #11155 fixes this issue.

closed 2022-07-05 by DanielBoudreau