JMRI/JMRI #11189: Extraneous log.info in mqtt.MqttSignalMast

jerryg2003, 2022-07-15:
Using 5.1.2 under Windows 10, Java 11.  Found the following line among the startup of my connections (Sprog Sim, MQTT, C/MRI Sim):

2022-07-14 20:40:49,408 simulator.SimulatorAdapter            INFO  - SPROG Simulator Started [SPROG Simulator]
2022-07-14 20:40:49,878 mqtt.MqttSignalMast                   INFO  - sendTopicPrefix set to signalmast/ [main]
2022-07-14 20:40:49,884 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2

This was before any panels loaded so no signal masts defined.  Looked at code and found 5 instances of log.info.  Should these be changed to log.debug? (I'll do it if the author is OK with that).  Note that no log.info appear in MqttLights or MqttSensors, for example.

bobjacobsen, 2022-07-15:
Is that the only thing that MQTT puts in the startup log? Or is there something like the "started" and "starting to connect" messages for MQTT too? Being able to see in the log that MQTT is starting up is a good thing, so there needs to be at least one line that indicates that.

jerryg2003, 2022-07-15:
Agree, as you can see Sprog and CMRI startup messages (indeed, MQTT is my second connection so its message comes in between).  However, this is the only MQTT message and it isn’t like the others “XXX Started” or “XXX Starting to Connect”.  I loaded 4.99.1 and got the same result.

Also, would the other instances of log.info put messages that are useful info or were they originally there to help in debugging?

bobjacobsen, 2022-07-16:
Not entirely sure what you're asking here.  The "sendTopicPrefix set to signalmast/" is meant to help support people who don't understand why their MQTT setup isn't seeing activity, but could be set to `debug` if you'd prefer.

jerryg2003, 2022-07-16:
Admittedly, I’m big on “consistency” across different aspects of JMRI:  if Sprog and C/MRI (and presumably other connection types) have a “XXX is starting up message,” why not MQTT (perhaps “Connecting/connected to MQTT broker XXX” would be more appropriate).  

Not sure that any other “info” messages are necessary.  “warn” or “error” messages make sense when something unexpected happens (e.g. drops connection to broker).  But why put in “info” messages on signal masts for someone who doesn’t use them, even before a panel is loaded? 

Obviously not a really big deal, but I do try to document “user noticeable” anomalies when I find them…

Thanks for listening.

bobjacobsen, 2022-07-16:
See #11191

jerryg2003, 2022-07-16:
Looks good. Thanks.

closed 2022-07-16 by jerryg2003