JMRI/JMRI #11190: Spontaneous sensor activation in JMRI 5.0

jb60, 2022-07-15:
Issue Type: __Bug__

In JMRI 5.0 on Linux Mint 20.3 a running locomotive causes spontaneous sensor activation all over the layout -- also switching turnouts, perhaps as a consequence. Going back to JMRI 4.26 and all is fine.

PanelPro 4.26+R381c8dfc32
OpenJDK 64-Bit Server VM Private Build 11.0.15+10-Ubuntu-0ubuntu0.18.04.1 mixed mode, sharing
Linux 5.0.0-32-generic amd64

dsand47, 2022-07-15:
To diagnose this issue, we will need some supporting documentation.
- The `~/.jmri/log/session.log` file from a session when the problem occurs.
- The PanelPro layout xml data and any supporting components such as custom icons.
- Any scripts.
- Ideally, the connection specific monitor file.

devel-bobm, 2022-07-15:
My first thought is that some occupancy detection hardware is known to be unintentionally sensitive to the electrical behaviors of some locomotive decoders, and can cause "random-seeming" detection events in unoccupied detection sections far removed from the location of the decoder.

As such, please provide information about your command station, boosters, and occupancy detection hardware.

dsand47, 2022-07-15:
I assume that the DCC-EX connection in JMRI is DCC++.  Since I don't have the required hardware, I am using simulation mode for DCC++ and CMRI.

I do get random sensor changes but they occur for both 5.0 and 4.26.  This implies they are related to simulation mode.

There are 4 errors due to missing script files.  A zip file of the script files would be helpful.
- hbKreisAussen.py
- hbKreisInnen.py
- hbADurchfahrt.py
- hbKreisMitte.py

jb60, 2022-07-16:
[hbADurchfahrt.txt](https://github.com/JMRI/JMRI/files/9125275/hbADurchfahrt.txt)
[hbAutoInit.txt](https://github.com/JMRI/JMRI/files/9125276/hbAutoInit.txt)
[hbKreisAussen.txt](https://github.com/JMRI/JMRI/files/9125277/hbKreisAussen.txt)
[hbKreisInnen.txt](https://github.com/JMRI/JMRI/files/9125278/hbKreisInnen.txt)
[hbKreisMitte.txt](https://github.com/JMRI/JMRI/files/9125279/hbKreisMitte.txt)

Right. In JMRI I choose DCC++. In addition to the scripts you ask for, I included hbAutoInit.py, which runs at startup.

jb60, 2022-07-16:
[scripts.tar.gz](https://github.com/JMRI/JMRI/files/9125291/scripts.tar.gz)

Here is the zip.

dsand47, 2022-07-16:
The DCC++ simulator generates random sensor messages.  I disabled the random messages and JMRI behaves properly.

Real hardware will be required to determine the problem.

In the meantime, you could run the "DCC++ Traffic Monitor" and see if the extra sensor messages are coming from the layout.  

Another idea is to test each of the 4.99 test releases and see which release causes the problem.  This helps narrow the search for code changes.

KenC57, 2022-07-19:
What is the chance this is a DCC++ firmware interaction issue? Did it get considered as one of the other
possible variables to how/why this is happening?

pdevlin13, 2022-09-12:
I have just recently upgraded to JMRI 5.0 and have had a similar issue, hence finding this thread.
I am running DCC++ with I/O on an Arduino Mega with serial CMRI.  I have block detection with a MERG current sensors with the input via CMRI, oddly in version 5.0 I can have the sensor input showing Active in the sensor table but the block in the block table showing unoccupied.  I also have a script with WaitsensorActive that continues to wait even though the sensor table is showing the input Active. It seems to happen when the next block is occupied but could not prove this repeatedly.
I originally suspected a floating input but scoping the inputs they are pulled up when not active and seem clean. I can repeatedly solve the problem by rolling back to 4.26 and repeatedly recreate the problem by going back to 5.0. I am currently as of tonight back to 4.26 and all is working again.

dsand47, 2022-09-12:
Try 5.1.3.

Dave Sand

pdevlin13, 2022-09-13:
Hi Dave 

Thanks for the advice, 5.1.3 has fixed the problem 😊

 

Paul.

jb60, 2022-09-13:
I also switched to 5.1.3. For each sensor I now receive a message: "Could not create sensor system Name ..." The turnouts table is lost, logixs are still there. I will have to start from scratch as far as sensors and turnouts are concerned.

dsand47, 2022-09-13:
What is the full text of the sensor name issue?   Better yet, post the contents of the system console.

I suspect you have a different problem.

Dave Sand

jb60, 2022-09-13:
Th full text is:
Could not create sensor name "DS1"
in adaptor of type jmri.jmrix.dccpp.configurexml.DCCppSensorManagerXml
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for more information

dsand47, 2022-09-13:
The most likely issue is that the DCC++ connection prefix is not actually "D".  Check Preferences => Connections.

Also review the system console for any errors during JMRI start up and verify that the connection was actually created and opened.

Dave Sand

jb60, 2022-09-14:
Connection prefix is D, connection name is DCC++

The warning concerning turnouts is:

Unexpected error (Exception)
while load(/home/…….xml)
in adaptor of type jmri.jmrix.dccpp.configurexml.DCCppTurnoutManagerXml
Exception: java.lang.IllegalArgumentException: Invalid system name for turnout: DT1 needed CT
See http://jmri.org/help…..

dsand47, 2022-09-14:
Do you have both CMRI and DCC++ connections?  

We are going to need to see the system console.  Use "Help => System Console".  Click on the "Copy to clipboard" button and paste into a reply.  Do not edit out any details.

Dave Sand

jb60, 2022-09-14:
Thanks for the hint about the System Console and all the other hints I was given here. Things are looking up. I am beginning to make better use of the diagnostics tools. Yes, I switched to 5.13. I will be able to continue on my own.

closed 2022-09-14 by jb60