JMRI/JMRI #11203: Missing dash in blockvalues.xml causes exception on startup

jerryg2003, 2022-07-20:
Per user group conversation (https://groups.io/g/jmriusers/topic/java_exception_during_startup/92467340?p=,,,50,0,0,0::recentpostdate/sticky,,,50,2,0,92467340,previd%3D9223372036854775807,nextid%3D1657499045717122019&previd=9223372036854775807&nextid=1657499045717122019), the  “Written by” comment in blockvalues.xml is missing a terminating dash, causing an exception to be thrown on JMRI startup.  Found in releases 5.0 and 5.1.2.

`

dsand47, 2022-07-20:
It does not appear to be a general bug.  I have blockvalues.xml files from 5.0, 5.1.1plus, 5.1.2, and 5.1.2plus and all of them have a proper comment.

jerryg2003, 2022-07-20:
Interesting:  I can consistently cause the issue by:

1. Deleting blockvalues.xml
2. Start JMRI
3. Load a panel file with blocks defined but no values.
4. Put a value in one of the blocks.
5. Save panel file.
6. Close JMRI

The blockvalues.xml that then appears has the incorrect comment line.

danielb987, 2022-07-20:
@jerryg2003 
Do you have this error on standard JMRI or on a version of JMRI that you have compiled yourself and made changes to?

If you have made changes to the JMRI sources and compiled it, and now get the error with that version, you need to check if you have the same error with unchanged JMRI sources.

jerryg2003, 2022-07-20:
Download my JMRI from the “releases” page on the web site.  I do not have the capability to compile my own version.

dsand47, 2022-07-20:
Are there any messages in session.log file after shutdown?

What tool are you using to look at the xml file?

jerryg2003, 2022-07-20:
Session.log just has normal shutdown message.  I'm using Notepad++ to look at the xml file.

HOWEVER, whatever you guys said above must have scared my system into working as I can no longer reproduce this problem!  

Here are the facts:
- I know I got an exception on startup (copied into user group message).
- I know I had a blockvalues.xml that was missing a dash (copied into user group, Dave Sand actually discovered it).
- I know I was able to reproduce this error on the day in question.
But I've been trying every combination of having block value, not having block value, having blockvalues.xml and deleting it, and I can't get JMRI to produce a blockvalues.xml with only a single dash on the comment line. So for now, I guess this issue is dormant...
Unless someone can point me to the code where the "Written by" comment is produced and I'll see if I can have a go at reading the code.

Thanks.

dsand47, 2022-07-20:
The code is in jmri.jmrit.XmlFile:
```
    static public void addDefaultInfo(Element root) {
        String content = "Written by JMRI version " + jmri.Version.name()
                + " on " + (new Date()).toString();
        Comment comment = new Comment(content);
        root.addContent(comment);
    }
```

There is nothing that can go wrong here, so the problem is either in Java, the `org.jdom2.Comment` module, or the operating system.  Or some unknown or random event such as a bit change caused by a stray cosmic particle.

jerryg2003, 2022-07-20:
Thanks for the info.  I’ll keep watch.  Does JMRI support cosmic ray sensors?

closed 2022-09-04 by danielb987