JMRI/JMRI #11207: Uninterpreted Lissy messages

arthkl, 2022-07-21:
Uhlenbrock's Lissy sensors (Reporters) causing some challenges on my layout.
All Lissys are configured to use the Uhlenbrock format to report: Loco address, Loco category, Loco speed and direction.
The current implementation interprets the loco address but does not seem to handle the other parameters. 
Instead it interprets the category as wheel count and seems to ignore the others.

Reading something like
_Unable to parse LocoNet message.
contents: E4 08 02 44 03 01 2B 7C_

it would be good enhance to implementation to consider the different formats Lissy can send information to LocoNet 

Send format Uhlenbrock with loco address and category
Send format Digitrax with loco address and block free
Send format Digitrax with loco address and block occupied

bobjacobsen, 2022-07-21:
Is there documentation available somewhere on the format and content of these messages?

arthkl, 2022-07-21:
I'm in contact with Uhlenbrock trying to obtain that info

arthkl, 2022-07-25:
While waiting for the documents, I did additional analysis with Lissys in all 4 possible configurations: (see LNCV=15)
Lissy not sending transmission protocol
Lissy using Uhlenbrock transmission protocol
Lissy using DigiTrax (train + block free)
Lissy using DigiTarx (train + block occupied)

It turns out that only one Lissy message contains a valid train number since DigiTrax compatible message don't get reported as Lissy messages at all.

I've made a small change to LnReporter (pushed the change to forked repository arthkl) to only report Lissy messages when bit 6 in the third element is set.
With that change all 22 Lissy receivers report correct train statuses (incl. directions)
Once I get more details from Uhlenbrock, I will change the LocoNetMessageInterprt as well.

Trace of all messages below:
**LNCV15 = 0 - No format**
2022-07-25 13:52:44,091 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 03 27  - Block occupied
 [AWT-EventQueue-0]
2022-07-25 13:53:11,914 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 02 26  - Block free
 [AWT-EventQueue-0]

**LNCV15 = 1 - Uhlenbrock format**
2022-07-25 13:57:23,327 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 02 44 12 _**01 2B**_ 6D - Train info northbound
 [AWT-EventQueue-0]
2022-07-25 13:57:23,554 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 03 27 - Block occupied
 [AWT-EventQueue-0]
2022-07-25 13:58:24,973 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 02 26 - Block free
 [AWT-EventQueue-0]

2022-07-25 13:59:00,665 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 02 64 12 **_01 2B_** 4D - Train info southbound
 [AWT-EventQueue-0]
2022-07-25 13:59:04,724 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 02 64 12 **_01 2B_** 4D - Train info southbound
 [AWT-EventQueue-0]

**LNCV15 = 2 - Digitrax (train + block frei)**
Transponder address 171 absent at LR531 () (BDL16x Board ID 34 RX4 zone B or BXP88 Board ID 67 section 3 or the BXPA1 Board ID 531 section). - DigiTrax not LISSY format
2022-07-25 14:01:58,458 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 03 27 - Block occupied
 [AWT-EventQueue-0]
2022-07-25 14:02:47,203 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 02 26 - Block free
 [AWT-EventQueue-0]

**LNCV15 = 3 - Digitrax (train + block occupied)**
Transponder address 171 present at LR531 () (BDL16x BLNCV15 = 0 - No format
2022-07-25 13:52:44,091 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 03 27  - Block occupied
 [AWT-EventQueue-0]
2022-07-25 13:53:11,914 messageinterp.LocoNetMessageInterpret DEBUG - Message from LISSY: contents: E4 08 01 24 12 00 02 26  - Block free
 [AWT-EventQueue-0]

bobjacobsen, 2022-07-25:
Very nice work!

devel-bobm, 2022-07-25:
I'd say this information is "a nice start".  But it is lacking the level of specific detail that is needed to truely interpret the LocoNet messages and to make JMRI's reporters "do something useful".  This effort will require either documentation from the manufacturer or a _lot_ more reverse-engineering work, with appropriate notes describing each test and its results.

arthkl, 2022-07-25:
Couldn't agree more. Let's see what Uhlenbrock can provide.

arthkl, 2022-07-30:
Since I haven't gotten any additional information from Uhlenbrock I went ahead and did some additional testing and ultimately code changes to the LocoNetMessageInterpreter. 
All Lissy messages are now interpreted and consistent with the IntelliBox II status display.

I've tested with a 3 sensor Lissy, 
[image]

configured to:
- send Uhlenbrock format (LNCV15 Option 1=1)
- locomotive speed (LNCV14>0)
- auto release block 10 seconds after loco left block (LNCV10 = 10+256) (256 means automatically switch signal back to red)

There are basically 3 main scenarios to consider.
A) Locomotive entering the block from the left (referring to the picture above) with the block signal being red (stopping at the signal (single sensor))
B) Locomotive entering the block from the left with the block signal being green (passing through the block)
C) Locomotive entering the block from the right (expected no changes to locomotive speed or block occupancy)

The recorded messages were as follow
A)
15:59:24.209: Lissy 18 IR Report: Loco 171 (category 3) moving north
15:59:24.228: Lissy 18 IR Report: Speed 68
15:59:24.449: Lissy 18 IR Report: Block occupied

After leaving the block (Manually switching block signal from red to green and automatically from green to red after 10 seconds)
16:02:06.168: Lissy 18 IR Report: Block free

B)
16:06:50.395: Lissy 18 IR Report: Loco 171 (category 3) moving north
16:06:50.413: Lissy 18 IR Report: Speed 68
16:**06:50**.597: Lissy 18 IR Report: Block occupied
16:**07:00**.570: Lissy 18 IR Report: Block free _(10 seconds delay based on LNCV10)_

C)
16:25:22.405: Lissy 18 IR Report: Loco 171 (category 3) moving south _(passing single sensor)_
16:25:26.059: Lissy 18 IR Report: Loco 171 (category 3) moving south _(passing double sensor)_
16:25:26.078: Lissy 18 IR Report: Speed 41

No block messages because block has been passed in the opposite direction.

As indicated, the status display at the IntellyBox II matches those message interpretations.
Please note: Nahverkehrszug is a configured description for category 3 
[image]

Since there is no uninterpreted Lissy message anymore on my layout, I would like to conclude this analysis.
I've pushed the changes to my forked repository for public review.
Also added the changed messages to the Dutch and Czech language files but encourage a native speaker to review.

Cheers, Arco

bobjacobsen, 2022-07-30:
Should be create a PR and merge these changes into JMRI itself?

devel-bobm, 2022-07-30:
I have concerns about the `loconet.LnReporter.lissyReport()` method as it exists, as well as the proposed change.  

- The proposed code specifically looks for bit 6 of LocoNet message byte 3 to be set, where the previous code does not look at that bit.  

- Neither the current JMRI code nor the proposed code pays any attention to bytes 2 or 4.  This risks inappropriate "aliasing" of messages other than reporter messaging as reporter messaging.

- I _do not see any test code.

- I do not know what hardware was used in preparing the original code and if/how it might differ from the specific hardware used by @arthkl .  Does the modified code work with _all_ hardware which makes use of this message?  There is risk that it might not.

I have no experience with Lissy hardware, and none available to me, so I have little to add beyond my observations and concerns.  Perhaps @pabender or @dsand47 can advise, as they have made modifications to the `lissyReporter()` method in the past.

dsand47, 2022-07-30:
I can provide no input since I have never touched LnReporter.

arthkl, 2022-07-30:
The current code is reporting a random number of wheels moving down south which is not only wrong because Uhlenbook's Lissys do not report wheels, it also messes up the reported id tags. 
Instead of reporting LD171 (which would be the code of a specific locomotive) it's nor reporting LD2 - or similar.

Is there anyone on that chat who is actually using Lissys? It would be nice to have that type of feedback.

Uhlenbrock  68610 is the hardware in question here. 
To be clear 68600 (which is the predecessor( does not report wheels either.

Also - the proposed code looks for byte 2 - as this is the category in case bit 6 is set
Also - byte 4 represents the unit (Lissy LNCV0 & 0x7F) something the reports does not seem to case about

arthkl, 2022-07-31:
BTW: The reporter manager LNReporterManager reads the 4th byte:
```
            case LnConstants.OPC_LISSY_UPDATE:
                if (l.getElement(1) == 0x08) {
                    addr =  (l.getElement(4) & 0x7F);
                    break;
                }
                return;
```
 It is used to create the SystemName for the reporter. It is therefor correct that the a LnReporter is ignoring that information and not reporting its own name. It is the reporter with that name reporting a status.

bobjacobsen, 2022-07-31:
@arthkl Your 'master' branch has both this change and your fix to the icon rotation problem in it.  I'd like to go ahead and make two PRs from those so they can be separately examined.  Is that OK with you?

arthkl, 2022-07-31:
Yes, absolutely. Please go ahead.

bobjacobsen, 2022-07-31:
See PR #11262.

It has some CI issues
 - There's a logging call that needs an update
 - Some of the tests need to be updated to changed message content:
```
Error:    LnReporterTest.testLnReporterLissy1:76 Lissy message 1 expected:<8501[] seen southbound> but was:<8501[:1] seen southbound>
Error:    LnReporterTest.testLnReporterLissy2:84 Lissy message 2 expected:<8501[] seen northbound> but was:<8501[:1] seen northbound>
Error:    LocoNetMessageInterpretTest.testLissy:884 Lissy message test 1 expected:<...R Report: Loco 8501 []moving south but was:<...R Report: Loco 8501 [(category 1) ]moving south
```

Once the discussion here converges, I'll be happy to fix those.

bobjacobsen, 2022-08-02:
The conversation seems to have died down.

To me, it looks like this code is progress, even if not known to completely and perfectly address Lissy messages.  So I think it would be reasonable to merge it. Other opinions welcome, particularly if they come with a plan for addressing any issues they raise.

arthkl, 2022-08-02:
Coincidently I received another email from Uhlenbrock today, indicating that they won't be able to help and suggesting that I should analyze the bytes myself. I've exhausted that avenue over the course of the last days and won't be able to provide any additional information.
However, since the initial issue "Uninterpreted Lissy messages" has been resolved, the issue can be closed from my point of view.
I realize that the test cases might now be out of line as some of the initial interpretation have changed. Happy to look into this if needed.
Cheers, Arco

bobjacobsen, 2022-08-02:
Do you have anything else on your master branch? I'd like to merge just these changes as a unit. If you have nothing else there, yes, please go ahead and fix those test comparisons.

Also, LocoNetMessageInterpret.java line 3783 should be changed to something like:
```
log.debug("Message from LISSY: {}", Bundle.getMessage("LN_MONITOR_MESSAGE_RAW_HEX_INFO", l.toString()));
```

devel-bobm, 2022-08-02:
@bobjacobsen,

> To me, it looks like this code is progress, even if not known to completely and perfectly address Lissy messages. So I think it would be reasonable to merge it. 

I agree, subject to the following comments.

@arthkl,
> I realize that the test cases might now be out of line as some of the initial interpretation have changed. 

If you intend to say that the "test cases may now need fixes" to support your changes: be aware that it is _your_ responsibility to investigate and fix any "broken" test cases, while ensuring proper JMRI behavior.

It is not unknown that a "broken" testcase reveals a flaw in the code changes that was not readily apparent to the developer; it is your responsibility to identify any such issues and provide the appropriate fix - either in the code, the testcase, or both.  It is also your responsibility to provide new testcases, as required, in order to maintain or improve the level of test "code coverage" on the JAVA source code files you have touched.

arthkl, 2022-08-02:
@devel-bobm I trust we all meaning well and I believe I said: "I'm happy to look into this". So please let's not engage in some generic discussions what everyones responsibilities might be. Thank you.

arthkl, 2022-08-02:
@bobjacobsen, 

Changes have been pushed into the master branch.
Running jmri.jmrix.loconet.messageinterp.LocoNetMessageInterpretTest does not show any errors.

devel-bobm, 2022-08-02:
@arthkl,

It was not my intent to offend, or to limit your contributions to JMRI in any way.  

It was only my intent to ensure that you, as a new contributor, know what is expected of you as a JMRI developer, so that collectively, _we_ can work toward a better JMRI product.

closed 2022-10-02 by stale[bot]