JMRI/JMRI #11208: BlockContentsIcon do not retain orientation when rendered

arthkl, 2022-07-21:
A blockContentsIcon with a rotation != 0 (270 degrees in my case) placed on a layout does not get rendered correctly.

When placed on the layout it appears correctly. 
However, after toggling (block free / block occupied again) the label gets rendered horizontally (rotation = 0).
Looking at the xml file, the entry 
    <BlockContentsIcon blockcontents="NW P1" x="1298" y="499" level="4" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" **degrees="270"** size="13" style="0" red="0" green="0" blue="0" hasBackground="no" justification="left" selectable="no" class="jmri.jmrit.display.configurexml.BlockContentsIconXml" />

Suspecting a rendering issue here.

bobjacobsen, 2022-07-21:
Which editor? Layout, Panel, Control Panel?

Also, could you attach the panel file please?  You'll need to temporarily add a .txt extension to it for GitHub to allow the upload.

arthkl, 2022-07-21:
That would be the Layout editor.
Apologies, what it the panel file, please? Is this the xml file where I've copied the section from?

bobjacobsen, 2022-07-21:
Yes, please.

arthkl, 2022-07-21:
Thanks - file attached.
[Moorbach.txt](https://github.com/JMRI/JMRI/files/9160603/Moorbach.txt)

arthkl, 2022-07-24:
Hi, after doing some debugging I believe the main reason for the issue is the setText(..) function in PositionableLabel.
It should call the rotate(..) function not only if it has a namedIcon but also if the rotation angle is !=0.

After making the change and testing for some unwanted null pointers, 
the layout started to show 90 and 270 degree rotated textlabels (see attached screen shot)
[image]

I was not able yet, to investigate why this only works for fixed dimensions - I had to add fixedWidth and fixedHeight to those BlockContentsIcon that have a degrees!=0. Otherwise the resulting dimension is 5x5 hardly displaying anything.

I've pushed the changes to a forked branch (arthkl) for further considerations.

PR #11261 (2022-08-02): Resolve issue with block icon rotations #11208 
Code from @arthkl to resolve #11208 on block icons retaining rotation. 

> setText should call rotated if the PositionableLabel has a rotation degree !=0 to allow a named icon to be built for text labels
> 
> Also fixed a few null pointer exceptions. Those exceptions were thrown
> during the edit mode.
closed 2022-08-02 by danielb987