JMRI/JMRI #11231: PR #10844 (4.99.5) can cause a NPE

dsand47, 2022-07-26:
The attached test has 3 blocks:  A-B-C.  If A and C are occupied and B becomes occupied, the value determination fails with a NPE.  The NPE can occur if the neighbor block values are null.  In this situation line 1066 effectively does a null.equals(null).

```
15:03:59,278 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
15:03:59,281 util.Log4JUtil                        INFO  - This log is appended to file: /Users/das/Library/Preferences/JMRI/log/messages.log [main]
15:03:59,281 util.Log4JUtil                        INFO  - This log is stored in file: /Users/das/Library/Preferences/JMRI/log/session.log [main]
15:03:59,337 apps.Apps                             INFO  - PanelPro version 5.1.2+Ra3b171c8ba starts under Java 11.0.13 on Mac OS X x86_64 v12.4 at Tue Jul 26 15:03:59 CDT 2022 [main]
15:04:02,729 apps.Apps                             INFO  - Starting with profile EdgeConnectors.3f43c6ad [main]
15:04:03,002 node.NodeIdentity                     INFO  - Using f1804893-46ed-4cb9-99c8-25c574503394 as the JMRI storage identity for profile id 3f43c6ad [AWT-EventQueue-0]
15:04:03,575 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
15:04:03,576 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
15:04:04,571 util.FileUtilSupport                  INFO  - File path program: is /Users/das/JMRI/Apps/JMRI_5.1.2/ [main]
15:04:04,571 util.FileUtilSupport                  INFO  - File path preference: is /Users/das/JMRI/_Profiles/EdgeConnectors.jmri/ [main]
15:04:04,571 util.FileUtilSupport                  INFO  - File path profile: is /Users/das/JMRI/_Profiles/EdgeConnectors.jmri/ [main]
15:04:04,571 util.FileUtilSupport                  INFO  - File path settings: is /Users/das/Library/Preferences/JMRI/ [main]
15:04:04,571 util.FileUtilSupport                  INFO  - File path home: is /Users/das/ [main]
15:04:04,572 util.FileUtilSupport                  INFO  - File path scripts: is /Users/das/JMRI/_Profiles/EdgeConnectors.jmri/ [main]
15:04:05,483 PanelPro.PanelPro                     INFO  - Main initialization done [main]
15:04:15,881 configurexml.LoadXmlConfigAction      INFO  - Loading selected file: /Users/das/JMRI/_Profiles/EdgeConnectors.jmri/block test.xml [AWT-EventQueue-0]
15:04:32,335 jmri.Block                            DEBUG - Block A goes UNOCCUPIED [AWT-EventQueue-0]
15:04:33,830 jmri.Block                            DEBUG - Block A goes OCCUPIED [AWT-EventQueue-0]
15:04:33,830 jmri.Block                            DEBUG - Sensor ACTIVE came out of nowhere, no neighbors active for block A. Value not set. [AWT-EventQueue-0]
15:04:36,046 jmri.Block                            DEBUG - Block B goes UNOCCUPIED [AWT-EventQueue-0]
15:04:38,146 jmri.Block                            DEBUG - Block C goes UNOCCUPIED [AWT-EventQueue-0]
15:04:38,800 jmri.Block                            DEBUG - Block C goes OCCUPIED [AWT-EventQueue-0]
15:04:38,801 jmri.Block                            DEBUG - Sensor ACTIVE came out of nowhere, no neighbors active for block C. Value not set. [AWT-EventQueue-0]
15:04:43,651 jmri.Block                            DEBUG - Block B goes OCCUPIED [AWT-EventQueue-0]
15:04:43,652 jmri.Block                            DEBUG - Block B has 2 active linked blocks, comparing directions [AWT-EventQueue-0]
15:04:43,652 jmri.Block                            DEBUG - comparing A (None) to B (East) [AWT-EventQueue-0]
15:04:43,652 jmri.Block                            DEBUG - comparing C (None) to B (West) [AWT-EventQueue-0]
15:04:43,655 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.Block.goingActive(Block.java:1066)
	at jmri.Block.handleSensorChange(Block.java:895)
	at jmri.Block.lambda$setNamedSensor$0(Block.java:303)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:293)
	at jmri.implementation.AbstractSensor.setOwnState(AbstractSensor.java:226)
	at jmri.implementation.AbstractSensor.setKnownState(AbstractSensor.java:165)
	at jmri.jmrit.display.SensorIcon.doMouseClicked(SensorIcon.java:617)
	at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseClicked(LayoutEditor.java:3906)
	at jmri.util.swing.JmriMouseListener$1.mouseClicked(JmriMouseListener.java:23)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6638)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6400)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4556)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

[block test.xml.zip](https://github.com/JMRI/JMRI/files/9193047/block.test.xml.zip)

bobjacobsen, 2022-07-26:
I'll fix it. Thanks for the test file!

danielb987, 2022-07-28:
Fixed by  #11238

closed 2022-07-28 by danielb987