JMRI/JMRI #11245: Change to speedometer default causes "save panels" dialog box

jerryg2003, 2022-07-28:
Using 5.1.2 on Windows 10:

I opened JMRI and opened a speedometer window.  Changed distances and hit "Save as default."  When I went to quit JMRI (never having opened a panel file), I got the dialog box saying panel info had changed and did a I want to save it.  I said "no."  FYI, the file PanelProSpeedometer.xml was correctly updated.

Note that in 5.1.1, I got that same dialog box whenever I opened a speedometer window, even without changing anything.  That was fixed in 5.1.2

dsand47, 2022-07-28:
Look at the system console.  It will show the first difference.

My guess is that the speedometer sensors were created by the speedometer dialog.

closed 2022-07-28 by jerryg2003

dsand47, 2022-07-28:
> @dsand47 <https://github.com/dsand47> System console doesn't show anything related to sensors or speedometer. When I open the speedometer, I checked the Sensor Table and nothing there. When I click "Save as deault", 3 new sensors are created corresponding to what is in the speeometer so indeed that is why I am being asked to "save" a new panel file. Thanks for this explanation of unexpected behavior. [I'll add a note on this into help file I am working on...]

Jerry,

I may have misled you a bit.  If a panel is never loaded, there is nothing to compare against.  So the default is to assume that something was changed.

Dave Sand